jsk-ros-pkg/jsk_recognition #2272: fcn test failing???

k-okada, 2018-04-25:


k-okada, 2018-04-25:
also something wrong on classifier @wkentaro 
```
jsk_perception/test_results/jsk_perception/rostest-fcn_object_segmentation__use_mask_true.xml: 1 tests, 1 errors, 0 failures, 0 skipped
jsk_perception/test_results/jsk_perception/rostest-test_fcn_object_segmentation.xml: 1 tests, 1 errors, 0 failures, 0 skipped
jsk_perception/test_results/jsk_perception/rostest-test_label_image_classifier.xml: 1 tests, 1 errors, 0 failures, 0 skipped
jsk_perception/test_results/jsk_perception/rostest-test_probability_image_classifier.xml: 1 tests, 1 errors, 0 failures, 0 skipped
```

wkentaro, 2018-04-25:
The classifier uses fcn and fcn uses chainer, so we have these failures.

wkentaro, 2018-04-26:
I tested on Jenkins with below few commands:

```
% sudo docker run -it ros-ubuntu:16.04 /bin/bash
$ fc -l 1
1        ls
2        cd 
3        ls
4        python
5        pip list | grep chainer
6        ls
7        cd 
8        ls
9        pip show chainercv
10       ls
11       cd
12       ls
13       mkdir catkin_ws
14       ls
15       cd catkin_ws/
16       ls
17       mkdir src
18       actkin init
19       catkin init
20       ls
21       sudo apt install python-catkin-tools -y -q
22       apt install python-catkin-tools -y -q
23       apt autoremove
24       ls
25       cd src
26       ls
27       lsa
28       cd ..
29       catkin init
30       ls
31       cd src
32       lsa
33       wstool init
34       ls
35       wstool set jsk-ros-pkg/jsk_recognition https://github.com/jsk-ros-pkg/jsk_recognition.git -v master -y -q
36       wstool set jsk-ros-pkg/jsk_recognition https://github.com/jsk-ros-pkg/jsk_recognition.git -v master -y -u
37       wstool set jsk-ros-pkg/jsk_recognition https://github.com/jsk-ros-pkg/jsk_recognition.git -v master -y -u --git
38       ls
39       source /opt/ros/kinetic/setup.bash
40       ls
41       rosdep update
42       rosdep init
43       rosdep update
44       clear
45       ls
46       rosdep install --from-path . -r -y
47       ls
48       apt update
49       rosdep install --from-path . -r -y
50       ls
51       catkin b jsk_perception
52       source ~/catkin_ws/devel/setup.bash
53       roslaunch jsk_perception sample_fcn_object_segmentation.launch GPU:=-1 gui:=false
54       tmux
55       apt install tmux -y
56       ls
57       tmux
58       ls
59       history
60       ls
61       cat ~/.bash_history 
62       fc -l
63       fc -l 1
```

Below shows some other information

```
root@6508769bcc1f:~# pip list | grep chainer
chainer (4.0.0)
chainercv (0.9.0)
^[[AYou are using pip version 8.1.1, however version 10.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
root@6508769bcc1f:~# pip list | grep cupy   
You are using pip version 8.1.1, however version 10.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
root@6508769bcc1f:~# 
```

PR #2280 (2018-05-01): Re-enable tests which use chainer inside them
Close #2272
closed 2018-05-01 by k-okada