jsk-ros-pkg/jsk_recognition #2288: virutal_mono_camera occasionally does not work?

furushchev, 2018-05-12:
I found that `virtual_mono_camera` occasionally publishes noised images (though I don't know if it is a bug or just a result from passing invalid parameters)

[image]

This node subscribes following messages:

```
Subscriptions: 
 * /tf [tf2_msgs/TFMessage]
 * /target_polygon [unknown type]
 * /tf_static [unknown type]
 * /view_point [unknown type]
 * /narrow_stereo/left/image_raw [unknown type]
 * /narrow_stereo/left/camera_info [unknown type]
```

We need to record `/view_point` and `/target_polygon` into bagfile on next time.

furushchev, 2018-05-12:
CC: @Affonso-Gui It was the positional problem that PR2 did not find 7th floor yesterday. (He could not find `↑ 7` on his camera (center))

YutoUchimi, 2019-11-12:
This is not a noise but the result of interpolation.
It occurs when the input image is enlarged too much.

I created a PR to enable changing the interpolation method in #2470 , so please try it.

closed 2019-11-12 by YutoUchimi