jsk-ros-pkg/jsk_recognition #2298: pointcloud_moveit_filter cannot be complied on kinetic?

sshige, 2018-05-31:
When I try to submit pull request #2297 , I got following error message in travis.

As I didn't change pointcloud_moveit_filter, I couldn't understand why the error happens. Does someone have idea why the following error happens?

I would appreciate if you would response this.

Errors << jsk_pcl_ros:make /workspace/ros/ws_jsk_recognition/logs/jsk_pcl_ros/build.make.000.log
In file included from /workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/pointcloud_moveit_filter.h:48:0,

                 from /workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_pcl_ros/src/pointcloud_moveit_filter.cpp:36:

/opt/ros/kinetic/include/moveit/occupancy_map_monitor/occupancy_map_monitor.h:44:38: fatal error: pluginlib/class_loader.hpp: No such file or directory

compilation terminated.

make[2]: *** [CMakeFiles/jsk_pcl_ros_moveit.dir/src/pointcloud_moveit_filter.cpp.o] Error 1

make[1]: *** [CMakeFiles/jsk_pcl_ros_moveit.dir/all] Error 2

make[1]: *** Waiting for unfinished jobs....

make: *** [all] Error 2

sshige, 2018-05-31:
Thanks @knorth55 !

Do you know how to avoid this error? Just wait ?

sshige, 2018-05-31:
ok. Thanks a lot!

knorth55, 2018-05-31:
I found out that `ros-ubuntu:16.04` docker image in `jenkins` is too old.
Therefore, there is no new `pluginlib` in the docker image.

knorth55, 2018-05-31:
@sshige I think https://github.com/jsk-ros-pkg/jsk_travis/pull/356 solve this problem, but Im not sure .

knorth55, 2018-05-31:
@sshige Travis is not updated, so the problem is not solved yet. Please wait my PR jsk-ros-pkg/jsk_travis#356 and its application to this repo.

sshige, 2018-05-31:
Ok. Thank you for your kind response.

knorth55, 2018-05-31:
I close PR jsk-ros-pkg/jsk_travis#356 and I updated docker image in jenkins.
I restart the build again.

knorth55, 2018-05-31:
Travis passed!

closed 2018-05-31 by knorth55