jsk-ros-pkg/jsk_recognition #2308: how to run tabletop tracking example

k-okada, 2018-07-26:
I ran following commands:
```
rosrun jsk_pcl_ros organized_multi_plane_segmentation
roslaunch jsk_pcl_ros tabletop_tracking.launch
roslaunch jsk_pcl_ros tracking.launch
```
Is it right?

cc: @sanketrahul

k-okada, 2018-07-31:
@knorth55 'I could not get the result.' is not  enough information. whenever you have something not working, you need detailed information. please remember https://huanghe389.wordpress.com/2009/06/02/the-confessional-method-of-debugging/

% I'm using intel zr300 ,which may slightly different from kinect?/xtion? but should be similar

1. start `roslaunch zr300_nodelet_rgbd.launch`
2. start `roslaunch jsk_pcl_ros tracking.launch input_point:=/camera/depth/points`
3. run rviz, subscribe `/camera/depth/points` or `/camera/depth_registered/points`
4. visualize '/particle_filter_tracker/track_result_pose` in rviz
5 add panel named "SelectPointCloudPublishAction" (see http://jsk-visualization.readthedocs.io/en/latest/jsk_rviz_plugins/panels/select_point_cloud_publish_action.html)
6 Follow instruction `Push the "Select" button at the top bar , drag and surround the target poincloud which you want to track in the rectangle area.Then, finally, push the "SelectPointCloudPublishActoin" button at SelectPointCloudPublishAction Panel. The tracker will start tracking the target.` (https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/doc/jsk_pcl_ros/nodes/particle_filter_tracking.md)

[image]

knorth55, 2018-07-31:
I ran with the following command, but no `/track_result_pose` returned.

```bash
roslaunch openni2_launch openni2.launch
roslaunch jsk_pcl_ros tracking.launch input_point:=/camera/depth_registered/points
rviz -d $(rospack find jsk_pcl_ros)/sample/rviz/tracking.rviz
```

Even I push `SelectPointCloudPublishAction` with selected point cloud, the launch output says noting.

```bash
$ roslaunch jsk_pcl_ros tracking.launch input_point:=/camera/depth/points
... logging to /home/shingo/.ros/log/97e294a4-94b4-11e8-8c87-ac1f6b2628a2/roslaunch-ecublens-39546.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ecublens:45705/

SUMMARY
========

CLEAR PARAMETERS
 * /particle_filter_tracker/
 * /selected_cloud_downsampler/
 * /voxelgrid/

PARAMETERS
 * /particle_filter_tracker/BASE_FRAME_ID: NONE
 * /particle_filter_tracker/align_box: False
 * /particle_filter_tracker/track_target_name: track_result
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /selected_cloud_downsampler/filter_limit_field_name: z
 * /selected_cloud_downsampler/filter_limit_max: 5.0
 * /selected_cloud_downsampler/filter_limit_min: 0.0
 * /selected_cloud_downsampler/leaf_size: 0.02
 * /voxelgrid/filter_limit_field_name: z
 * /voxelgrid/filter_limit_max: 5.0
 * /voxelgrid/filter_limit_min: 0.0
 * /voxelgrid/leaf_size: 0.02

NODES
  /
    particle_filter_tracker (nodelet/nodelet)
    renew_tracking (jsk_pcl_ros/renew_tracking.py)
    selected_cloud_downsampler (nodelet/nodelet)
    tracking_manager (nodelet/nodelet)
    voxelgrid (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

process[tracking_manager-1]: started with pid [39569]
process[voxelgrid-2]: started with pid [39570]
process[particle_filter_tracker-3]: started with pid [39571]
type is pcl/VoxelGrid
process[selected_cloud_downsampler-4]: started with pid [39576]
type is jsk_pcl/ParticleFilterTracking
type is pcl/VoxelGrid
process[renew_tracking-5]: started with pid [39589]
[ INFO] [1533036405.950063175]: Initializing nodelet with 48 worker threads.
[ INFO] [1533036406.615297545]: trans: [0.000000, 0.000000, 0.000000]
[ WARN] [1533036411.553058520]: '/particle_filter_tracker' subscribes topics only with child subscribers.
[ INFO] [1533036693.418856337]: trans: [0.060898, 0.018464, 0.534566]
[ INFO] [1533036693.418937203]: RESET TARGET MODEL
```

k-okada, 2018-07-31:
```
k-okada@p51s:/opt/ros/kinetic/share/jsk_pcl_ros/sample/rviz$ grep -i topic
tracking.rviz
          Topic: /particle_filter_tracker/track_result
          Topic: /particle_filter_tracker/particle
          Topic: /voxelgrid/output
      Topic: /initialpose
      Topic: /move_base_simple/goal
      Topic: /clicked_point
```
did you really select point cloud?

sanketrahul, 2018-08-03:
@k-okada How to do 4th step "visualize '/particle_filter_tracker/track_result_pose` in rviz" ?

k-okada, 2018-08-03:
Choose Add button from Displays panel , select Pose and set `
/particle_filter_tracker/track_result_pose` in topic
This is just to subscribe the output of tracking node, so it is equivalent
to `rostopic echo /particle_filter_tracker/track_result_pose`

sanketrahul, 2018-08-03:
@k-okada Thanks for the clarification. I did all the steps suggested by you but it didnot worked for me. 
[image]
As you can see I selected the white ball area and then moved it. But there is no tracking done.
I am getting following error message on terminal running command "roslaunch jsk_pcl_ros tracking.launch input_point:=/camera/depth/points" :
[ INFO] [1533316228.542621600]: Initializing nodelet with 12 worker threads.
[ INFO] [1533316229.499059214]: trans: [0.000000, 0.000000, 0.000000]
[ WARN] [1533316234.305023645]: '/particle_filter_tracker' subscribes topics only with child subscribers.
[selected_cloud_downsampler-4] process has died [pid 24585, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet standalone pcl/VoxelGrid ~input:=selected_pointcloud __name:=selected_cloud_downsampler __log:=/home/baxter2/.ros/log/fd941b6c-9741-11e8-a850-000af70eb628/selected_cloud_downsampler-4.log].
log file: /home/baxter2/.ros/log/fd941b6c-9741-11e8-a850-000af70eb628/selected_cloud_downsampler-4*.log

sanketrahul, 2018-08-06:
I cannot run tar command. It gives me error: tar: Removing leading `/' from member names
/home/baxter2/.ros/log/latest

sanketrahul, 2018-08-06:
For or portion when I run the command 'roslaunch jsk_pcl_ros tracking.launch input_point:=/camera/depth_registered/points' I get following error:
[tracking.launch] is neither a launch file in package [jsk_pcl_ros] nor is [jsk_pcl_ros] a launch file name
The traceback for the exception was written to the log file

k-okada, 2018-08-06:
did you run sudo apt-get install ros-kinetic-jsk-pcl-ros?

sanketrahul, 2018-08-07:
@k-okada Thanks for the suggestion. I have run 'sudo apt-get install ros-kinetic-jsk-pcl-ros' and also 'sudo apt-get install ros-kinetic-jsk-visualization' to run the OR section command successfully. But still I don't think code is tracking selected point cloud very well as shown in below screen sho
[image]
t:

k-okada, 2018-08-07:
it seems you still have
```
[rospack] Error: failed to parse command-line options: too many positional options have been specified on the command line
```
did you check `echo $ROS_PACAKGE_PATH` and that do include your workspace , after `source /opt/ros/kinetic/setup.bash`

also start from scene where the target object is not close to other objects and have enough points.
For example, hang your target object from the ceiling hooking on a string or something like that, or try to track backrest of chair.

sanketrahul, 2018-08-09:
Yes it works well. The was making a big mistake to move the object coming in front of the kinect camera because of which loss the point cloud track. When I moved from back of the table it worked well. 
[image]
For me it worked with following commands:
1. source /opt/ros/kinetic/setup.bash
2. roslaunch openni_launch openni.launch
3. roslaunch jsk_pcl_ros tracking.launch input_point:=/camera/depth_registered/points
4. run rviz, subscribe /camera/depth/points or /camera/depth_registered/points
5. visualize '/particle_filter_tracker/track_result_pose` in rviz by selecting 'Pose' in the display and further selecting ''/particle_filter_tracker/track_result_pose`' in topic.
6. add panel named "SelectPointCloudPublishAction" (see http://jsk-visualization.readthedocs.io/en/latest/jsk_rviz_plugins/panels/select_point_cloud_publish_action.html)
6.Follow instruction Push the "Select" button at the top bar , drag and surround the target poincloud which you want to track in the rectangle area.Then, finally, push the "SelectPointCloudPublishActoin" button at SelectPointCloudPublishAction Panel. The tracker will start tracking the target. (https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/doc/jsk_pcl_ros/nodes/particle_filter_tracking.md)