jsk-ros-pkg/jsk_recognition #2316: Z axis of bounding box from ClusterPointIndicesDecomposer is flipped

pazeshun, 2018-08-03:
I found z axis of bounding box generated by `ClusterPointIndicesDecomposer` is flipped from target frame.
[image]
This is because the default value of `force_to_flip_z_axis` param is true.
https://github.com/jsk-ros-pkg/jsk_recognition/blob/1.2.5/jsk_pcl_ros/src/cluster_point_indices_decomposer_nodelet.cpp#L101

I've already sent PR (#2315 ) to adapt doc to this.

However, I think this is unnatural behavior. I want to fix this, but I'm afraid I'll break existing recognition code.
Should I fix this?

wkentaro, 2018-08-03:
It seems the reason `force_to_flip_z_axis` is set as `true` in default is that the camera's z-axis is usually reverse to the z-axis of the world coordinates.
For someone who uses `align_bboxes_with_plane: true` for tabletop object manipulation, maybe flipping z-axis is a natural behavior.
Please check it by using [`sample_cluster_point_indices_decomposer.launch`](https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/sample/sample_cluster_point_indices_decomposer.launch#L19-L44)

Since changing the default value of this seems very crucial to the motion planning in existing euslisp programs, I recommend not to change the default value.

pazeshun, 2018-08-03:
@wkentaro 
Thank you very much for your detailed response.
I decide I don't change the default value, so I'll close this issue.

closed 2018-08-03 by pazeshun