jsk-ros-pkg/jsk_recognition #2317: [openni2_local.launch] is neither a launch file in package [jsk_pcl_ros] nor is [jsk_pcl_ros] a launch file name

pratikshaandhare, 2018-08-03:
Hi,

I am calibrating the primesense caramine 1.09.  I am new to Ros. I had installed Ros kinetic 1.12.13 on Ubuntu 16.04. I am following the tutorial on https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/doc/jsk_pcl_ros/calibration.md.  I had done the monocular camera calibration and saved the camera matrix in .yml file. 
For depth calibration I had run the command `rscore` on Terminal1 and `roslaunch jsk_pcl_ros openni2_local.launch` on the Terminal2 but getting the error `[openni2_local.launch] is neither a launch file in package [jsk_pcl_ros] nor is [jsk_pcl_ros] a launch file name The traceback for the exception was written to the log file`. Could you please tell me what should I install and at what location? How to install jsk_pcl_ros openni2_local.launch?

Thanks in advance.

iory, 2018-08-03:
Did you install ros-kinetic-jsk-pcl-ros?

pratikshaandhare, 2018-08-03:
Thanks a lot for reply. Yes I installed it and it works fine now.

pratikshaandhare, 2018-08-03:
Hi,

I had run `rosrun rviz rviz` on the terminal. RViz* window appears but where can I get the pose and point cloud of live streaming of primesense?

pratikshaandhare, 2018-08-03:
Hi,
I had run the command `rosrun jsk_pcl_ros depth_error_calibration.py --model quadratic-uv-quadratic-abs` but I am not getting any window of live streaming primesense caramine in order to move checker board for depth calibration.  I got the window of Figure1 with X and Y axes and nothing is plotted there.

pratikshaandhare, 2018-08-04:
Yes, I had launched `roslaunch jsk_pcl_ros openni2_local.launch` and `roslaunch jsk_pcl_ros openni2_remote.launch`. Screenshot of Rviz and window after running the command `rosrun jsk_pcl_ros depth_error_calibration.py --model quadratic-uv-quadratic-abs` is attched here.

[image]

[image]

iory, 2018-08-04:
When you launchd ```roslaunch jsk_pcl_ros depth_error.launch```, you can see the checkerboard detector's window.
You can see it?

pratikshaandhare, 2018-08-04:
No, no. I am getting error`ERROR: cannot launch node of type [checkerboard_detector/checkerboard_detector]: checkerboard_detector
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/opt/ros/kinetic/share
ERROR: cannot launch node of type [checkerboard_detector/objectdetection_tf_publisher.py]: checkerboard_detector
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/opt/ros/kinetic/share
`
Here is the screen shot of terminal

[image]

iory, 2018-08-04:
Run following command.
```
sudo apt install ros-kinetic-checkerboard-detector
```

and, carefully follow this steps
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/doc/jsk_pcl_ros/calibration.md#depth-calibrationavailable-only-in-jsk_pcl_ros-package

pratikshaandhare, 2018-08-04:
@iory Thanks a lot. I will follow it.

iory, 2018-08-05:
@pratikshaandhare 
If you solved this issue, please close this.

closed 2018-08-06 by pratikshaandhare