jsk-ros-pkg/jsk_recognition #2348: [jsk_perception/people_pose_estimation2d.py] Question about executing sample

kochigami, 2018-10-23:
I'd like to use `people_pose_estimation2d.py`, but I'm a super beginner and need someone's help :)

I followed this documentation: https://jsk-recognition.readthedocs.io/en/latest/jsk_perception/nodes/people_pose_estimation_2d.html
When I ran `roslaunch jsk_perception sample_people_pose_estimation_2d.launch GPU:=0`,
I was said to install chainer.
Is it ok to just follow these instructions?, or is there any recommendations?: http://docs.chainer.org/en/stable/install.html

My environment is ubuntu 14.04, ROS indigo.
My PC seems to have a GPU.
```
kanae@kanae-ThinkPad-T440p:~$ lspci | grep -i nvidia
02:00.0 VGA compatible controller: NVIDIA Corporation GK208M [GeForce GT 730M] (rev a1)
```
`jsk_recognition` (deb) version:
```
kanae@kanae-ThinkPad-T440p:~$ sudo apt-cache policy ros-indigo-jsk-recognition
ros-indigo-jsk-recognition:
  Installed: 1.2.4-0trusty-20181006-130827-0800
  Candidate: 1.2.4-0trusty-20181006-130827-0800
  Version table:
 *** 1.2.4-0trusty-20181006-130827-0800 0
        500 http://packages.ros.org/ros/ubuntu/ trusty/main amd64 Packages
        100 /var/lib/dpkg/status
```

Whole log is here:
```
kanae@kanae-ThinkPad-T440p:~$ roslaunch jsk_perception sample_people_pose_estimation_2d.launch GPU:=0
... logging to /home/kanae/.ros/log/be984734-d68a-11e8-97ff-28d244f02e4c/roslaunch-kanae-ThinkPad-T440p-17893.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://133.11.216.132:33102/

SUMMARY
========

PARAMETERS
 * /camera/camera_nodelet_manager/num_worker_threads: 4
 * /camera/depth_rectify_depth/interpolation: 0
 * /camera/depth_registered_rectify_depth/interpolation: 0
 * /people_mask_publisher/limb_part: RHand
 * /people_mask_publisher/person_indices: 0
 * /people_mask_publisher/queue_size: 50
 * /people_pose_estimation_2d/gpu: 0
 * /people_pose_estimation_2d/model_file: /opt/ros/indigo/s...
 * /people_pose_estimation_2d/scales: [0.38]
 * /people_pose_estimation_2d/stride: 8
 * /people_pose_estimation_2d/with_depth: True
 * /rosdistro: indigo
 * /rosversion: 1.11.21
 * /use_sim_time: True

NODES
  /camera/rgb/
    republish (image_transport/republish)
  /camera/depth_registered/
    republish (image_transport/republish)
  /
    camera_base_link (tf/static_transform_publisher)
    camera_base_link1 (tf/static_transform_publisher)
    camera_base_link2 (tf/static_transform_publisher)
    camera_base_link3 (tf/static_transform_publisher)
    people_mask_publisher (jsk_perception/people_mask_publisher.py)
    people_pose_estimation_2d (jsk_perception/people_pose_estimation_2d.py)
    people_poses_to_poses (jsk_recognition_msgs/people_pose_array_to_pose_array.py)
    rosbag_play (rosbag/play)
    rviz (rviz/rviz)
  /camera/
    camera_nodelet_manager (nodelet/nodelet)
    depth_metric (nodelet/nodelet)
    depth_metric_rect (nodelet/nodelet)
    depth_points (nodelet/nodelet)
    depth_rectify_depth (nodelet/nodelet)
    depth_registered_hw_metric_rect (nodelet/nodelet)
    depth_registered_metric (nodelet/nodelet)
    depth_registered_rectify_depth (nodelet/nodelet)
    points_xyzrgb_hw_registered (nodelet/nodelet)
    rgb_rectify_color (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[rosbag_play-1]: started with pid [17911]
process[camera/rgb/republish-2]: started with pid [17912]
process[camera/depth_registered/republish-3]: started with pid [17918]
process[camera/camera_nodelet_manager-4]: started with pid [17931]
[FATAL] [1540276076.356191611]: Error opening file: /opt/ros/indigo/share/jsk_perception/sample/data/2017-06-20-12-00-00_people_images_in_lab.bag
process[camera/rgb_rectify_color-5]: started with pid [17948]
process[camera/depth_rectify_depth-6]: started with pid [17955]
process[camera/depth_metric_rect-7]: started with pid [17965]
process[camera/depth_metric-8]: started with pid [17979]
process[camera/depth_points-9]: started with pid [17997]
process[camera/depth_registered_rectify_depth-10]: started with pid [18014]
process[camera/points_xyzrgb_hw_registered-11]: started with pid [18032]
process[camera/depth_registered_hw_metric_rect-12]: started with pid [18041]
process[camera/depth_registered_metric-13]: started with pid [18056]
process[camera_base_link-14]: started with pid [18068]
process[camera_base_link1-15]: started with pid [18085]
process[camera_base_link2-16]: started with pid [18104]
process[camera_base_link3-17]: started with pid [18119]
process[people_pose_estimation_2d-18]: started with pid [18145]
process[people_mask_publisher-19]: started with pid [18170]
ERROR: cannot launch node of type [jsk_recognition_msgs/people_pose_array_to_pose_array.py]: can't locate node [people_pose_array_to_pose_array.py] in package [jsk_recognition_msgs]
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/jsk_perception/people_pose_estimation_2d.py", line 9, in <module>
    import chainer
ImportError: No module named chainer
process[rviz-21]: started with pid [18183]
[people_pose_estimation_2d-18] process has died [pid 18145, exit code 1, cmd /opt/ros/indigo/lib/jsk_perception/people_pose_estimation_2d.py ~input:=/camera/rgb/image_rect_color ~input/info:=/camera/rgb/camera_info ~input/depth:=/camera/depth_registered/hw_registered/image_rect_raw __name:=people_pose_estimation_2d __log:=/home/kanae/.ros/log/be984734-d68a-11e8-97ff-28d244f02e4c/people_pose_estimation_2d-18.log].
log file: /home/kanae/.ros/log/be984734-d68a-11e8-97ff-28d244f02e4c/people_pose_estimation_2d-18*.log
[ INFO] [1540276076.524468453]: Initializing nodelet with 4 worker threads.
[rosbag_play-1] process has died [pid 17911, exit code 1, cmd /opt/ros/indigo/lib/rosbag/play /opt/ros/indigo/share/jsk_perception/sample/data/2017-06-20-12-00-00_people_images_in_lab.bag --clock --loop __name:=rosbag_play __log:=/home/kanae/.ros/log/be984734-d68a-11e8-97ff-28d244f02e4c/rosbag_play-1.log].
log file: /home/kanae/.ros/log/be984734-d68a-11e8-97ff-28d244f02e4c/rosbag_play-1*.log
[rviz-21] process has finished cleanly
log file: /home/kanae/.ros/log/be984734-d68a-11e8-97ff-28d244f02e4c/rviz-21*.log
^C[people_mask_publisher-19] killing on exit
[camera_base_link3-17] killing on exit
[camera_base_link2-16] killing on exit
[camera_base_link1-15] killing on exit
[camera_base_link-14] killing on exit
[camera/depth_registered_metric-13] killing on exit
[camera/depth_registered_hw_metric_rect-12] killing on exit
[camera/depth_registered_rectify_depth-10] killing on exit
[camera/points_xyzrgb_hw_registered-11] killing on exit
[camera/depth_points-9] killing on exit
[camera/depth_metric-8] killing on exit
[camera/depth_metric_rect-7] killing on exit
[camera/depth_rectify_depth-6] killing on exit
[camera/rgb_rectify_color-5] killing on exit
[camera/camera_nodelet_manager-4] killing on exit
[camera/depth_registered/republish-3] killing on exit
[camera/rgb/republish-2] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

YutoUchimi, 2018-10-24:
@kochigami 
- Probably
    ```
    ERROR: cannot launch node of type [jsk_recognition_msgs/people_pose_array_to_pose_array.py]: can't locate node [people_pose_array_to_pose_array.py] in package [jsk_recognition_msgs]
    ```
    appears because `people_pose_array_to_pose_array.py` installed via apt is not executable.
    I'm trying to fix this in https://github.com/jsk-ros-pkg/jsk_recognition/pull/2345
    For temporary fix, please make this script executable by
    ```
    sudo chmod +x /opt/ros/indigo/lib/jsk_recognition_msgs/scripts/*
    ```

- Maybe
    ```
    [FATAL] [1540276076.356191611]: Error opening file: /opt/ros/indigo/share/jsk_perception/sample/data/2017-06-20-12-00-00_people_images_in_lab.bag
    ```
    appears because
    ```
    rosrun jsk_perception install_sample_data.py
    ```
    was not executed while installing or before launching this sample.
    So please install this rosbag file manually.

kochigami, 2018-11-03:
just a memo 

I didn't succeed in installing cupy.
I don't know how to search for the best version.

--- old memo
However, I have trouble in executing sample...
When I executed `roslaunch jsk_perception sample_people_pose_estimation_2d.launch GPU:=
0`, rviz display was launched, but no image (rosbag data) was shown.
I got `OSError: model does not exists: "/home/leus/jsk_recognition_ws/src/jsk_recognition/jsk_perception/trained_data/pose_estimation_2d_chainermodel.pkl"`

What I did:
I installed CUDA, cuDNN, chainer and cupy following the expert's instruction. 
(I'll write a documentation later for a super beginner like me!)

CUDA 10.0
```
sudo dpkg --list | grep cuDNN
ii  libcudnn7                                    7.3.1.20-1+cuda10.0                                   amd64        cuDNN runtime libraries
ii  libcudnn7-dev                                7.3.1.20-1+cuda10.0                                   amd64        cuDNN development libraries and headers
```
```
leus@leus-ThinkStation-P320-Tiny:~$ python -c 'import chainer; chainer.print_runtime_info()'
Platform: Linux-4.15.0-38-generic-x86_64-with-Ubuntu-16.04-xenial
Chainer: 5.0.0
NumPy: 1.15.3
CuPy: Not Available
iDeep: Not Available
```

- Then, I cloned `jsk_recognition` from source.
```
leus@leus-ThinkStation-P320-Tiny:~/jsk_recognition_ws$ catkin build
------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/kinetic
Workspace:                   /home/leus/jsk_recognition_ws
------------------------------------------------------------------
Source Space:       [exists] /home/leus/jsk_recognition_ws/src
Log Space:          [exists] /home/leus/jsk_recognition_ws/logs
Build Space:        [exists] /home/leus/jsk_recognition_ws/build
Devel Space:        [exists] /home/leus/jsk_recognition_ws/devel
Install Space:      [unused] /home/leus/jsk_recognition_ws/install
DESTDIR:            [unused] None
------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
------------------------------------------------------------------
Workspace configuration appears valid.
------------------------------------------------------------------
[build] Found '9' packages in 0.0 seconds.                                     
[build] Package table is up to date.                                           
Starting  >>> jsk_recognition_msgs                                             
Finished  <<< jsk_recognition_msgs                   [ 1.2 seconds ]           
Starting  >>> checkerboard_detector                                            
Starting  >>> jsk_recognition_utils                                            
Finished  <<< checkerboard_detector                  [ 0.1 seconds ]           
Finished  <<< jsk_recognition_utils                  [ 0.2 seconds ]           
Starting  >>> imagesift                                                        
Starting  >>> jsk_pcl_ros_utils                                                
Starting  >>> resized_image_transport                                          
[... 238 lines cut ...]
process[people_mask_publisher-20]: started with pid [9759]
process[people_poses_to_poses-21]: started with pid [9775]
process[rviz-22]: started with pid [9780]
[ INFO] [1541214732.301330460]: Initializing nodelet with 4 worker threads.
[rosbag_play-2] process has died [pid 9547, exit code 1, cmd /opt/ros/kinetic/lib/rosbag/play /home/leus/jsk_recognition_ws/src/jsk_recognition/jsk_perception/sample/data/2017-06-20-12-00-00_people_images_in_lab.bag --clock --loop __name:=rosbag_play __log:=/home/leus/.ros/log/414d8a20-df16-11e8-ab43-6c4b9021cca5/rosbag_play-2.log].
log file: /home/leus/.ros/log/414d8a20-df16-11e8-ab43-6c4b9021cca5/rosbag_play-2*.log
/usr/lib/python2.7/dist-packages/matplotlib/font_manager.py:273: UserWarning: Matplotlib is building the font cache using fc-list. This may take a moment.
  warnings.warn('Matplotlib is building the font cache using fc-list. This may take a moment.')
Traceback (most recent call last):
  File "/home/leus/jsk_recognition_ws/src/jsk_recognition/jsk_perception/node_scripts/people_pose_estimation_2d.py", line 807, in <module>
    PeoplePoseEstimation2D()
  File "/home/leus/jsk_recognition_ws/src/jsk_common/jsk_topic_tools/src/jsk_topic_tools/transport.py", line 26, in __call__
    obj = type.__call__(cls, *args, **kwargs)
  File "/home/leus/jsk_recognition_ws/src/jsk_recognition/jsk_perception/node_scripts/people_pose_estimation_2d.py", line 132, in __init__
    self._load_model()
  File "/home/leus/jsk_recognition_ws/src/jsk_recognition/jsk_perception/node_scripts/people_pose_estimation_2d.py", line 152, in _load_model
    self._load_chainer_model()
  File "/home/leus/jsk_recognition_ws/src/jsk_recognition/jsk_perception/node_scripts/people_pose_estimation_2d.py", line 158, in _load_chainer_model
    self.pose_net = PoseNet(pretrained_model=model_file)
[... cut ...]
[people_pose_estimation_2d-19] process has died [pid 9746, exit code 1, cmd /home/leus/jsk_recognition_ws/src/jsk_recognition/jsk_perception/node_scripts/people_pose_estimation_2d.py ~input:=/camera/rgb/image_rect_color ~input/info:=/camera/rgb/camera_info ~input/depth:=/camera/depth_registered/hw_registered/image_rect_raw __name:=people_pose_estimation_2d __log:=/home/leus/.ros/log/414d8a20-df16-11e8-ab43-6c4b9021cca5/people_pose_estimation_2d-19.log].
log file: /home/leus/.ros/log/414d8a20-df16-11e8-ab43-6c4b9021cca5/people_pose_estimation_2d-19*.log
```

YutoUchimi, 2018-11-03:
I'm just guessing, but perhaps you forgot to `rosdep install` after cloning `jsk_common` into your catkin workspace.

kochigami, 2018-11-07:
@YutoUchimi 
Thank you for your comment!
Actually, my biggest problem might be that installing CUDA 10,0, which cupy has not supported yet :)
My PC don't have any cupy.
I'll try your suggestion after installing CUDA 9.2.

---
memo

CUDA10.0 <= cupy has not supported yet according to https://docs-cupy.chainer.org/en/stable/install.html
So, when installing CUDA, I need to specify the version: 'sudo apt-get install cuda-9-2'.

chainer: I'm interested in what happens if I install chainer5.0. (I can install this newest version according to http://docs.chainer.org/en/stable/install.html) Then, I'll try `sudo pip install chainer==4.3.0`.

* uninstall CUDA
`sudo apt-get --purge remove nvidia-*`

* uninstall cuDNN
```
sudo dpkg --list | grep cuDNNii  libcudnn7                                    7.3.1.20-1+cuda10.0                                   amd64        cuDNN runtime libraries
ii  libcudnn7-dev                                7.3.1.20-1+cuda10.0                                   amd64        cuDNN development libraries and headers
leus@leus-ThinkStation-P320-Tiny:~$ sudo dpkg -r libcudnn7
dpkg: dependency problems prevent removal of libcudnn7:
 libcudnn7-dev depends on libcudnn7 (= 7.3.1.20-1+cuda10.0).

dpkg: error processing package libcudnn7 (--remove):
 dependency problems - not removing
Errors were encountered while processing:
 libcudnn7
;; 他のパッケージに依存しているもの（末端）から消していく
leus@leus-ThinkStation-P320-Tiny:~$ sudo dpkg -r libcudnn7-dev
(Reading database ... 319682 files and directories currently installed.)
Removing libcudnn7-dev (7.3.1.20-1+cuda10.0) ...
update-alternatives: removing manually selected alternative - switching libcudnn to auto mode
leus@leus-ThinkStation-P320-Tiny:~$ sudo dpkg -r libcudnn7
(Reading database ... 319676 files and directories currently installed.)
Removing libcudnn7 (7.3.1.20-1+cuda10.0) ...
Processing triggers for libc-bin (2.23-0ubuntu10) ...
```
* uninstall chainer
`pip uninstall chainer`

kochigami, 2018-11-08:
memo
(Maybe I did wrong thing to my PC environment.  I will re-install ubuntu16.04.)
The problem is that my PC tries to fetch matplotlib-3.0.1, which produces an error.

I installed:
CUDA 9.2
cuDNN for CUDA 9.2
chainer 5.0 / 4.3
cupy: cupy-cuda92

Then, 
```
cd 
rosdep update
sudo apt update
rosdep install --from-paths --ignore-src -y -r src
```
produces the error below:
```
executing command [sudo -H pip install -U fcn]
Collecting fcn
  Using cached https://files.pythonhosted.org/packages/07/e3/647f5bb7ee60beb063ace1758836d8d66f511916b0ef89c6181201683ef5/fcn-6.4.8.tar.gz
Requirement already up-to-date: chainer>=2.0.0 in /usr/local/lib/python2.7/dist-packages (from fcn)
Requirement already up-to-date: chainercv in /usr/local/lib/python2.7/dist-packages (from fcn)
Collecting gdown (from fcn)
  Using cached https://files.pythonhosted.org/packages/12/33/e9f21d0b3f85804ca570d124fb7a80c12a99948ff495cf54dfb72f18bf9e/gdown-3.6.0.tar.gz
Collecting tqdm (from fcn)
  Using cached https://files.pythonhosted.org/packages/91/55/8cb23a97301b177e9c8e3226dba45bb454411de2cbd25746763267f226c2/tqdm-4.28.1-py2.py3-none-any.whl
Collecting numpy (from fcn)
  Using cached https://files.pythonhosted.org/packages/de/37/fe7db552f4507f379d81dcb78e58e05030a8941757b1f664517d581b5553/numpy-1.15.4-cp27-cp27mu-manylinux1_x86_64.whl
Collecting scipy (from fcn)
  Using cached https://files.pythonhosted.org/packages/2a/f3/de9c1bd16311982711209edaa8c6caa962db30ebb6a8cc6f1dcd2d3ef616/scipy-1.1.0-cp27-cp27mu-manylinux1_x86_64.whl
Collecting matplotlib (from fcn)
  Using cached https://files.pythonhosted.org/packages/62/81/e394906a8a15c46b56110c558c222d4d9b3735f0595e254918eca47f98cf/matplotlib-3.0.1.tar.gz
    Complete output from command python setup.py egg_info:
    
    Matplotlib 3.0+ does not support Python 2.x, 3.0, 3.1, 3.2, 3.3, or 3.4.
    Beginning with Matplotlib 3.0, Python 3.5 and above is required.
    
    This may be due to an out of date pip.
    
    Make sure you have pip >= 9.0.1.
    
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-u55hT8/matplotlib/
You are using pip version 8.1.1, however version 18.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
You are using pip version 9.0.1, however version 18.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
ERROR: the following rosdeps failed to install
  pip: command [sudo -H pip install -U fcn] failed
  pip: Failed to detect successful installation of [fcn]
```
When I executed `pip install pip==9.0.1`, the same log appeared. (previous pip version is 8.x, I forgot x)

python version: 2.7.12
```
python --version
Python 2.7.12
```

matplotlib version: 1.5.1
```
import matplotlib
print matplotlib.__version__
1.5.1
```

knorth55, 2018-11-08:
### CUDA 10 Uninstallation
You may fail to uninstall `cuda-10` or build `cupy` correctly.
That is because you can see `CUDA ~~: 10000` in `chainer.print_runtime_info`.

```python
  CUDA Build Version    : 9020
  CUDA Driver Version   : 10000
  CUDA Runtime Version  : 9020
```

My env is like this.

```python
In [3]: chainer.print_runtime_info()
Platform: Linux-4.4.0-128-generic-x86_64-with-Ubuntu-16.04-xenial
Chainer: 6.0.0a1
NumPy: 1.15.3
CuPy:
  CuPy Version          : 5.0.0rc1
  CUDA Root             : /usr/local/cuda
  CUDA Build Version    : 9010
  CUDA Driver Version   : 9010
  CUDA Runtime Version  : 9010
  cuDNN Build Version   : 7103
  cuDNN Version         : 7103
  NCCL Build Version    : 2115
iDeep: Not Available
```

### Matplotlib

My env is like below.
```bash
$ pip --version
pip 9.0.3 from /usr/local/lib/python2.7/dist-packages (python 2.7)
```

```python
In [6]: matplotlib.__version__
Out[6]: '2.2.3'
```

Can you try `pip install 'pip<10.0.0' -U` or `pip install matplotlib<3.0.0`?

kochigami, 2018-11-08:
@knorth55 

Thank you for your comment!
Actually, I re-installed ubuntu16.04 before reading your comment, and re-installed CUDA and others...

And finally, I succeeded in executing a sample (`roslaunch jsk_perception sample_people_pose_estimation_2d.launch GPU:=0`)!

Thank you @iory , too!

I'll write a simple documentation about required setup for using GPU for a super beginner like me.

---
memo

I installed:
```
ubuntu 16.04
ros kinetic
CUDA 9.2
cuDNN for 9.2 (cuDNN runtime/ development libraries)
sudo pip install chainer==4.3.0
sudo pip install cupy-cuda92==4.3.0
jsk_recognition from source 
```

One trouble I encountered was that I had to update pip version when executing `rosdep install --from-paths --ignore-src -y -r src` because installing `fcn` was failed.
`sudo python -m pip install pip==9.0.1`
% I shouldn't execute `pip install -U pip` because new version causes another error.

```
 chainer.print_runtime_info()
Platform: Linux-4.15.0-38-generic-x86_64-with-Ubuntu-16.04-xenial
Chainer: 5.0.0
NumPy: 1.15.4
CuPy:
  CuPy Version          : 5.0.0
  CUDA Root             : /usr/local/cuda
  CUDA Build Version    : 9020
  CUDA Driver Version   : 10000
  CUDA Runtime Version  : 9020
  cuDNN Build Version   : 7201
  cuDNN Version         : 7201
  NCCL Build Version    : 2213
iDeep: Not Available
```
```
leus@leus-ThinkStation-P320-Tiny:~$ pip --version
pip 9.0.1 from /usr/local/lib/python2.7/dist-packages (python 2.7)
```

```
In [4]: matplotlib.__version__
Out[4]: '2.2.3'
```

closed 2018-11-08 by kochigami