jsk-ros-pkg/jsk_recognition #2358: ROS Melodic release?

esteve, 2018-11-06:
Over at the https://github.com/CPFL/Autoware project, we'd like to migrate to Melodic, but unfortunately some of our dependencies haven't been released into Melodic.

Is there anything I can help with to release the `jsk_recognition` packages into Melodic? Thanks.

k-okada, 2018-11-07:
@esteve thanks for using our package.

We're working this on https://github.com/jsk-ros-pkg/jsk_recognition/pull/2355 and it said real blocker is our jsk_common packages, and that should be ready in a few days by  https://github.com/ros/rosdistro/pull/19417

```
imagesift: Cannot locate rosdep definition for [jsk_topic_tools]
resized_image_transport: Cannot locate rosdep definition for [jsk_tools]
jsk_pcl_ros: Cannot locate rosdep definition for [jsk_tools]
checkerboard_detector: Cannot locate rosdep definition for [dynamic_tf_publisher]
jsk_recognition_utils: Cannot locate rosdep definition for [jsk_tools]
jsk_perception: Cannot locate rosdep definition for [jsk_tools]
jsk_pcl_ros_utils: Cannot locate rosdep definition for [jsk_tools]
```

Due to limited resources of our lab and also most of the real robot still using indigo/kinetic including pr2, fetch, baxter, pepper, nao and so on..., The priority of Melodic migration is very low for us.
And also we do not have enough resources to support multiple branch such as `kinetic-devel` and `melodic-devel` for one package, we'd like to have one 'master' branch that covers all working ROS distros, which are currently Indigo/Kinetic and Melodic near future, we do not want to drop features/dependencies just because it is not available on Melodic. That's why I started http://wiki.ros.org/OrphanedPackage project and released necessary packages such as http://repositories.ros.org/status_page/ros_melodic_default.html?q=ORPHANED

So if you would need quick release of our jsk_* packages on Meldoic, please help us to release it's dependencies to Meldoic first, by becoming the maintainer of these packages.

Another idea is move our code into `rviz` or `topic_tools`, if you feel these features are useable. Before we started `jsk_rviz_plugins`, we contributed to `rviz` packages but we switched to `jsk_rviz_plugin` inorder to accelerate our developing speed and also accept experimental code. So I think it's time to back usable code into upstream packages.

esteve, 2018-11-07:
@k-okada thank you so much for the quick response. I can help with the orphaned packages, any in particular you'd like me to take maintenance over to start with? `ml-classifiers` seems a good candidate.

k-okada, 2018-11-07:
@esteve thanks `ml-classifiers` would be great. I also think you might be interested in `openslam_gmapping` and `teleop_twist_joy` . `gmapping` is now become BSD license so I think it worth to have new version (  https://github.com/ros-perception/openslam_gmapping/pull/20 )

http://repositories.ros.org/status_page/blocked_releases_melodic.html?q=label%3D%22UNBLOCKED_BLOCKING%22&s=6&r=1
http://repositories.ros.org/status_page/blocked_releases_lunar.html?q=label%3D%22UNBLOCKED_BLOCKING%22&s=6&r=1

christian-rauch, 2018-11-28:
@esteve I was working on replacing `tf` with `tf2_ros` for a `melodic` build in PR https://github.com/jsk-ros-pkg/jsk_recognition/pull/2335. It builds on melodic if you manually check out it's dependency `octomap_mapping` from source.
If someone would be willing to maintain/release `octomap_mapping` to melodic (https://github.com/OctoMap/octomap_mapping/issues/49), my PR would build on the build server and could be merged.

esteve, 2018-11-28:
@christian-rauch `octomap_mapping` is now marked as unmantained, so we could release it as an orphaned package. @k-okada is there anything I have to do to join the orphaned package maintainers group? REP-152.rst is not merged yet https://github.com/ros-infrastructure/rep/pull/150

esteve, 2019-01-08:
@k-okada would it be feasible to move `jsk_recognition_msgs` to its own repository, just like `jsk_common_msgs`, that way it could be released into Melodic.

We're using `jsk_recognition_msgs` in a few packages and that's the only thing that's preventing us from migrating to Melodic. Moreover, `jsk_rviz_plugins` and `jsk_recognition_utils` are still in Kinetic because they depend on `jsk_recognition_msgs`, the rest of the dependencies seem to be already in Melodic.

jinfagang, 2019-02-13:
@esteve Any updates on this thread? Would u like to opensource the updated melodic version jsk messages?

k-okada, 2019-02-13:
please look at https://travis-ci.org/jsk-ros-pkg/jsk_recognition/jobs/492475446, and propose for fix

christian-rauch, 2019-02-13:
@k-okada The error `error: no match for ?operator=? (operand types are ?boost::shared_ptr<tf::Transformer>? and ?const std::shared_ptr<tf2_ros::Buffer>?)` is fixed by https://github.com/jsk-ros-pkg/jsk_recognition/pull/2335

k-okada, 2019-02-14:
released ->  https://github.com/ros/rosdistro/pull/20308

JWhitleyAStuff, 2019-02-27:
@esteve / @k-okada - I've taken maintenance for ml-classifiers and updated it for Melodic (I wasn't aware until now that @esteve planned to do it): https://github.com/astuff/ml_classifiers. PRs for releases:

https://github.com/ros/rosdistro/pull/20386
https://github.com/ros/rosdistro/pull/20387
https://github.com/ros/rosdistro/pull/20389

esteve, 2019-02-27:
@JWhitleyAStuff no worries, thank you so much for taking care of it.

jinfagang, 2019-03-01:
@esteve @JWhitleyAStuff Still can not install jsk-recognition-msgs with this command:

```
 sudo apt install ros-melodic-jsk-recognition-msgs
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-melodic-jsk-recognition-msgs

```

JWhitleyAStuff, 2019-03-01:
@jinfagang - this is because it has been added to the rosdep index but it will not be released to the build farm until the next package sync which should be sometime next week. Watch https://discourse.ros.org/c/release/melodic for the announcement.

k-okada, 2019-03-01:
that's only available on shadow-fixed repository,
http://repositories.ros.org/status_page/ros_melodic_default.html?q=jsk_recognition_msgs
please wait for next sync, that should be done in 1-2 week.
https://discourse.ros.org/c/release/melodic
--
◉ Kei Okada

jinfagang, 2019-03-01:
Thank u all

JWhitleyAStuff, 2019-03-13:
All packages released in [melodic sync](https://discourse.ros.org/t/new-packages-for-melodic-2019-03-04/8120).

closed 2019-04-10 by YutoUchimi

3038922, 2019-10-02:
my system is ubuntu1804 
ros --version is melodic .
if i input `rosdep install -i -y --from-paths ./naoqi_driver`
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
naoqi_driver: Cannot locate rosdep definition for [naoqi_libqicore]
```