jsk-ros-pkg/jsk_recognition #2359: Missing dependencies for Kinetic

esteve, 2018-11-06:
This commit (https://github.com/jsk-ros-pkg/jsk_recognition/commit/63538fda553e58a48ec39936afe4c04a8cf70504) readded `ml_classifiers`, which is not available in Kinetic, so `jsk_pcl_ros` 1.2.6 can't be released into Kinetic (or later)

JWhitleyAStuff, 2019-02-22:
@k-okada and @esteve - I am updating ml_classifiers. I have reached out to the maintainer via email. Until that time, I've created a fork with a Melodic-compatible version at astuff/ml_classifiers.

JWhitleyAStuff, 2019-02-24:
@k-okada - Can you please verify that the version which is now at https://github.com/astuff/ml_classifiers works correctly in your implementation? Once it's been tested, I'll create a new release for Kinetic, Lunar, and Melodic.

k-okada, 2019-02-26:
@JWhitleyAStuff thanks for you support to ml_classifiers, I have confirmed that your package is ok for us.

JWhitleyAStuff, 2019-02-26:
https://github.com/ros/rosdistro/pull/20386 - Kinetic
https://github.com/ros/rosdistro/pull/20387 - Lunar
https://github.com/ros/rosdistro/pull/20389 - Melodic

JWhitleyAStuff, 2019-03-13:
Released in all 3 ROS versions. This issue can probably be closed but you might want to change ml_classifiers to a ROS dependency instead of a subrepo for jsk_recognition now. :)

closed 2019-04-10 by YutoUchimi