jsk-ros-pkg/jsk_recognition #2360: How to I install this package?

yuqli, 2018-11-06:
` sudo apt-get install ros-kinetic-jsk-recognition-msg` is not working so I guess the only way is to build from source? I'm very new to ROS, any comments / confirmation will be helpful. Thanks!

Also, will using catkin_make and catkin build be a difference? The latter command belongs to a set of tools in the catkin command line tools https://www.google.com/search?client=ubuntu&channel=fs&q=catkin+command+line+tools&ie=utf-8&oe=utf-8

k-okada, 2018-11-07:
`ros-kinetic-jsk-recognition-msgs` should be installable from both public and shadow-fixed repositories. Please check again.
http://repositories.ros.org/status_page/ros_kinetic_default.html?q=jsk_recognition_msgs

yuqli, 2018-11-07:
Thanks for the reply. I figured it out: I should use hyphen instead of
underscore. Use jsk-recognition works fine now.

YutoUchimi, 2018-12-26:
@yuqli If you have solved your problem, please close this issue.

closed 2019-03-01 by knorth55