jsk-ros-pkg/jsk_recognition #2366: How to install and use OrganizedEdgeDetector

km60713, 2018-11-13:
I'd like to say thank you for the wonderful packages to JSK and now I'd like to use OrganizedEdgeDetector. However, I'm stucked in installation and need a help.
I installed jsk_recognition by `$ sudo apt-get install ros-kinetic-jsk-recognition`.
Once I tried to run it with the sample launch file which was included in the package, there was some error as below. It says that OrganizedEdgeDetector is not existing.
> [ERROR] [1542059871.979218954, 1541865071.258148383]: Failed to load nodelet [/edge_detector] of type [jsk_pcl/OrganizedEdgeDetector] even after refreshing the cache: MultiLibraryClassLoader: Could not create object of class type jsk_pcl_ros::OrganizedEdgeDetector as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()

After that, I tried to install jsk-pcl-ros from the source code. The build went through good but the OrganizedEdgeDetector does not seem to exist yet.

How can I install and use OrganizedEdgeDetector?

OS: Ubuntu 16.04LTS
ROS: kinetic
PCL: 1.7(installed by `$ sudo apt install libpcl-dev`. I believe it is 1.7.2)

Thank you in advance.

iory, 2018-11-13:
Thank you for using jsk_recognition and your issue report!

If you would like to use ```OrganizedEdgeDetector```, you have to install ```pcl-1.8```.
Can you try following scripts?

```
sudo apt update && sudo apt install -y \
                        libboost-all-dev \
                        libeigen3-dev \
                        libflann-dev \
                        libqhull-dev \
                        libvtk5-dev

cd /tmp && \
    wget https://github.com/PointCloudLibrary/pcl/archive/pcl-1.8.0rc2.tar.gz && \
    tar zxf pcl-1.8.0rc2.tar.gz

cd /tmp/pcl-pcl-1.8.0rc2 && \
    mkdir build && \
    cd build && \
    cmake .. -DCMAKE_BUILD_TYPE=Release && \
    make -j2 && \
    sudo make install

mkdir -p ~/ros/ws_jsk_recognition/src && \
    cd ~/ros/ws_jsk_recognition/src && \
    apt-get install -y python-rosinstall-generator && \
    wstool init && \
    wstool set --git jsk-ros-pkg/jsk_recognition https://github.com/jsk_recognition -y
    rosinstall_generator --tar --rosdistro kinetic \
                         pcl_conversions \
                         pcl_ros \
                         octomap_server \
                         > $$.rosinstall && \
        wstool merge -y $$.rosinstall && \
        wstool up -j -1

cd ~/ros/ws_jsk_recognition/src && \
    rosdep update && \
    rosdep install --from-paths --ignore-src -y -r .

cd ~/ros/ws_jsk_recognition && \
    source /opt/ros/kinetic/setup.sh && \
    catkin build
```

km60713, 2018-11-13:
Thank you for the scripts.
Finally, The installation of `pcl-1.8` seems to be successful.
On the other hand, during this scrips, huge amount(more than 50) of ros packages have removed from my computer. For example, ros-kinetic-pcl-ros, ros-kinetic-opencv-apps, ros-kinetic-image-view and more. Even ros-kinetic-jsk-recognition or ros-kinetic-jsk-pcl-ros have removed. I found a directry which name is jsk-ros-pkg in ~/ros/ws_jsk_recognition . However, it was empty.
I'll try to re-install all these packages from the repository but I have a question. Is it good for installing jsk packages from the repository or do I need to build from the source?

iory, 2018-11-13:
>  I found a directry which name is jsk-ros-pkg in ~/ros/ws_jsk_recognition . However, it was empty.

Please let me know the following command results.
```
ls -althr ~/ros/ws_jsk_recognition/src/
```

iory, 2018-11-14:
> For my understanding, all I need to do is as below. Is it correct?

Please try it.

km60713, 2018-11-14:
Unfortunately, catkin build failed by some error. It was during pcl_ros:make.
I attached the log file in below.
[build.make.000.log](https://github.com/jsk-ros-pkg/jsk_recognition/files/2578818/build.make.000.log)
Sorry for bothering you but could you give me a little more advice?

iory, 2018-11-14:
In kinetic, there are some errors. So I'll fix it.
However, if you can use docker, you can try ```OrganizedEdgeDetector``` easily.

```
docker run --rm --net=host -it jskrobotics/ros-indigo-jsk-recognition-with-pcl-1.8
```

In bash on docker, please source jsk_recognition workspace.
```
source ~/ros/ws_jsk_recognition/devel/setup.bash
```

After that you can try jsk_recognition's nodes.

km60713, 2018-11-14:
Thank you for introducing another solution. I don't know much about docker but I'll try it.
I'm little worried that many other packages are running on ros-kinetic at non-docker environment. So, it may not receive the result of the OrganizedEdgeDetector.

iory, 2018-11-14:
install manual is here.
https://docs.docker.com/install/linux/docker-ce/ubuntu/

> I'm little worried that many other packages are running on ros-kinetic at non-docker environment. So, it may not receive the result of the OrganizedEdgeDetector.

you can use a same ip address on docker instance. so you can  receive the result of the OrganizedEdgeDetector by using ROS like as a another PC in same network.

km60713, 2018-11-15:
Thank you for your kind support. I'll try to run it in docker.
Also, looking forward to the source code which can build in kinetic.

iory, 2018-12-25:
@km60713 

I confirmed jsk_recognition build with pcll1.8 in kinetic.
Please check following steps.

```
sudo apt update && sudo apt install -y \
                       wget \
                       libboost-all-dev \
                       libeigen3-dev \
                       libflann-dev \
                       libqhull-dev \
                       libvtk5-dev \
                       python-catkin-tools \
                       python-rosinstall-generator \
                       python-wstool

# VTK install
cd ~ && \
    wget -q http://www.vtk.org/files/release/7.1/VTK-7.1.0.tar.gz && \
    tar -xf VTK-7.1.0.tar.gz && \
    cd ~/VTK-7.1.0 && mkdir build && cd build && \
    cmake -DCMAKE_BUILD_TYPE=Release .. && \
    make -j2 && \
    sudo make install

# PCL install
cd ~ && \
    wget -q https://github.com/PointCloudLibrary/pcl/archive/pcl-1.8.0rc2.tar.gz && \
    tar zxf pcl-1.8.0rc2.tar.gz && \
    cd ~/pcl-pcl-1.8.0rc2 && \
    mkdir build && \
    cd build && \
    cmake -DCMAKE_BUILD_TYPE=Release -DPCL_ENABLE_SSE:BOOL=FALSE .. && \
    make -j2 && \
    sudo make install

mkdir -p ~/ros/ws_jsk_recognition/src && \
    cd ~/ros/ws_jsk_recognition/src && \
    rosinstall_generator --tar --rosdistro ${ROS_DISTRO} \
        pcl_conversions \
        pcl_ros \
        octomap_server \
        > .rosinstall && \
    wstool up -j -1

cd ~/ros/ws_jsk_recognition/src && \
    git clone --depth 1 https://github.com/jsk-ros-pkg/jsk_recognition && \
    cd ~/ros/ws_jsk_recognition && \
    rosdep update && \
    sudo rosdep install --from-paths --ignore-src -y -r src && \
    catkin build
```