jsk-ros-pkg/jsk_recognition #2368: DepthImageCreator does not publish organized pointcloud

yuki-shark, 2018-11-16:
I tried to use DepthImageCreator. Although [the external document](https://jsk-recognition.readthedocs.io/en/latest/jsk_pcl_ros/nodes/depth_image_creator.html) says DepthImageCreator publishes `~output_cloud` as organized pointcloud, it publishes not organized but unorganized pointcloud.
Is this right node behavior? The other topics are successfully published.

I also tried to make organized pointcloud from `~output` and `~output_image` published by DepthImageCreator using [point_cloud_xyzrgb](http://wiki.ros.org/depth_image_proc). However, it does not work (not publish any topics).

I do not know whether these bugs are from my faults or not.

iory, 2018-11-16:
Thanks for your issue report!

Can you take a rosbag and upload it for debug?

yuki-shark, 2018-11-16:
Sure! Thank you for your support.
I uploaded the bag file on [my drive](https://drive.google.com/file/d/1XR4sjEo124nB6bfAUdZR-hYeOJL4Be3R/view?usp=sharing).
[Here](https://drive.google.com/open?id=1nq_H3cp5t91Y79-4aw_WAehf-V--xR4a) is rviz file for your debug.

*Description*
`~input` : `/concatenate_data/output` (unorganized)
`~info` :  `/static_virtual_camera/camera_info`

Although here I am using a static virtual camera, `~output_cloud` was still unorganized when I used camera info published by openni2. I also tried to use organized pointcloud from raw data as `~input` instead of unorganized pointcloud, the result was same.

iory, 2018-11-30:
memo 

we can create organized point cloud from depth_image_proc
```
  <group ns="depth_image_creator" >
   <node name="points_xyzrgb_sw_registered"
         pkg="nodelet" type="nodelet"
         args="standalone depth_image_proc/point_cloud_xyzrgb"
         respawn="true" >
     <!-- Explicit topic remappings, shouldn't need all of these -->
     <remap from="rgb/image_rect_color"        to="output_image" />
     <remap from="rgb/camera_info"             to="/right_hand_left_camera/rgb/camera_info" />
     <remap from="depth_registered/image_rect" to="output" />
     <remap from="depth_registered/points"     to="output/points" />
   </node>
 </group>
```

YutoUchimi, 2019-08-20:
@yuki-shark I fixed this problem in #2446 . Please check it.

yuki-shark, 2019-08-21:
@YutoUchimi Thanks a lot for making PR instead of me.