jsk-ros-pkg/jsk_recognition #2387: Failed to load resized_image_transport/ImageResizer

arwtyxouymz, 2019-01-16:
Hi
I followed [this calibration instruction](https://github.com/jsk-ros-pkg/jsk_recognition/blob/1.2.6/doc/jsk_pcl_ros/calibration.md).
At execution of `roslaunch jsk_pcl_ros openni2_remote.launch`, I faced this error.

```
process[camera_remote_uncalibrated/depth_registered_hw_metric_rect-9]: [ERROR] [1547647604.027012086]: Failed to load nodelet [/camera_remote/rgb_half] of type [resized_image_transport/ImageResizer] even after refreshing the cache: Could not find library corresponding to plugin resized_image_transport/ImageResizer. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1547647604.027058084]: The error before refreshing the cache was: Could not find library corresponding to plugin resized_image_transport/ImageResizer. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[FATAL] [1547647604.027208867]: Failed to load nodelet '/camera_remote/rgb_half` of type `resized_image_transport/ImageResizer` to manager `camera_remote_nodelet_manager'
[camera_remote/rgb_half-23] process has died [pid 191, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load resized_image_transport/ImageResizer camera_remote_nodelet_manager ~input/image:=rgb/image_rect_color ~output/image:=~image_rect_color __name:=rgb_half __log:=/root/.ros/log/fbadf928-198f-11e9-bffa-3c6aa7f49241/camera_remote-rgb_half-23.log].
log file: /root/.ros/log/fbadf928-198f-11e9-bffa-3c6aa7f49241/camera_remote-rgb_half-23*.log
[ERROR] [1547647604.397675250]: Failed to load nodelet [/camera_remote/rgb_quater] of type [resized_image_transport/ImageResizer] even after refreshing the cache: Could not find library corresponding to plugin resized_image_transport/ImageResizer. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1547647604.397713937]: The error before refreshing the cache was: Could not find library corresponding to plugin resized_image_transport/ImageResizer. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[FATAL] [1547647604.397898802]: Failed to load nodelet '/camera_remote/rgb_quater` of type `resized_image_transport/ImageResizer` to manager `camera_remote_nodelet_manager'
[camera_remote/rgb_quater-24] process has died [pid 198, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load resized_image_transport/ImageResizer camera_remote_nodelet_manager ~input/image:=rgb/image_rect_color ~output/image:=~image_rect_color __name:=rgb_quater __log:=/root/.ros/log/fbadf928-198f-11e9-bffa-3c6aa7f49241/camera_remote-rgb_quater-24.log].
```

Probably nodelet cannot find `resized_image_transport/ImageResizer`.
How can i fix it?

I tried this in two ways, local kinetic and docker indigo container.
But in both cases same error occurred.

My local configuration:
- Ubuntu 16.04.5 LTS
- ROS Kinetic

My docker configuration:
```dockerfile
FROM osrf/ros:indigo-desktop-full

RUN apt-get update \
    && apt-get install -y --no-install-recommends \
    ros-indigo-jsk-recognition \
    && rm -rf /var/lib/apt/lists/*

RUN echo "source /opt/ros/indigo/setup.bash" >> /root/.bashrc
```

k-okada, 2019-01-17:
can you find `resized_image_transport` plugins?
```
$ rospack plugins --attrib=plugin  nodelet | grep resized_image_transport
resized_image_transport /opt/ros/kinetic/share/resized_image_transport/nodelet.xml
```
if not please, run `apt-get install ros-kinetic-resized-image-transport`.

knorth55, 2019-01-17:
I got into this error with `deb` install.
It does not happen when I install and build from source.

k-okada, 2019-01-17:
did you run apt-get dist-upgrade?
--
◉ Kei Okada

knorth55, 2019-01-17:
No, i didn’t
Should i try that?

knorth55, 2019-01-25:
I tried `apt dist-upgrade` but it does not help.

k-okada, 2019-01-25:
what is the output of

rospack plugins --attrib=plugin  nodelet | grep resized_image_transport

knorth55, 2019-01-28:
```bash
$ rospack plugins --attrib=plugin nodelet | grep resized_image_transport 
resized_image_transport /opt/ros/kinetic/share/resized_image_transport/nodelet.xml
```

knorth55, 2019-01-29:
I fix this problem in #2390