jsk-ros-pkg/jsk_recognition #2420: rosbag files are not decompressed automatically

sktometometo, 2019-04-09:
Rosbag files automatically downloaded to jsk_pcl_ros/sample/data/ are not decompressed.
```
$ rosbag info vision.compressed.bag
path:         vision.compressed.bag
version:      2.0
duration:     12.4s
start:        Jul 06 2016 12:16:43.76 (1467775003.76)
end:          Jul 06 2016 12:16:56.16 (1467775016.16)
size:         360.4 MB
messages:     3213
compression:  bz2 [655/655 chunks; 34.08%]
uncompressed:   1.0 GB @ 85.2 MB/s
compressed:   360.2 MB @ 29.0 MB/s (34.08%)
types:        sensor_msgs/CameraInfo  [c9a58c1b0b154e0e6da7578cb991d214]
              sensor_msgs/Image       [060021388200f6f0f447d0fcd9c64743]
              sensor_msgs/PointCloud2 [1158d486dd51d683ce2f1be655c3c181]
              tf2_msgs/TFMessage      [94810edda583a504dfda3829e70d7eec]
topics:       /camera_remote/depth_registered/points    327 msgs    : sensor_msgs/PointCloud2
              /camera_remote/rgb/camera_info            327 msgs    : sensor_msgs/CameraInfo 
              /camera_remote/rgb/image_rect_color       327 msgs    : sensor_msgs/Image      
              /tf                                      2232 msgs    : tf2_msgs/TFMessage
```

YutoUchimi, 2019-04-09:
from `jsk_pcl_ros/scripts/install_sample_data.py`:
```
    download_data(
        path='sample/data/2016-07-06-12-16-43-person-in-point-cloud.tgz',
        url='https://drive.google.com/uc?id=0B_NiLAzvehC9cHhnOFd0YUN2N1U',
        md5='c59067adc429fb9f1cf180d350a2da43',
        extract=True,
        compressed_bags=[
            'sample/data/2016-07-06-12-16-43-person-in-point-cloud/vision.compressed.bag',  # NOQA
        ],
    )
```

In this case, `extract=True` correctly works because corresponding directory exists.
```
$ ls -l jsk_pcl_ros/sample/data/2016-07-06-12-16-43-person-in-point-cloud/
-rwxrwxrwx 1 sktometometo sktometometo 377952961  7月 21  2016 vision.compressed.bag
-rwxrwxrwx 1 sktometometo sktometometo 377952961  7月 21  2016 vision.compressed.orig.bag
```

But `compressed_bags=['sample/data/2016-07-06-12-16-43-person-in-point-cloud/vision.compressed.bag']` seems not working correctly.

i.e. Although `vision.compressed.bag` and `vision.compressed.orig.bag`, which should be a decompressed bag and a backup compressed bag respectively, were created correctly, but `vision.compressed.bag` wasn't decompressed.

YutoUchimi, 2019-04-10:
This problem happens when you run `install_sample_data.py` or `catkin build jsk_pcl_ros` 2 times or more.
After the first `catkin build`, `vision.compressed.bag` will be decompressed correctly.
But on the next build, `2016-07-06-12-16-43-person-in-point-cloud.tgz` is extracted again and decompressed `vision.compressed.bag` will be overwritten by compressed `vision.compressed.bag`.

iory, 2019-04-10:
That's right.

YutoUchimi, 2019-04-25:
@sktometometo
I think this issue is fixed by https://github.com/jsk-ros-pkg/jsk_common/pull/1626.
Please check it.

closed 2019-07-29 by sktometometo