jsk-ros-pkg/jsk_recognition #2424: failed to try fcn_object_segmentation

tongtybj, 2019-04-11:
@knorth55 

I am trying the [this example]( https://jsk-docs.readthedocs.io/en/latest/jsk_recognition/doc/jsk_perception/nodes/fcn_object_segmentation.html), but failed with following error.

Did I miss something?

```
chou@desktop:/opt/ros/kinetic/share/jsk_perception/node_scripts$ roslaunch jsk_perception sample_fcn_object_segmentation.launch
... logging to /home/chou/.ros/log/d5a0d526-5c8c-11e9-9167-107b44b0f539/roslaunch-desktop-4425.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://desktop:46835/

SUMMARY
========

PARAMETERS
 * /fcn_object_segmentation/gpu: 0
 * /fcn_object_segmentation/model_file: /opt/ros/kinetic/...
 * /fcn_object_segmentation/model_name: fcn8s
 * /fcn_object_segmentation/target_names: ['background', 'a...
 * /fcn_object_segmentation/use_mask: False
 * /image_publisher/file_name: /opt/ros/kinetic/...
 * /image_publisher/publish_info: False
 * /label_image_decomposer/queue_size: 100
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    fcn_object_segmentation (jsk_perception/fcn_object_segmentation.py)
    image_publisher (jsk_perception/image_publisher.py)
    image_view (image_view/image_view)
    label_image_decomposer (jsk_perception/label_image_decomposer.py)

auto-starting new master
process[master]: started with pid [4435]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to d5a0d526-5c8c-11e9-9167-107b44b0f539
process[rosout-1]: started with pid [4448]
started core service [/rosout]
process[image_publisher-2]: started with pid [4459]
process[fcn_object_segmentation-3]: started with pid [4466]
process[label_image_decomposer-4]: started with pid [4467]
process[image_view-5]: started with pid [4468]
[INFO] [1555009558.255891]: Loading trained model: /opt/ros/kinetic/share/jsk_perception/trained_data/fcn8s_voc.npz
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/jsk_perception/node_scripts/fcn_object_segmentation.py", line 216, in <module>
    FCNObjectSegmentation()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/jsk_topic_tools/transport.py", line 26, in __call__
    obj = type.__call__(cls, *args, **kwargs)
  File "/opt/ros/kinetic/share/jsk_perception/node_scripts/fcn_object_segmentation.py", line 41, in __init__
    self._load_model()
  File "/opt/ros/kinetic/share/jsk_perception/node_scripts/fcn_object_segmentation.py", line 48, in _load_model
    self._load_chainer_model()
  File "/opt/ros/kinetic/share/jsk_perception/node_scripts/fcn_object_segmentation.py", line 78, in _load_chainer_model
    S.load_npz(model_file, self.model)
  File "/usr/local/lib/python2.7/dist-packages/chainer/serializers/npz.py", line 187, in load_npz
    with numpy.load(file) as f:
  File "/usr/lib/python2.7/dist-packages/numpy/lib/npyio.py", line 362, in load
    fid = open(file, "rb")
IOError: [Errno 2] No such file or directory: '/opt/ros/kinetic/share/jsk_perception/trained_data/fcn8s_voc.npz'
================================================================================REQUIRED process [fcn_object_segmentation-3] has died!
process has died [pid 4466, exit code 1, cmd /opt/ros/kinetic/share/jsk_perception/node_scripts/fcn_object_segmentation.py ~input:=image_publisher/output ~input/mask:=mask_image_generator/output __name:=fcn_object_segmentation __log:=/home/chou/.ros/log/d5a0d526-5c8c-11e9-9167-107b44b0f539/fcn_object_segmentation-3.log].
log file: /home/chou/.ros/log/d5a0d526-5c8c-11e9-9167-107b44b0f539/fcn_object_segmentation-3*.log
Initiating shutdown!
================================================================================
[image_view-5] killing on exit
[label_image_decomposer-4] killing on exit
[fcn_object_segmentation-3] killing on exit
[image_publisher-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

knorth55, 2019-04-12:
you need to install trained model first.
can you try `rosrun jsk_recognition install_trained_model.py`
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_perception/scripts/install_trained_data.py

knorth55, 2019-04-12:
@tongtybj I'm sorry, but currently we cannot use the script in debian package.
You have to build this package from source if you want to use this FCN model.
cc. @YutoUchimi 
related: https://github.com/jsk-ros-pkg/jsk_3rdparty/blob/master/3rdparty/julius/Makefile.julius

tongtybj, 2019-04-12:
@knorth55 

Thank you for your kind help! I will try to build from source. 
BTW, it is also possible to build this source in Jetson TX2, right?

knorth55, 2019-04-12:
probably,it is possible.

closed 2019-04-17 by knorth55