jsk-ros-pkg/jsk_recognition #2433: HeightmapToPointCloud does not make organized pointcloud

yuki-shark, 2019-05-30:
Is this intended behavior?

I would like to use `jsk_pcl_ros OrganizedMultiPlaneSegmentation` and input `accumulated_heightmap_pointcloud/output` generated by `jsk_pcl_ros/launch/heightmap.launch`.

Following is the process which `heightmap_converter.launch`

- current_heightmap (organized pointcloud 1600 x 1600)
v
- filtered_current_heightmap (organized pointcloud 1600 x 1600)
v
- accumulated_heightmap (Image 1600 x 1600)
v
- accumulated_heightmap_pointcloud (unorganized pointcloud)

Should I convert the last unorganized topic to organized topic to get organized accumulated heightmap pointcloud?

knorth55, 2019-05-30:
`HeightmapToPointcloud` uses `convertHeightMapToPCL`.
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/src/heightmap_to_pointcloud_nodelet.cpp#L92

`covnertHeightMapToPCL`  skips some points, so the node publish unorganized cloud.
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/include/jsk_pcl_ros/heightmap_utils.h#L52-L75

if you want have organized point cloud, you need to put `NaN` value when `v = -FLT_MAX`
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/include/jsk_pcl_ros/heightmap_utils.h#L65

and also you need to set `cloud.is_dense =false; cloud.width = image.width; cloud.height = image.height;`
http://docs.pointclouds.org/1.1.0/classpcl_1_1_point_cloud.html#a2b9aa6d655847c7dad1af27439d4663b

yuki-shark, 2019-05-30:
Thank you for your kind help!

I found `convertHeightMapToPCLOrganize` function in [heightmap_utils.h](https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/include/jsk_pcl_ros/heightmap_utils.h#L77-L103).
It looks for making organized pointcloud from heightmap image.

Does it also work to change `covnertHeightMapToPCL` to  `convertHeightMapToPCLOrganize` in 
[heightmap_to_pointcloud_nodelet.cpp](https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/src/heightmap_to_pointcloud_nodelet.cpp#L92)?

knorth55, 2019-05-30:
@yuki-shark I think it is better to keep publishing unorganized pointcloud by default and publish organized pointcloud with `rosparam` `organized:=true`.

yuki-shark, 2019-05-31:
I made PR for this extension.

closed 2019-05-31 by yuki-shark