jsk-ros-pkg/jsk_recognition #2438: How to get organized pointcloud from Euclidean Clustering?

ri-ceres, 2019-06-14:
I'd like to create a 2D pixel mask using the output of Euclidean Clustering, but the pointcloud is not organized. I've also tried using DepthImageCreator to organize the point cloud, but encountered the same result as in https://github.com/jsk-ros-pkg/jsk_recognition/issues/2368./

How can I achieve this?

knorth55, 2019-06-15:
can you give us rosbag file and launch file to reproduce your problem?
I want to know if the input point cloud is organized or not.
Making organized point cloud from unorganized point cloud would be a bit challenging.

ri-ceres, 2019-06-15:
Thanks for the reply. The point cloud is unorganized. For instance, if I run:

```
rosrun pcl_ros convert_pointcloud_to_image input:=/pcl_nodelet/cluster_decomposer/output00 output:=/mask_output
```

The message I receive is:
```
Input point cloud is not organized, ignoring!
```

I have been able to separately convert the unorganized point cloud in Python and perform the masking operation as follows:

```
points = pc2.read_points(msg_pointcloud, skip_nans=True, field_names=("x", "y", "z"))
camera_model.fromCameraInfo(intrinsics)

organized_points = []
for point in points:
    p = [point[0], point[1], point[2]]
    pp = camera_model.project3dToPixel(pc)
    organized_points.append(pp)

organized_points = np.array(organized_points, dtype=int)
```

However, I was hoping to do this using jsk rather than building a separate node.

YutoUchimi, 2019-09-20:
@ri-ceres 
> I've also tried using DepthImageCreator to organize the point cloud, but encountered the same result as in #2368

I fixed `DepthImageCreator` in #2446 , so please try it.

e.g.
unorganized point cloud and camera info -> `jsk_pcl/DepthImageCreator` -> organized point cloud -> `jsk_pcl/EuclideanClustering` -> cluster point indices -> `jsk_pcl_utils/ClusterPointIndicesToPointIndices` -> point indices -> `jsk_pcl_utils/PointIndicesToMaskImage` -> 2D pixel mask

closed 2019-10-23 by YutoUchimi