jsk-ros-pkg/jsk_recognition #2451: [jsk_perception/point_pose_extractor] setting parameter `viewer_window` returns NULL window handler

kochigami, 2019-09-24:
When I set `viewer_window` to `False`, the following error appeared:

```
OpenCV Error: Null pointer (NULL window handler) in cvSetMouseCallback, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/highgui/src/window_QT.cpp, line 700
terminate called after throwing an instance of 'cv::Exception'
  what():  /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/highgui/src/window_QT.cpp:700: error: (-27) NULL window handler in function cvSetMouseCallback
```

I use ubuntu 16.04, ROS kinetic.
My sample code is below, and I wrote `set_template.py` following [this link](https://jsk-recognition.readthedocs.io/en/latest/jsk_perception/nodes/point_pose_extractor.html#example-of-how-to-run-set-template-service).
```
<launch>
  <node name="nodelet_manager"
        pkg="nodelet" type="nodelet"
        args="manager" />
  <node name="imagesift"
        pkg="nodelet" type="nodelet"
        args="load imagesift/ImageSift nodelet_manager">
    <remap from="image" to="/pepper_robot/camera/front/image_raw"/>
    <remap from="camera_info" to="/pepper_robot/camera/front/camera_info" />
  </node>
  <node name="point_pose_extractor" pkg="jsk_perception" type="point_pose_extractor"
        respawn="true" output="screen">
    <!-- disable viewer -->
    <param name="viewer_window" value="false"/>
  </node>
  <node name="set_template" pkg="my_pkg" type="set_template.py"
        output="screen"/>
</launch>
```

pazeshun, 2019-09-24:
This may be because #2414 is not released.
You use `jsk_perception` from apt (latest version: 1.2.10), don't you?
Can you try to use it from source?

kochigami, 2019-09-24:
Thank you very much for your advice. I could also understand why this error happened.
My problem was solved with the latest `jsk_perception` from source!

closed 2019-09-24 by kochigami