jsk-ros-pkg/jsk_recognition #2474: consensus_trackingのtestについて

taichiH, 2019-11-19:
`consensus_tracking.test` をみると
https://github.com/jsk-ros-pkg/jsk_recognition/blob/81b136d28dfa6f036e55125cb3e2861093123cce/jsk_perception/test/consensus_tracking.test#L12-L15

このようになっているのですが，これは良いのでしょうか？

```
      topic_0: /consensus_tracking/output/mask
      timeout_0: 30
      topic_1: /consensus_tracking/debug/image
      timeout_1: 30
```

これはこれで問題だと思うのですが，
本題としてはこれを上記のように直して，
```
rostest jsk_perception consensus_tracking.test
```
すると，以下のようにtestが成功します．
https://gist.github.com/taichiH/377f83e4a22408b578611ec3228c7710

しかしsampleを立ち上げると
```
roslaunch jsk_perception sample_consensus_tracking.launch
```
このようにプロセスが終了してしまいます．
```
[consensus_tracking-3] process has died [pid 31885, exit code -11, cmd /home/taichi/ros/kinetic/devel/lib/jsk_perception/consensus_tracking ~input:=/camera/rgb/image_rect_color ~input/polygon:=/camera/rgb/image_rect_color/screenrectangle __name:=consensus_tracking __log:=/home/taichi/.ros/log/c498267c-0afa-11ea-8c29-d05099843c5b/consensus_tracking-3.log].
log file: /home/taichi/.ros/log/c498267c-0afa-11ea-8c29-d05099843c5b/consensus_tracking-3*.log

```

test時に生成されるログファイルを見てみると
同様にプロセスがkillされていることが確認できます．
https://gist.github.com/taichiH/66612cf76d2ce8daf5464e28b6839ce6#file-rostest-pluto-31316-log-L117

以下のconsensus_trackingのログにはなにも出力されていませんでした．\
`~/.ros/log/5c6d8fac-0af9-11ea-8c29-d05099843c5b/consensus_tracking-2-stdout.log`

あまり詳しくないので，質問したいのですが，
この場合，なぜテストが通ってしまうのでしょうか？
@YutoUchimi 

自分のPCの環境としては以下のようになっています．

PCの情報
```
Linux pluto 4.15.0-66-generic #75~16.04.1-Ubuntu SMP Tue Oct 1 14:01:08 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```

rosのバージョン
```
kinetic
```

consensus_trackingのバージョン
```
commit c291920fc7017c43e1af32a7813f4bfbee62e222
Author: Kentaro Wada <www.kentaro.wada@gmail.com>
Date:   Thu Apr 20 18:54:52 2017 +0900

    Add ~approximate_sync param to ConsensusTracking
    
    Modified:
      - doc/jsk_perception/nodes/consensus_tracking.rst
      - jsk_perception/include/jsk_perception/consensus_tracking.h
      - jsk_perception/src/consensus_tracking.cpp

```

jsk_toolsのバージョン
```
taichi@pluto:~/ros/kinetic/src/jsk-ros-pkg/jsk_recognition/jsk_perception/src$ sudo apt-cache policy ros-kinetic-jsk-tools 
ros-kinetic-jsk-tools:
  Installed: 2.2.10-0xenial-20190608-123645-0800
  Candidate: 2.2.10-0xenial-20191029-002759+0000
  Version table:
     2.2.10-0xenial-20191029-002759+0000 500
        500 http://packages.ros.org/ros/ubuntu xenial/main amd64 Packages
 *** 2.2.10-0xenial-20190608-123645-0800 100
        100 /var/lib/dpkg/status
```

YutoUchimi, 2019-11-20:
Good issue report!

> `consensus_tracking.test` をみると
https://github.com/jsk-ros-pkg/jsk_recognition/blob/81b136d28dfa6f036e55125cb3e2861093123cce/jsk_perception/test/consensus_tracking.test#L12-L15
> このようになっているのですが，これは良いのでしょうか？

As you pointed out, this is a bug of test.
Parameter is overrided by the last one, so in this case only `/consensus_tracking/debug/image` is tested.

> 本題としてはこれを上記のように直して，
> ```
> rostest jsk_perception consensus_tracking.test
> ```
> すると，以下のようにtestが成功します．
https://gist.github.com/taichiH/377f83e4a22408b578611ec3228c7710

The test finished with `ROSInterruptException` before the timeout of `test_consensus_tracking` reached, so the test result is actually neither success nor failure. (Test result below indicates it).
```
SUMMARY
 * RESULT: SUCCESS
 * TESTS: 0
 * ERRORS: 0
 * FAILURES: 0
```
This occurs because `rosbag_play` node is `required="true"` and it shutdowns automatically at the end of play back (Loop is somehow disabled for now.)

> しかしsampleを立ち上げると
> ```
> roslaunch jsk_perception sample_consensus_tracking.launch
> ```
> このようにプロセスが終了してしまいます．
> ```
> [consensus_tracking-3] process has died [pid 31885, exit code -11, cmd [...]

 I will try to fix this.

Affonso-Gui, 2019-11-20:
Also had some troubles with ConsensusTracking sample some time ago. Waiting for updates :+1:

PR #2475 (2020-02-28): Fix test for consensus_tracking
Close #2474 

~Waiting for https://github.com/jsk-ros-pkg/jsk_3rdparty/pull/179 to be released.~
This is now released.
closed 2020-02-28 by k-okada