jsk-ros-pkg/jsk_recognition #2494: Asus Xtionを起動できない．

shmpwk, 2020-05-13:
Ubuntuで，[Xtion_PRO_LIVE](https://www.asus.com/jp/3D-Sensor/Xtion_PRO_LIVE/HelpDesk_Warranty/)と思われるカメラを[install_xtion2.rst](https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/doc/install_xtion2.rst)に従ってセットアップしていました．ググると，xtion pro liveもxtion2と同じようにセットアップすればよいと書いてあったので，そうしました．

問題点は，`./SimpleViewer`で
```
Can't find video camera!
Can't find Depth camera!
Cannot identify '/dev/video0': 2, No such file or directory
Cannot identify '/dev/video0': 2, No such file or directory
After initialization:

SimpleViewer: Device open failed:
	DeviceOpen using default: no devices found
```
と表示されます．

lsusbで以下のように，カメラを接続しているときのみ，”ASUS ”が出てくるので，PCに認識自体はされているのかと思います．
```
$ lsusb
Bus 005 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 004 Device 006: ID 06cb:00bd Synaptics, Inc. 
Bus 004 Device 005: ID 04f3:2884 Elan Microelectronics Corp. 
Bus 004 Device 004: ID 13d3:56bc IMC Networks 
Bus 004 Device 003: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 004 Device 002: ID 8087:0025 Intel Corp. 
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 007: ID 0458:003a KYE Systems Corp. (Mouse Systems) NetScroll+ Mini Traveler / Genius NetScroll 120
Bus 002 Device 008: ID 1d27:0600 ASUS 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
また，/dev/videoや/dev/video0というフォルダは，PCにありませんでした．
```
$ ls -al /dev/video*
ls: '/dev/video*' にアクセスできません: そのようなファイルやディレクトリはありません
```
`./SimpleViewer`より前の手順では，特にエラーは出ていませんでした．

他に試すことを教えていただきたいです．

k-okada, 2020-05-13:
あ，そういうことはしないで
roslaunch openni2_launch openni2.launch で動くものを渡したつもりです．sudo
cpして何かインストールするものがあったら，削除してデフォルトの状況に戻してください．
--
◉ Kei Okada

shmpwk, 2020-05-14:
了解しました．
sudo cpしたものを消去した後，ビルドしてlaunchしようとしましたが，エラーがでました．

まず，build時ですが，errorは出ていないものの，以下のようなwarningが出ました．
```
  catkin_package() DEPENDS on 'libopenni2' but neither
  'libopenni2_INCLUDE_DIRS' nor 'libopenni2_LIBRARIES' is defined.
```
build全体の出力は以下のようになっています．
```
$ catkin b
==> Expanding alias 'b' from 'catkin b' to 'catkin build'
----------------------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/melodic
Workspace:                   /home/shumpeiwakabayashi/xtion_ws
----------------------------------------------------------------------
Build Space:        [exists] /home/shumpeiwakabayashi/xtion_ws/build
Devel Space:        [exists] /home/shumpeiwakabayashi/xtion_ws/devel
Install Space:      [unused] /home/shumpeiwakabayashi/xtion_ws/install
Log Space:         [missing] /home/shumpeiwakabayashi/xtion_ws/logs
Source Space:       [exists] /home/shumpeiwakabayashi/xtion_ws/src
DESTDIR:            [unused] None
----------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
----------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
----------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
----------------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
----------------------------------------------------------------------
[build] Found '2' packages in 0.0 seconds.                                                          
[build] Updating package table.                                                                     
Starting  >>> catkin_tools_prebuild                                                                 
Finished  <<< catkin_tools_prebuild                [ 2.4 seconds ]                                  
Starting  >>> openni2_camera                                                                        
____________________________________________________________________________________________________
Warnings   << openni2_camera:cmake /home/shumpeiwakabayashi/xtion_ws/logs/openni2_camera/build.cmake.000.log
CMake Warning at /opt/ros/melodic/share/catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on 'libopenni2' but neither
  'libopenni2_INCLUDE_DIRS' nor 'libopenni2_LIBRARIES' is defined.
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  CMakeLists.txt:20 (catkin_package)

cd /home/shumpeiwakabayashi/xtion_ws/build/openni2_camera; catkin build --get-env openni2_camera | catkin env -si  /usr/bin/cmake /home/shumpeiwakabayashi/xtion_ws/src/ros-drivers/openni2_camera/openni2_camera --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/shumpeiwakabayashi/xtion_ws/devel/.private/openni2_camera -DCMAKE_INSTALL_PREFIX=/home/shumpeiwakabayashi/xtion_ws/install; cd -
....................................................................................................
____________________________________________________________________________________________________
Warnings   << openni2_camera:make /home/shumpeiwakabayashi/xtion_ws/logs/openni2_camera/build.make.000.log
In file included from /home/shumpeiwakabayashi/xtion_ws/src/ros-drivers/openni2_camera/openni2_camera/include/openni2_camera/openni2_frame_listener.h:35:0,
                 from /home/shumpeiwakabayashi/xtion_ws/src/ros-drivers/openni2_camera/openni2_camera/src/openni2_frame_listener.cpp:33:
/home/shumpeiwakabayashi/xtion_ws/src/ros-drivers/openni2_camera/openni2_camera/include/openni2_camera/openni2_device.h:67:48: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
   OpenNI2Device(const std::string& device_URI) throw (OpenNI2Exception);
                                                ^~~~~
/home/shumpeiwakabayashi/xtion_ws/src/ros-drivers/openni2_camera/openni2_camera/include/openni2_camera/openni2_device.h:99:47: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
   void setImageRegistrationMode(bool enabled) throw (OpenNI2Exception);
                                               ^~~~~
/home/shumpeiwakabayashi/xtion_ws/src/ros-drivers/openni2_camera/openni2_camera/include/openni2_camera/openni2_device.h:100:40: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
   void setDepthColorSync(bool enabled) throw (OpenNI2Exception);
                                        ^~~~~
/home/shumpeiwakabayashi/xtion_ws/src/ros-drivers/openni2_camera/openni2_camera/include/openni2_camera/openni2_device.h:102:43: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
[... 449 lines cut ...]
[FATAL] [1589384285.998636739]: Failed to load nodelet '/camera/depth_metric` of type `depth_image_proc/convert_metric` to manager `camera_nodelet_manager'
process[camera_base_link-11]: started with pid [21394]
process[camera_base_link1-12]: started with pid [21400]
process[camera_base_link2-13]: started with pid [21408]
process[camera_base_link3-14]: started with pid [21413]
[ INFO] [1589384286.022624727]: Initializing nodelet with 4 worker threads.
[... cut ...]
[camera/depth_rectify_depth-4] process has died [pid 21353, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load image_proc/rectify camera_nodelet_manager --no-bond image_mono:=depth/image_raw image_rect:=depth/image_rect_raw __name:=depth_rectify_depth __log:=/home/shumpeiwakabayashi/.ros/log/5b2a5e3c-9522-11ea-be50-287fcf9948c3/camera-depth_rectify_depth-4.log].
log file: /home/shumpeiwakabayashi/.ros/log/5b2a5e3c-9522-11ea-be50-287fcf9948c3/camera-depth_rectify_depth-4*.log
```
warningの内容を色々ググってみましたが，結局原因が分かりませんでした．．

k-okada, 2020-05-15:
あ，コンパイルする必要もなくて`apt`で入るものを使ってください．
基本的にはよく使わているもの＝バグが少ないと期待されるもの＝色々なところで有効だと思われているもの，は`apt`でリリースされているので，コンパイルしなければいけないようなことは無いものだと持ってすすめるのが吉です．

shmpwk, 2020-05-15:
なるほど，わかりました．
aptで以下は入っているのですが，同じエラーがでます．
```
sudo apt-get install ros-$ROS_DISTRO-openni2-launch ros-$ROS_DISTRO-openni2-camera libopenni2-dev
```
パッケージが見つからないというエラーは見当たらないので，他にも原因ありますでしょうか．

k-okada, 2020-05-15:
エラーというのは
```
[FATAL] [1589384286.132364521]: Failed to load nodelet '/camera/driver` of type `openni2_camera/OpenNI2DriverNodelet` to manager `camera_nodelet_manager'
```
でしょうか．

基本的には渡した状態のPCであれば，何も問題なく動くはずなので，そこまで戻すのを目標にしましょう．
```
rm -fr  /home/shumpeiwakabayashi/xtion_ws
```
していらないワークスペースは消して，
```
/tmp/ASUS-Linux-x64-OpenNI2.2/install.sh
```
をみるとuninstall というオプションがあるからこれで消して，
```
            ln -s libglut.so.3.9.0 ${GLUT_PATH}/libglut.so.3
```
も無理やりイントールするように見えるんだけど，
```
sudo aptr-get install --reinstall freeglut3
```
して，
それでもダメなら
```
ldd /opt/ros/melodic/lib/libopenni2_camera_nodelet.so 
```
みにしてNot Foundみたいなのがでている気がする．

これ以外に何か自分で`sudo apt` いがいの`sudo`コマンドを実行してコピーしたりしている事があると，そこも原因になりえます．

shmpwk, 2020-05-15:
ご丁寧にありがとうございます．
> エラーというのは
[FATAL] [1589384286.132364521]: Failed to load nodelet '/camera/driver` of type `openni2_camera/OpenNI2DriverNodelet` to manager `camera_nodelet_manager'
でしょうか

はい，そうです．
> sudo aptr-get install --reinstall freeglut3
して，

この後も，先程と同じエラーがでました．
lddで共有ライブラリの依存関係はたくさん出てきました．
はい，過去にsudoしてしまったものを念入りに調べてみようと思います．

```
ldd /opt/ros/melodic/lib/libopenni2_camera_nodelet.so 
	linux-vdso.so.1 (0x00007fff5bbfe000)
	libopenni2_driver_lib.so => /opt/ros/melodic/lib/libopenni2_driver_lib.so (0x00007f1cd693e000)
	libnodeletlib.so => /opt/ros/melodic/lib/libnodeletlib.so (0x00007f1cd66d6000)
	libclass_loader.so => /opt/ros/melodic/lib/libclass_loader.so (0x00007f1cd64af000)
	libroscpp.so => /opt/ros/melodic/lib/libroscpp.so (0x00007f1cd611b000)
	libboost_system.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.65.1 (0x00007f1cd5f16000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f1cd5cf7000)
	libconsole_bridge.so.0.4 => /usr/lib/x86_64-linux-gnu/libconsole_bridge.so.0.4 (0x00007f1cd5af2000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f1cd5769000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f1cd5551000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f1cd5160000)
	libopenni2_wrapper.so => /opt/ros/melodic/lib/libopenni2_wrapper.so (0x00007f1cd4f2f000)
	libcamera_info_manager.so => /opt/ros/melodic/lib/libcamera_info_manager.so (0x00007f1cd4d10000)
	libdynamic_reconfigure_config_init_mutex.so => /opt/ros/melodic/lib/libdynamic_reconfigure_config_init_mutex.so (0x00007f1cd4b08000)
	libimage_transport.so => /opt/ros/melodic/lib/libimage_transport.so (0x00007f1cd488a000)
	librosconsole.so => /opt/ros/melodic/lib/librosconsole.so (0x00007f1cd4654000)
	libroscpp_serialization.so => /opt/ros/melodic/lib/libroscpp_serialization.so (0x00007f1cd4451000)
	librostime.so => /opt/ros/melodic/lib/librostime.so (0x00007f1cd4231000)
	libboost_thread.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.65.1 (0x00007f1cd400c000)
	libbondcpp.so => /opt/ros/melodic/lib/libbondcpp.so (0x00007f1cd3ddb000)
	libroslib.so => /opt/ros/melodic/lib/libroslib.so (0x00007f1cd3bc8000)
	libtinyxml2.so.6 => /usr/lib/x86_64-linux-gnu/libtinyxml2.so.6 (0x00007f1cd39b4000)
	libboost_filesystem.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.65.1 (0x00007f1cd379a000)
	libPocoFoundation.so.50 => /usr/lib/libPocoFoundation.so.50 (0x00007f1cd33f1000)
	libxmlrpcpp.so => /opt/ros/melodic/lib/libxmlrpcpp.so (0x00007f1cd31d2000)
	libcpp_common.so => /opt/ros/melodic/lib/libcpp_common.so (0x00007f1cd2fc7000)
	libboost_chrono.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_chrono.so.1.65.1 (0x00007f1cd2dc2000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f1cd2a24000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f1cd6da6000)
	libOpenNI2.so.0 => /usr/lib/libOpenNI2.so.0 (0x00007f1cd27a8000)
	libcamera_calibration_parsers.so => /opt/ros/melodic/lib/libcamera_calibration_parsers.so (0x00007f1cd2581000)
	libmessage_filters.so => /opt/ros/melodic/lib/libmessage_filters.so (0x00007f1cd237c000)
	librosconsole_log4cxx.so => /opt/ros/melodic/lib/librosconsole_log4cxx.so (0x00007f1cd2161000)
	librosconsole_backend_interface.so => /opt/ros/melodic/lib/librosconsole_backend_interface.so (0x00007f1cd1f5f000)
	liblog4cxx.so.10 => /usr/lib/x86_64-linux-gnu/liblog4cxx.so.10 (0x00007f1cd1b96000)
	libboost_regex.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_regex.so.1.65.1 (0x00007f1cd188e000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f1cd1686000)
	libuuid.so.1 => /lib/x86_64-linux-gnu/libuuid.so.1 (0x00007f1cd147f000)
	librospack.so => /opt/ros/melodic/lib/librospack.so (0x00007f1cd123c000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f1cd1038000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f1cd0dc6000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f1cd0ba9000)
	libyaml-cpp.so.0.5 => /usr/lib/x86_64-linux-gnu/libyaml-cpp.so.0.5 (0x00007f1cd092f000)
	libapr-1.so.0 => /usr/lib/x86_64-linux-gnu/libapr-1.so.0 (0x00007f1cd06fa000)
	libaprutil-1.so.0 => /usr/lib/x86_64-linux-gnu/libaprutil-1.so.0 (0x00007f1cd04cf000)
	libicui18n.so.60 => /usr/lib/x86_64-linux-gnu/libicui18n.so.60 (0x00007f1cd002e000)
	libicuuc.so.60 => /usr/lib/x86_64-linux-gnu/libicuuc.so.60 (0x00007f1ccfc76000)
	libboost_program_options.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_program_options.so.1.65.1 (0x00007f1ccf9f5000)
	libpython2.7.so.1.0 => /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0 (0x00007f1ccf478000)
	libcrypt.so.1 => /lib/x86_64-linux-gnu/libcrypt.so.1 (0x00007f1ccf240000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007f1ccf00e000)
	libicudata.so.60 => /usr/lib/x86_64-linux-gnu/libicudata.so.60 (0x00007f1ccd465000)
	libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007f1ccd262000)
```

k-okada, 2020-05-15:
なるほど．次は
 roslaunch openni2_launch openni2.launch debug:=true
でしょうか．
--
◉ Kei Okada

k-okada, 2020-05-15:
あとは 
```
sudo apt-get dist-upgrade
```
でしょうか．

shmpwk, 2020-05-15:
> あとは
sudo apt-get dist-upgrade
でしょうか．

こちら解決できませんでした．

> なるほど．次は
 roslaunch openni2_launch openni2.launch debug:=true
でしょうか．

これでデバッガーを開いたところ，以下のようにpthread_cond_wait.cが存在しないと出ました．しかしながら，解決法がわかりませんでした．

[image]

k-okada, 2020-05-15:
複数のスレッドがはしっているので
https://sourceware.org/gdb/current/onlinedocs/gdb/Threads.html
で，info threads, thread 2, thread 3... として違うスレッドがどこでエラーになっているか試してみてください．
あと，
sudo apt-get install ros-melodic-openni2-camera-dbgsym
とすると -g でコンパイルしたバイナリが入ります．

あと，ASUS-Linux-x64-OpenNI2.2/install.sh  みると

            cp ${SCRIPT_PATH}/ASUS/Xtion2/lib/${VIDEO_RULES}
/etc/udev/rules.d/${VIDEO_RULES}
            chmod a+r /etc/udev/rules.d/${VIDEO_RULES}

があるけど，uninstall.sh できえないから念の為これもけしておく．．．

shmpwk, 2020-05-16:
[これ](https://doss.eidos.ic.i.u-tokyo.ac.jp/html/gdb_step_into_libraries.html)を見ながら，試してみましたが，やはり未解決です．

12個のthreadを１つ１つ見てエラーとなる場所を探しましたが，エラーとなる部分がありませんでした．
例えば，1つ目のthreadを見ると，大体pthread_cond_waitでした．[これ](https://linuxjm.osdn.jp/html/glibc-linuxthreads/man3/pthread_cond_wait.3.html)をみるとgdbでstepし続けてもエラーがでないと考えました．実際，ずっとstepを連打したところ，最終的には，Single stepping until exit from function which has no line number informationとでました．
[image]
[image]
~ 中略 ~
[image]

[これ](https://ch.nicovideo.jp/famos/blomaga/ar1288539)をみて，gdbを最新版にしましたが，だめでした．
あとは，-gでコンパイルしたものと，しなかったものが混ざっているのが原因だったり，-gではなく-ggdbでないとダメだったなどという解決法がでてきましたが，具体的に今回できることがなさそうだと思いました．

他のthreadについては，全体の約半分のthreadはthread１と同じ挙動をします．残りのthreadについては，1 step 進めただけでずっとに停止しないままになってしまいます．
例えば，thread4です．
[image]

ASUS-Linux-x64-OpenNI2.2/install.shでインストールしてしまったものは全て消去したはずです．

gbdを初めて使用して，使いこなせていない部分があるかもしれませんが，他に解決手段があれば教えていただきたいです．

k-okada, 2020-05-17:
> これをみて，gdbを最新版にしましたが，だめでした．

基本的には{packages,jp.archive}.ubuntu.com/ubuntu 以下からダウンロードされるものでちゃんと動く様になっていて，この環境で世界中の人が動かしていて，なにかおかしいことがあったら，誰かがどこかのそのトラブルに見舞われて，その修正はやはりubuntu.com/ubuntu に反映される，というのがオープンソースの仕組みなので，デフォルトのままつかえば必ず動く，という強い信念を持つのがとっても大切です．
なので，これは一旦戻しましょう．
次に信じていいのはpacakges.ros.orgです．他からインストールするものは使わないのが良いです．
同様にインターネット上の情報もstackoverflowかaskubuntu以外の情報は信じないのが良いです．

とにかくどこかのファイルがオリジナルのものではなくなっている可能性が高く，それをロードする際にエラーになっているように見えるので
１）　gdbですが，step ではなく，where でどこでエラーになっているか，その元の関数を見つけるのが大切です．
2)  strace/ltrace も便利なデバッグツールなので```wget https://raw.githubusercontent.com/ros-drivers/openni2_camera/4128c9d2487b42b3d89a73d32ae9cdd07bc697bf/openni2_launch/launch/openni2.launch ```
して
```
roslaunch --screen ./openni2.launch debug:=true num_worker_threads:=1 2> >(tee log.txt)
```
してください．以下のように変更すれば，
```
diff --git a/openni2_launch/launch/openni2.launch b/openni2_launch/launch/openni2.launch
index 40b2b8c..d23b321 100644
--- a/openni2_launch/launch/openni2.launch
+++ b/openni2_launch/launch/openni2.launch
@@ -79,7 +79,7 @@
 
   <!-- Optionally launch manager in GDB, for debugging -->
   <!-- <arg name="debug" default="false" /> -->
-  <arg if="$(arg debug)" name="launch_prefix" value="xterm -e strace" />
+  <arg if="$(arg debug)" name="launch_prefix" value="strace" />
   <arg unless="$(arg debug)" name="launch_prefix" value="" />
   <!-- Worker threads -->
   <!-- <arg name="num_worker_threads" /> --
```
今後はlog.txtに
```
stat("/opt/ros/melodic/lib/lib/libopenni2_camera_nodelet.so", 0x7ffca7decdb0) = -1 ENOENT (No such file or directory)
stat("/opt/ros/melodic/lib//libopenni2_camera_nodelet.so", {st_mode=S_IFREG|0644, st_size=35352, ...}) = 0
openat(AT_FDCWD, "/opt/ros/melodic/lib//libopenni2_camera_nodelet.so", O_RDONLY|O_CLOEXEC) = 10
read(10, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0000;\0\0\0\0\0\0"..., 832) = 832
fstat(10, {st_mode=S_IFREG|0644, st_size=35352, ...}) = 0
mmap(NULL, 2130400, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 10, 0) = 0x7fb9c7ae2000
mprotect(0x7fb9c7ae9000, 2097152, PROT_NONE) = 0
mmap(0x7fb9c7ce9000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 10, 0x7000) = 0x7fb9c7ce9000
close(10)      
```
みたいになりって，このへんで落ちているんでは何かと思います．

shmpwk, 2020-05-17:
> なので，これは一旦戻しましょう．
次に信じていいのはpacakges.ros.orgです．他からインストールするものは使わないのが良いです．
同様にインターネット上の情報もstackoverflowかaskubuntu以外の情報は信じないのが良いです．

了解しました．

> １）　gdbですが，step ではなく，where でどこでエラーになっているか，その元の関数を見つけるのが大切です．

こちら関数の呼び出し関係を見たところ，エラーを起こしている箇所がないように
見えます．どのthreadも他が動くのをwaitしているという感じでしょうか．

[image]
[image]
[image]
[image]
[image]
[image]

> 2)  strace/ltrace も便利なデバッグツールなので...

log.txtを見たところ，
まず，libopenni2_camera_nodelet.soは次のように読み込めているのかなと判断しました．
```
stat("/home/shumpeiwakabayashi/catkin_ws/devel/lib/lib/libopenni2_camera_nodelet.so", 0x7ffc6f538cb0) = -1 ENOENT (No such file or directory)
stat("/home/shumpeiwakabayashi/catkin_ws/devel/lib//libopenni2_camera_nodelet.so", 0x7ffc6f538cb0) = -1 ENOENT (No such file or directory)
stat("/opt/ros/melodic/lib/lib/libopenni2_camera_nodelet.so", 0x7ffc6f538cb0) = -1 ENOENT (No such file or directory)
stat("/opt/ros/melodic/lib//libopenni2_camera_nodelet.so", {st_mode=S_IFREG|0644, st_size=35352, ...}) = 0
openat(AT_FDCWD, "/opt/ros/melodic/lib//libopenni2_camera_nodelet.so", O_RDONLY|O_CLOEXEC) = 14
read(14, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0000;\0\0\0\0\0\0"..., 832) = 832
fstat(14, {st_mode=S_IFREG|0644, st_size=35352, ...}) = 0
mmap(NULL, 2130400, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 14, 0) = 0x7f65f1bc4000
mprotect(0x7f65f1bcb000, 2097152, PROT_NONE) = 0
mmap(0x7f65f1dcb000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 14, 0x7000) = 0x7f65f1dcb000
close(14)    
```
この後，様々な.soの共有ライブラリが同様に読み込まれていました．
読み込まれていなさそうなのは，
```
access("/usr/lib/OpenNI.ini", F_OK)     = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/OpenNI2/Drivers", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = -1 ENOENT (No such file or directory)
```
より，OpenNI.iniとOpenNI2/Driversなのかと思いました．
OpenNI.iniはetc/openni2にあり，OpenNI2/Driversはコンピュータ上にどこにも存在しませんでした．

その後に
```
[31m[FATAL] [1589724268.147412856]: Failed to load nodelet '/camera/driver` of type `openni2_camera/OpenNI2DriverNodelet` to manager `camera_nodelet_manager'[0m
```
とlogがあったことからも，OpenNI.iniとOpenNI2/Driversが読み込まれていないことが原因なのかなと思いました．
ですが，これは，[XtionのSDK](https://jsk-docs.readthedocs.io/projects/jsk_recognition/en/latest/install_xtion2.html#install-sdk-try-sample)に関係するもので，rosとは直接関係ないのかと思いました．

この後，しばしメモリをreadしようとしてResource temporarily unavailableになりながら，libcompressed_image_transport.so やimage_transportが繰り返されました．
これは，既にfailed to load nodeletしているので，当然失敗するのかなと思いました．そして，connection timed outしました．

以上で一通りログを見たつもりではありますが，見落としているポイントはありますでしょうか．

k-okada, 2020-05-18:
>    1. strace/ltrace も便利なデバッグツールなので...

全てのログをgist.github.com などにおいて共有してください．

トラブっているときは自分の思い込みで見るところを間違っているる可能性が高いので，とにかく全ログを共有するのが重要で鵜s．

shmpwk, 2020-05-18:
[こちらに](https://gist.github.com/shmpwk/5ed6d93b58e7b98750e48880e595dbb8)ログを置きました．ご確認お願いします．

k-okada, 2020-05-21:
```
openat(AT_FDCWD, "/home/shumpeiwakabayashi/catkin_ws/devel/lib/tls/x86_64/x86_64/libnodeletlib.so", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/home/shumpeiwakabayashi/cat
```
.bashrc に
```
source ~/catkin_ws/devel/setup.bash
```
などを書いてある気がします．

`~/.bashrc` が `/etc/skel/.bashrc` と同じになるようにして，とにかく自分の手を付かたものは全てバグを生み出していると考えて，デフォルトのシステムと同じ状態から，OSRFがバイナリでリリースしているものだけを信じて`source /opt/ros/melodic/setup.bash` してたちあげてください．
```
$ echo $ROS_PACKAGE_PATH 
/opt/ros/melodic/share
$ echo $LD_LIBRARY_PATH 
/opt/ros/melodic/share/euslisp/jskeus/eus//Linux64/lib:/opt/ros/melodic/lib:/opt/ros/melodic/lib/x86_64-linux-gnu
```

shmpwk, 2020-05-21:
ありがとうございます．
ただ，やはり同じエラーが出てしまいました．

行ったことは，
1. ~/.bashrcの内容を全て/etc/skel/.bashrcからコピペで上書きした．(バックアップはあります)
2.  `export ROS_PACKAGE_PATH=/opt/ros/melodic/share`
3.  `export LD_LIBRARY_PATH=/opt/ros/melodic/share/euslisp/jskeus/eus//Linux64/lib:/opt/ros/melodic/lib:/opt/ros/melodic/lib/x86_64-linux-gnu`とした．
4. `source /opt/ros/melodic/setup.bash`
5. `roslaunch openni2_launch openni2.launch`
このような手順を踏んで，最初と同じエラーが出ました．
再起動や`sudo apt-get dist-upgrade`も行いました．

また，log.txtの出力は[こちら](https://gist.github.com/shmpwk/e5cf8cbafee31c502919bc123f374358)です．catkin_ws関係の出力は消えたものの，その他の出力はメモリアドレスが違う点以外，読み込んだファイルやエラーが同じように見えます．

k-okada, 2020-05-22:
export を自分でやると間違う可能性があるので，とにかくOSRFを信じて`source /opt/ros/melodic/setup.bash` だけやるのがよいです．

で，
```
access("/usr/lib/OpenNI.ini", F_OK)     = -1 ENOENT (No such file or directory)
```
となるけど，
```
$ apt-file search OpenNI.ini
libopenni2-0: /etc/openni2/OpenNI.ini
```
となって，そういうファイルはないので，おそらくその前の
```
openat(AT_FDCWD, "/usr/lib/libOpenNI2.so.0", O_RDONLY|O_CLOEXEC) = 10
```
のバージョンが違うのではないか？
```
$ dpkg -L libopenni2-0
/.
/etc
/etc/openni2
/etc/openni2/OpenNI.ini
/lib
/lib/udev
/lib/udev/rules.d
/lib/udev/rules.d/60-libopenni2-0.rules
/usr
/usr/lib
/usr/lib/OpenNI2
/usr/lib/OpenNI2/Drivers
/usr/lib/OpenNI2/Drivers/libDummyDevice.so.0
/usr/lib/OpenNI2/Drivers/libOniFile.so.0
/usr/lib/OpenNI2/Drivers/libPS1080.so.0
/usr/lib/OpenNI2/Drivers/libPSLink.so.0
/usr/lib/libOpenNI2.so.0
/usr/share
/usr/share/doc
/usr/share/doc/libopenni2-0
/usr/share/doc/libopenni2-0/TODO.Debian
/usr/share/doc/libopenni2-0/changelog.Debian.gz
/usr/share/doc/libopenni2-0/copyright
```
となっており，
```
$ strings /usr/lib/libOpenNI2.so.0 | grep OpenNI.ini
/etc/openni2/OpenNI.ini
```
というところから，`libOpenNI2.so.0`から違うディレクトリの.iniを読んでいるような気がする．
なので，まだ，前自分でインストールしたlibOpenNI2.soが残っていないか．
```
apt-get install --reinstall libopenni2-0
```
でなおらないだろうか．

shmpwk, 2020-05-22:
ありがとうございます．できました！

> `openat(AT_FDCWD, "/usr/lib/libOpenNI2.so.0", O_RDONLY|O_CLOEXEC) = 10`
のバージョンが違うのではないか？

おっしゃる通りだったようです．
`OpenNI.ini`が見つからない点は気づいていましたが，その前の`libOpenNI2.so.0`のバージョンが違うという発想に至りませんでした．

今回，今まで知らなかったデバッグ法を沢山知れて大変勉強になりました．
お忙しい中，お時間をかなり割いて頂いて恐縮でしたが，本当に助かりました．
今後も教えていただいたことを実践していきたいです．
ありがとうございました．

shmpwk, 2020-06-09:
また，Xtionが起動しなくなりました．
エラーはでず，警告が出て止まります．

```
$ roslaunch openni2_launch openni2.launch 
... logging to /home/shumpeiwakabayashi/.ros/log/cd2be5ae-aa54-11ea-8e3e-287fcf9948c3/roslaunch-shumpeiwakabayashi-ThinkPad-T495s-15219.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://shumpeiwakabayashi-ThinkPad-T495s:38037/

SUMMARY
========

PARAMETERS
 * /camera/camera_nodelet_manager/num_worker_threads: 4
 * /camera/depth_rectify_depth/interpolation: 0
 * /camera/driver/color_depth_synchronization: False
 * /camera/driver/color_mode: 5
 * /camera/driver/depth_camera_info_url: 
 * /camera/driver/depth_frame_id: camera_depth_opti...
 * /camera/driver/depth_mode: 5
 * /camera/driver/depth_registration: False
 * /camera/driver/device_id: #1
 * /camera/driver/id_manufacturer: 1d27
 * /camera/driver/id_product: 0601
 * /camera/driver/ir_mode: 5
 * /camera/driver/rgb_camera_info_url: 
 * /camera/driver/rgb_frame_id: camera_rgb_optica...
 * /rosdistro: melodic
 * /rosversion: 1.14.5

NODES
  /camera/
    camera_nodelet_manager (nodelet/nodelet)
    depth_metric (nodelet/nodelet)
    depth_metric_rect (nodelet/nodelet)
    depth_points (nodelet/nodelet)
    depth_rectify_depth (nodelet/nodelet)
    depth_registered_sw_metric_rect (nodelet/nodelet)
    driver (nodelet/nodelet)
    points_xyzrgb_sw_registered (nodelet/nodelet)
    register_depth_rgb (nodelet/nodelet)
    rgb_rectify_color (nodelet/nodelet)
  /
    camera_base_link (tf2_ros/static_transform_publisher)
    camera_base_link1 (tf2_ros/static_transform_publisher)
    camera_base_link2 (tf2_ros/static_transform_publisher)
    camera_base_link3 (tf2_ros/static_transform_publisher)

auto-starting new master
process[master]: started with pid [16156]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to cd2be5ae-aa54-11ea-8e3e-287fcf9948c3
process[rosout-1]: started with pid [16419]
started core service [/rosout]
process[camera/camera_nodelet_manager-2]: started with pid [16441]
process[camera/driver-3]: started with pid [16445]
process[camera/rgb_rectify_color-4]: started with pid [16451]
process[camera/depth_rectify_depth-5]: started with pid [16461]
process[camera/depth_metric_rect-6]: started with pid [16469]
process[camera/depth_metric-7]: started with pid [16476]
process[camera/depth_points-8]: started with pid [16484]
process[camera/register_depth_rgb-9]: started with pid [16495]
process[camera/points_xyzrgb_sw_registered-10]: started with pid [16501]
process[camera/depth_registered_sw_metric_rect-11]: started with pid [16515]
process[camera_base_link-12]: started with pid [16524]
[ INFO] [1591709179.832914906]: Initializing nodelet with 4 worker threads.
process[camera_base_link1-13]: started with pid [16534]
process[camera_base_link2-14]: started with pid [16547]
process[camera_base_link3-15]: started with pid [16559]
[ INFO] [1591709179.880492215]: Device "1d27/0600@2/3" found.
Warning: USB events thread - failed to set priority. This might cause loss of data...
[ WARN] [1591709180.321088366]: Reconnect has been enabled, only one camera should be plugged into each bus

```

カメラに関係するファイルをいじった記憶がないので，なぜ起動しなくなってしまったのかよくわかっていません．

shmpwk, 2020-06-09:
関係あるか分かりませんが，現在デフォルトで付いている内蔵カメラが認識されなくなりました．Thinkpad T495sです．

最初は使えていたのですが，他のXtion起動関係を修復しているうちにいつの間にか作動しなくなりました．
具体的には，zoomやskypeで内カメを使おうとしても，デバイスにアクセスできませんと表示されます．
lspciやlsusbでカメラを探してみましたが，見つかりませんでした．
もしかしたら，Xtionをいじっていた際に何か内部カメラの設定に関する重要なファイルを消してしまった可能性があるかもしれません．

```
$ lsusb
Bus 005 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 004 Device 006: ID 06cb:00bd Synaptics, Inc.
Bus 004 Device 005: ID 04f3:2884 Elan Microelectronics Corp.
Bus 004 Device 004: ID 13d3:56bc IMC Networks
Bus 004 Device 003: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 004 Device 002: ID 8087:0025 Intel Corp.
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 002: ID 0458:003a KYE Systems Corp. (Mouse Systems) NetScroll+ Mini Traveler / Genius NetScroll 120
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
$ lspci
00:00.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 15d0
00:00.2 IOMMU: Advanced Micro Devices, Inc. [AMD] Device 15d1
00:01.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe Dummy Host Bridge
00:01.2 PCI bridge: Advanced Micro Devices, Inc. [AMD] Device 15d3
00:01.3 PCI bridge: Advanced Micro Devices, Inc. [AMD] Device 15d3
00:01.4 PCI bridge: Advanced Micro Devices, Inc. [AMD] Device 15d3
00:01.5 PCI bridge: Advanced Micro Devices, Inc. [AMD] Device 15d3
00:01.7 PCI bridge: Advanced Micro Devices, Inc. [AMD] Device 15d3
00:08.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe Dummy Host Bridge
00:08.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Device 15db
00:14.0 SMBus: Advanced Micro Devices, Inc. [AMD] FCH SMBus Controller (rev 61)
00:14.3 ISA bridge: Advanced Micro Devices, Inc. [AMD] FCH LPC Bridge (rev 51)
00:18.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 15e8
00:18.1 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 15e9
00:18.2 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 15ea
00:18.3 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 15eb
00:18.4 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 15ec
00:18.5 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 15ed
00:18.6 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 15ee
00:18.7 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 15ef
01:00.0 Network controller: Intel Corporation Wireless-AC 9260 (rev 29)
02:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller SM981/PM981
03:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 0e)
03:00.1 Serial controller: Realtek Semiconductor Co., Ltd. Device 816a (rev 0e)
03:00.2 Serial controller: Realtek Semiconductor Co., Ltd. Device 816b (rev 0e)
03:00.3 IPMI Interface: Realtek Semiconductor Co., Ltd. Device 816c (rev 0e)
03:00.4 USB controller: Realtek Semiconductor Co., Ltd. Device 816d (rev 0e)
04:00.0 Wireless controller [0d40]: Intel Corporation Device 7360 (rev 01)
05:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS522A PCI Express Card Reader (rev 01)
06:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Picasso (rev d1)
06:00.1 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] Device 15de
06:00.2 Encryption controller: Advanced Micro Devices, Inc. [AMD] Device 15df
06:00.3 USB controller: Advanced Micro Devices, Inc. [AMD] Device 15e0
06:00.4 USB controller: Advanced Micro Devices, Inc. [AMD] Device 15e1
06:00.5 Multimedia controller: Advanced Micro Devices, Inc. [AMD] Device 15e2
06:00.6 Audio device: Advanced Micro Devices, Inc. [AMD] Device 15e3

```

knorth55, 2020-06-09:
まずPCの再起動してみてください。
繋がっているUSBを全部抜いてみて、```
lsusb
rosrun openni2_camera list_devices
```
で両者でデバイスが認識されているかを確認してみてください。

shmpwk, 2020-06-10:
ありがとうございます．再起動をした後

繋がっているUSBを全部抜いた結果です．
```
$ lsusb
Bus 005 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 004 Device 006: ID 06cb:00bd Synaptics, Inc. 
Bus 004 Device 005: ID 04f3:2884 Elan Microelectronics Corp. 
Bus 004 Device 004: ID 13d3:56bc IMC Networks 
Bus 004 Device 003: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 004 Device 002: ID 8087:0025 Intel Corp. 
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
```
$ rosrun openni2_camera list_devices
Found 0 devices:
```

Xtionのみを接続させたときです．
```
$ lsusb
Bus 005 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 004 Device 006: ID 06cb:00bd Synaptics, Inc. 
Bus 004 Device 005: ID 04f3:2884 Elan Microelectronics Corp. 
Bus 004 Device 004: ID 13d3:56bc IMC Networks 
Bus 004 Device 003: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 004 Device 002: ID 8087:0025 Intel Corp. 
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 005: ID 1d27:0600 ASUS 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
```
$ rosrun openni2_camera list_devices
[ INFO] [1591749083.022492250]: Device "1d27/0600@2/5" found.
Found 1 devices:

Device #0:
Uri: 1d27/0600@2/5 (Vendor: PrimeSense, Name: PS1080, Vendor ID: 1d27, Product ID: 600)

Warning: USB events thread - failed to set priority. This might cause loss of data...
Serial number: 1204110276
```

このとき，`$ roslaunch openni2_launch openni2.launch `のときと同じ警告`Warning: USB events thread - failed to set priority. This might cause loss of data...`がでました．

knorth55, 2020-06-10:
ん、これいま`lsusb`で`ASUS`でてるから認識されたのでは？
warning無視して`rosrun image_view image_view image:=/camera/rgb/image_raw`してみてください

shmpwk, 2020-06-10:
できました．
原因は，お恥ずかしながらうっかりしており，image_viewを立ち上げていなかったことです．カメラ映像が出力されないのも当然ですね，，，
お騒がせしました．

closed 2020-06-10 by shmpwk

knorth55, 2020-06-10:
`image_view`を使う以外でカメラ画像の出力をチェックする方法ってどうやってたんですか？
`rostopic hz /camera/rgb/image_raw`とかですか？

shmpwk, 2020-06-10:
いえ，image_viewを使っていました．僕の過去の記憶がなくなっただけです笑
いろいろ試してすぐにやったことを忘れてしまうので，メモをとっていきたいです．

knorth55, 2020-06-10:
なるほど（？）まぁ解決したならよかったです