jsk-ros-pkg/jsk_recognition #2496: Xtionカメラからの点群に色がつかない

Kanazawanaoaki, 2020-05-14:
Xtionカメラと openni2_launch/launch/openni2.launchを使って(https://github.com/Kanazawanaoaki/jsk_demos/blob/kanazawa-ow/jsk_2020_04_chahakobi/launch/xtion_test/xtion_test.launch )のようにして点群を試してみようと思ったのですが点群に色が付きませんでした。
[image]
Rvizの設定かと思いましたが、Color Transformerを変更しても色が付きませんでした。
[image]

Kanazawanaoaki, 2020-05-14:
PointCloudToPCDで点群をサブスクライブしてPCDファイルに保存するようにして(https://github.com/Kanazawanaoaki/jsk_demos/blob/kanazawa-ow/jsk_2020_04_chahakobi/launch/xtion_test/ptcloud2pcd_test.launch )
そのPCDファイル(https://raw.githubusercontent.com/Kanazawanaoaki/jsk_demos/kanazawa-ow/jsk_2020_04_chahakobi/pcd/xtion_test/xtion_pcd_1589422287446802.pcd )をみてみたら
```
# .PCD v0.7 - Point Cloud Data file format
VERSION 0.7
FIELDS x y z
SIZE 4 4 4
TYPE F F F
COUNT 1 1 1
WIDTH 640
HEIGHT 480
VIEWPOINT 0 0 0 1 0 0 0
POINTS 307200
DATA ascii
```
とFIELDS x y zになっていたのでそもそもの点群のトピックに色が着いていなかったのかなと思いました。  

jsk_pcl_rosのAddColorFromImageを使ってみようと思い、
```
<node name="add_color_from_image"
        pkg="nodelet" type="nodelet"
        args="load jsk_pcl/AddColorFromImage">
    <remap from="~input" to="/camera/depth/points" />
    <remap from="~input/image" to="/camera/rgb/image_raw" />
    <remap from="~input/camera_info" to="/camera/rgb/camera_info" />
  </node>
```
(https://github.com/Kanazawanaoaki/jsk_demos/blob/kanazawa-ow/jsk_2020_04_chahakobi/launch/xtion_test/xtion_color_test.launch )としてみましたが以下のようにlogが出てprocess has diedになってしまいました。
```
[add_color_from_image-17] process has died [pid 8069, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load jsk_pcl/AddColorFromImage ~input:=/camera/depth/points ~input/image:=/camera/rgb/image_raw ~input/camera_info:=/camera/rgb/camera_info __name:=add_color_from_image __log:=/home/kanazawa/.ros/log/67d7f26c-958a-11ea-8d9e-287fcf98ebf8/add_color_from_image-17.log].
log file: /home/kanazawa.ros/log/67d7f26c-958a-11ea-8d9e-287fcf98ebf8/add_color_from_image-17*.log
```

Kanazawanaoaki, 2020-05-14:
もしかするとgazeboの問題で今回の問題とは直接関係しないかもしれませんが、そもそも以前からRvizで点群の色がつかなくて、
```
roslaunch pr2eus_tutorials pr2_tabletop_sim.launch physics:=dart
```
をしても、
[image]
shadowsを無効化したworldを作っても色が出ませんでした。
[image]
この場合もRvizのPointCloud2の設定を変えても色は出ませんでしたが、点群をPCDファイルに保存すると色は着いていて、HSIなども使う事が出来ます。

k-okada, 2020-05-14:
`/camera/depth_registered/points`
をサブスクライブしている？ `/camera/depth/points` にたいして色情報をregisterしているものです．
後はこれをecho してみて以下のようにfields にrgbがはいあっている確認してみてください．

```
header: 
  seq: 2162
  stamp: 
    secs: 1589429563
    nsecs: 373323671
  frame_id: "camera_rgb_optical_frame"
height: 480
width: 640
fields: 
  - 
    name: "x"
    offset: 0
    datatype: 7
    count: 1
  - 
    name: "y"
    offset: 4
    datatype: 7
    count: 1
  - 
    name: "z"
    offset: 8
    datatype: 7
    count: 1
  - 
    name: "rgb"
    offset: 16
    datatype: 7
    count: 1
is_bigendian: False
point_step: 32
row_step: 20480
```

Kanazawanaoaki, 2020-05-14:
そうなんですね、`/camera/depth_registered/points` はパブリッシュされていないのか、
```
$ rostopic echo /camera/depth_registered/points
```
としても何も表示されませんし、
```
$ rostopic hz /camera/depth_registered/points
subscribed to [/camera/depth_registered/points]
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
```
となってしまうので`/camera/depth/points `を使っていました。
```
$ rostopic info /camera/depth_registered/points
Type: sensor_msgs/PointCloud2

Publishers: 
 * /camera/camera_nodelet_manager (http://kanazawa-ThinkPad-T495s:37157/)

Subscribers: None

```
とはなっているのですがよくわかりません。

k-okada, 2020-05-14:
あれ？送るの間違えちゃったかな．
```
$ lsusb | grep -i asus
Bus 001 Device 027: ID 1d27:0600 ASUS 
```

Kanazawanaoaki, 2020-05-14:
```
$ lsusb | grep -i asus
Bus 002 Device 006: ID 1d27:0600 ASUS 
```
となりました。カメラは問題なさそうですね。  
僕が作ったlaunchファイル(https://github.com/Kanazawanaoaki/jsk_demos/blob/kanazawa-ow/jsk_2020_04_chahakobi/launch/xtion_test/xtion_test.launch )ではrvizで点群をみられるようにcamera_depth_optical_frameにtfを出していたのですが、
```
  <include file="$(find openni2_launch)/launch/openni2.launch" />
  
  <node name="referene_tf_publisher"
        pkg="tf" type="static_transform_publisher"
        args="0 0 1 0 0 -1.57 map camera_depth_optical_frame 10" />

```
tf部分をコメントアウトすると、`/camera/depth_registered/points`が出るようになりました。
```
$ rostopic hz /camera/depth_registered/points 
subscribed to [/camera/depth_registered/points]
average rate: 26.638
	min: 0.014s max: 0.082s std dev: 0.01729s window: 10
average rate: 29.193
	min: 0.014s max: 0.082s std dev: 0.00998s window: 40
average rate: 29.532
```
tfを出さないとrvizで点群をみられないようなのですが、どうするのが良いでしょうか。
[image]

k-okada, 2020-05-14:
Xtion ,以下のようにカメラが２つついているのと３つついているのがあるんだけど，３つあるよね？２つだとRGBでてこないです．
https://www.asus.com/media/global/products/hahEFPMWY9UVDL7z/ODjGCbIYqA23dyML_500.jpg
https://www.asus.com/3D-Sensor/Xtion_PRO/
まずはFrameですが，rvizのdisplayのFixed Frameのcamera_linkを直打ちすると見えるはずです．
--
◉ Kei Okada

Kanazawanaoaki, 2020-05-14:
RvizでDisplays->Global Optinos -> Fixed Frameをcamera_depth_optical_frameにしたところ点群が表示されるようになりました。
[image]

`/camera/depth_registered/points`を指定してDisplays -> PointCloud2 の Color TransformerをRGB8に指定しても僕の環境ではRvizの点群の色は付きませんが、
```
$ rostopic echo /camera/depth_registered/points -n 1 | grep header -A 30
header: 
  seq: 12895
  stamp: 
    secs: 1589464248
    nsecs:  54828875
  frame_id: "camera_rgb_optical_frame"
height: 480
width: 640
fields: 
  - 
    name: "x"
    offset: 0
    datatype: 7
    count: 1
  - 
    name: "y"
    offset: 4
    datatype: 7
    count: 1
  - 
    name: "z"
    offset: 8
    datatype: 7
    count: 1
  - 
    name: "rgb"
    offset: 16
    datatype: 7
    count: 1
is_bigendian: False
point_step: 32
```
とfields にrgbが入っているのでXtionからのポイントクラウドのトピックは正しい動作になったのかなと思います。

Kanazawanaoaki, 2020-05-14:
> Xtion ,以下のようにカメラが２つついているのと３つついているのがあるんだけど，３つあるよね？２つだとRGBでてこないです． https://www.asus.com/media/global/products/hahEFPMWY9UVDL7z/ODjGCbIYqA23dyML_500.jpg https://www.asus.com/3D-Sensor/Xtion_PRO/ [...]

すみません見落としていました。
カメラは３つ着いています。https://www.asus.com/media/global/products/hahEFPMWY9UVDL7z/ODjGCbIYqA23dyML_500.jpg このタイプです。

k-okada, 2020-05-15:
camera_depth_optical_frame はcamera_linkか何かにならないかな？
rosrun tf view_framesの結果のPDFもつけてみてください。
--
◉ Kei Okada

Kanazawanaoaki, 2020-05-15:
RvizのDisplays->Global Optinos -> Fixed Frameでcamera_linkを指定することはできました。
[image]
トピックのframe_idは"camera_rgb_optical_frame"になっています。
```
$ rostopic echo /camera/depth_registered/points -n 1 | grep header -A 30
header: 
  seq: 564
  stamp: 
    secs: 1589520680
    nsecs: 953724377
  frame_id: "camera_rgb_optical_frame"
height: 480
width: 640
fields: 
  - 
    name: "x"
    offset: 0
    datatype: 7
    count: 1
  - 
    name: "y"
    offset: 4
    datatype: 7
    count: 1
  - 
    name: "z"
    offset: 8
    datatype: 7
    count: 1
  - 
    name: "rgb"
    offset: 16
    datatype: 7
    count: 1
is_bigendian: False
point_step: 32
```
`rosrun tf view_frames`をした時のPDFです。
[frames.pdf](https://github.com/jsk-ros-pkg/jsk_recognition/files/4632422/frames.pdf)

k-okada, 2020-05-15:
うーん，おかしいな．
```
$ roslaunch --screen openni2_launch openni2.launch 
```
としたときのログを確認するのと，`rviz`のColorTransformのところのRGB8をAxisColorとかにしたら色は出るんだよね？

Kanazawanaoaki, 2020-05-15:
```
$ roslaunch --screen openni2_launch openni2.launch 
```
のログは以下に載せて起きました。
https://gist.github.com/Kanazawanaoaki/1c81bb397db4f1fd56570466a04f9a81

[ WARN] [1589535122.070369930]: Reconnect has been enabled, only one camera should be plugged into each bus

[ WARN] [1589535145.688045656]: Camera calibration file /home/kanazawa/.ros/camera_info/rgb_PS1080_PrimeSense.yaml not found.
あたりがあやしいのでしょうか。

`rviz`のColorTransformは僕の環境だとどれを選択しても色は出ないです。これはXtionカメラだけでなく、GazeboのPR2のカメラでも色は出ないです。

k-okada, 2020-05-15:
だとするとgpu 変という説になりそうだけど，僕のt480s (intel + gpu なし）でも色は出るんだよね．AMDそこまでいまいちだろうか．．．

という仮説を確認するために
```
rosbag play -l $(rospack find jsk_pcl_ros)/sample/data/sample_door_handle_detector.bag
```
で`rviz`で結果を見てみてください．
ほぼ白黒で見づらいですが，RGBだと右側のチラシに緑色がでるはずです．

％　研究室，懐かしいですね．．．

[image]

MiyabiTane, 2020-05-15:
こちらのlaunchを立ち上げ、
https://github.com/MiyabiTane/rtmros_tutorials/blob/add_by_tanemoto/hironx_tutorial/launch/hiro_lunch_box.launch

roslaunch openni2_launch openni2.launch
を実行し、rvizで
Fixed Frame -> camera_link
Topic -> /camera/depth_registered/points
とすることで色付き点群を出すことができました。
私のPCはT495sです。

[image]

Kanazawanaoaki, 2020-05-15:
>Dsiplays->PointCloud2->Style も色々いじってみるのかな．
これでした。
StyleをPointsにすると色がつくようになりました。
[image]
[image]

closed 2020-05-17 by Kanazawanaoaki