jsk-ros-pkg/jsk_recognition #2499: Fail to publish static image topic

shmpwk, 2020-05-17:
jsk_recognitionのチュートリアルの最初の[publish_static_image_topic](https://jsk-docs.readthedocs.io/projects/jsk_recognition/en/latest/tutorials/publish_static_image_topic.html) が実行できません．

手順に従ってimage_viewをsubscribeしようとしたところ，セグフォになりました．
```
$ rosrun image_view image_view image:=/image_publisher/output
[ INFO] [1589707886.846567915]: Initializing nodelet with 8 worker threads.
[ INFO] [1589707886.927092045]: Using transport "raw"
Segmentation fault (コアダンプ)
```
上記のA single command versionでもエラーがでました．
```
~ 略 ~
[image_view-3] process has died [pid 16163, exit code -11, cmd /opt/ros/melodic/lib/image_view/image_view image:=image_publisher/output __name:=image_view __log:=/home/shumpeiwakabayashi/.ros/log/f7a84672-9822-11ea-ae8c-287fcf9948c3/image_view-3.log].
log file: /home/shumpeiwakabayashi/.ros/log/f7a84672-9822-11ea-ae8c-287fcf9948c3/image_view-3*.log
```

現在Xtionのカメラが立ち上がらないのも関係ありますでしょうか．

k-okada, 2020-05-17:
image_view はhttps://github.com/ros-perception/image_pipeline/pull/526
 が原因なのimage_viewのかわりにimage_view2を使ってください．
--
◉ Kei Okada

shmpwk, 2020-05-17:
ありがとうございます．無事できました．

```rosrun image_view image_view image:=/image_publisher/output```
を
```rosrun image_view2 image_view2 image:=/image_publisher/output```
に変更してできました．

また，launchする場合は，tutorial_image_publisher.launchにおいて
```<node name="image_view"  pkg="image_view" type="image_view">```
を
```<node name="image_view" pkg="image_view2" type="image_view2">```
に変更してできました．

closed 2020-05-17 by shmpwk