jsk-ros-pkg/jsk_recognition #2503: [PointcloudScreenpoint] Has anyone tried with gazebo PointCloud?

ishiguroJSK, 2020-05-23:
I'm trying to use PointcloudScreenpoint with PointCloud from gazebo.

The PointCloud source is kinect like plugin ↓
`<plugin filename="libgazebo_ros_openni_kinect.so" name="xtion_camera_controller">`
And it looks working fine.

I think topic names or params are remapped correctly,
but any PointCloud seems not to be subscribed.

When I set the logger level to DEBUG, 
I expected this debug print,
https://github.com/jsk-ros-pkg/jsk_recognition/blob/ccdb666ed109e565e89bd35ef7dfa14999550da0/jsk_pcl_ros/src/pointcloud_screenpoint_nodelet.cpp#L282-L283
but it doesn't appear without these suspicious messages.
```
[DEBUG] [1590247731.993518106, 529.245000000]: Connection::drop(2)
[DEBUG] [1590247731.993578584, 529.245000000]: Socket 14 closed with (ERR|HUP|NVAL) events 8196: Success
[DEBUG] [1590247731.993617895, 529.245000000]: TCP socket [14] closed
[DEBUG] [1590247731.993652704, 529.245000000]: Connection::drop(0)
[DEBUG] [1590247731.993672157, 529.245000000]: Connection::drop(2)
[DEBUG] [1590247731.993720129, 529.245000000]: Connection::drop(2)
[DEBUG] [1590247732.371960417, 529.408000000]: SteadyTimer deregistering callbacks.
[DEBUG] [1590247732.844752246, 529.601000000]: Incoming queue was full for topic "/clock". Discarded oldest message (current queue size [0])
```

Has anyone tried with gazebo?
Cc: @708yamaguchi @k-okada 

```
... logging to /home/leus/.ros/log/fd04abcc-9d06-11ea-8daa-7c7a914eb678/roslaunch-W540-2323.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://W540:42837/

SUMMARY
========

CLEAR PARAMETERS
 * /depth_image_proc_nodelet/
 * /pointcloud_screenpoint_nodelet/

PARAMETERS
 * /depth_image_proc_nodelet/queue_size: 30
 * /display_point/base_frame: xtion_link
 * /display_point/ray_srv: /pointcloud_scree...
 * /display_point/sensor_topic: /xtion/rgb/image_raw
 * /pointcloud_screenpoint_nodelet/crop_size: 10
 * /pointcloud_screenpoint_nodelet/publish_point: True
 * /pointcloud_screenpoint_nodelet/publish_points: False
 * /pointcloud_screenpoint_nodelet/queue_size: 10
 * /pointcloud_screenpoint_nodelet/search_size: 16
 * /pointcloud_screenpoint_nodelet/use_point: True
 * /pointcloud_screenpoint_nodelet/use_point_array: True
 * /pointcloud_screenpoint_nodelet/use_rect: False
 * /pointcloud_screenpoint_nodelet/use_sync: False
 * /rosdistro: melodic
 * /rosversion: 1.14.5
 * /xtion/rgb/screenpoint_view/autosize: True

NODES
  /xtion/rgb/
    screenpoint_view (image_view2/image_view2)
  /
    depth_image_proc_nodelet (nodelet/nodelet)
    display_point (jsk_pcl_ros/pointcloud_screenpoint.l)
    pointcloud_screenpoint_nodelet (nodelet/nodelet)
    screenpoint_manager (nodelet/nodelet)

WARNING: unrecognized tag sphinxdoc
[... 133 lines cut ...]
[DEBUG] [1590247750.937832685, 537.191000000]: Connection to publisher [TCPROS connection on port 33294 to [W540:36735 on socket 17]] to topic [/screenpoint_manager/bond] dropped
[DEBUG] [1590247750.937963652, 537.191000000]: getsockopt failed on socket [-1]
[DEBUG] [1590247750.938001409, 537.191000000]: Socket -1 closed with (ERR|HUP|NVAL) events 8193: Unknown error -1
[DEBUG] [1590247750.940467113, 537.192000000]: Socket 18 closed with (ERR|HUP|NVAL) events 8192: Success
[DEBUG] [1590247750.940605415, 537.192000000]: TCP socket [18] closed
[DEBUG] [1590247750.940671634, 537.192000000]: Connection::drop(0)
[DEBUG] [1590247750.940726221, 537.192000000]: Connection to subscriber [callerid=[/screenpoint_manager] address=[TCPROS connection on port 57829 to [127.0.0.1:51736 on socket 18]]] to topic [/screenpoint_manager/bond] dropped
[DEBUG] [1590247750.940756208, 537.192000000]: Connection::drop(2)
[DEBUG] [1590247750.940797603, 537.192000000]: Connection::drop(2)
[DEBUG] [1590247750.941962741, 537.192000000]: Socket [14] received 0/5 bytes, closing
[DEBUG] [1590247750.942028480, 537.192000000]: TCP socket [14] closed
[DEBUG] [1590247750.942059998, 537.192000000]: Connection::drop(0)
[DEBUG] [1590247750.942119596, 537.192000000]: Service client from [callerid=[/screenpoint_manager] address=[TCPROS connection on port 34196 to [W540:36735 on socket 14]]] for [/screenpoint_manager/unload_nodelet] dropped
[DEBUG] [1590247750.942238509, 537.192000000]: getsockopt failed on socket [-1]
[DEBUG] [1590247750.942266992, 537.192000000]: Socket -1 closed with (ERR|HUP|NVAL) events 8193: Unknown error -1
[DEBUG] [1590247750.942291897, 537.192000000]: Connection::drop(2)
[DEBUG] [1590247750.942536042, 537.192000000]: XML-RPC call [lookupService] returned an error (-1): [no provider]
[ WARN] [1590247750.956553880, 537.198000000]: ros::roseus-sigint-handler 2
[ INFO] [1590247750.956643320, 537.198000000]: cell* ROSEUS_EXIT(context*, int, cell**)
[ INFO] [1590247750.956698024, 537.198000000]: exiting roseus 1
shutting down processing monitor...
... shutting down processing monitor complete
done

```

708yamaguchi, 2020-05-23:
I have used `PointcloudScrennpoint` with gazebo last week.
Here is my guess.

I think this is not gazebo problem.
Because `PointcloudScrennpoint` inherits `jsk_topic_tools::ConnectionBasedNodelet`, the `PointcloudScrennpoint` node subscribes topics only when other nodes subscribe the topics which `PointcloudScrennpoint` node advertises.

inherit code:
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/include/jsk_pcl_ros/pointcloud_screenpoint.h#L56
ConnectionBasedNodelet:
https://jsk-common.readthedocs.io/en/latest/jsk_topic_tools/class/connection_based_nodelet.html

I think there are two solutions.
1. set `~always_subscribe` rosparam of `ConnectionBasedNodelet` as true.
2. subscribe topics which `ConnectionBasedNodelet` advertises. For example, `rostopic echo /pointcloud_screenpoint_nodelet/output`

ishiguroJSK, 2020-05-23:
Thanks! solution "2." works!
(I couldn't find "~always_subscribe"
```
$ rosparam get /pointcloud_screenpoint_nodelet/
/pointcloud_screenpoint_nodelet/approximate_sync  /pointcloud_screenpoint_nodelet/queue_size        /pointcloud_screenpoint_nodelet/use_point
/pointcloud_screenpoint_nodelet/crop_size         /pointcloud_screenpoint_nodelet/search_size       /pointcloud_screenpoint_nodelet/use_point_array
/pointcloud_screenpoint_nodelet/publish_point     /pointcloud_screenpoint_nodelet/synchronization   /pointcloud_screenpoint_nodelet/use_rect
/pointcloud_screenpoint_nodelet/publish_points    /pointcloud_screenpoint_nodelet/timeout           /pointcloud_screenpoint_nodelet/use_sync
```
)

closed 2020-05-23 by ishiguroJSK

708yamaguchi, 2020-05-23:
I'm glad it worked well.

For your information:
`~always_subscribe` is not explicitly declared, but works when set.

ishiguroJSK, 2020-05-23:
I see.
It works only if the "~always_subscribe" is already set before init.
So, I had to write it directly on launch file.
Anyway it works and thank you.

(and I found minor typo...
https://github.com/jsk-ros-pkg/jsk_recognition/blob/ccdb666ed109e565e89bd35ef7dfa14999550da0/jsk_pcl_ros/launch/pointcloud_screenpoint.launch#L73)

k-okada, 2020-06-01:
@ishiguroJSK please update documentation, and minor typo and send PR