jsk-ros-pkg/jsk_recognition #2505: ターゲットをmakeするルールがない

shmpwk, 2020-05-24:
[coral のセットアップ](https://github.com/knorth55/coral_usb_ros#workspace-build-melodic)をしているのですが，ワークスペースのビルド時にエラーが起こりました．
恐らく，これは僕の環境の問題であり，[Xttionを起動できない](https://github.com/jsk-ros-pkg/jsk_recognition/issues/2494)という問題に関係あるかと思い，こちらに質問させていただきました．
buildはimage_view2にのみ絞っています．

```

$ catkin build image_view2 
------------------------------------------------------------------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/melodic
Workspace:                   /home/shumpeiwakabayashi/coral_ws
------------------------------------------------------------------------------------------------------------------------------
Build Space:        [exists] /home/shumpeiwakabayashi/coral_ws/build
Devel Space:        [exists] /home/shumpeiwakabayashi/coral_ws/devel
Install Space:      [unused] /home/shumpeiwakabayashi/coral_ws/install
Log Space:          [exists] /home/shumpeiwakabayashi/coral_ws/logs
Source Space:       [exists] /home/shumpeiwakabayashi/coral_ws/src
DESTDIR:            [unused] None
------------------------------------------------------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
------------------------------------------------------------------------------------------------------------------------------
Additional CMake Args:       -DPYTHON_EXECUTABLE=/usr/bin/python3 -DPYTHON_INCLUDE_DIR=/usr/include/python3.6m -DPYTHON_LIBRARY=/usr/lib/x86_64-linux-gnu/libpython3.6m.so
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
------------------------------------------------------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
------------------------------------------------------------------------------------------------------------------------------
Workspace configuration appears valid.
------------------------------------------------------------------------------------------------------------------------------
[build] Found '15' packages in 0.0 seconds.                                                                                  
[build] Package table is up to date.                                                                                         
Starting  >>> cv_bridge                                                                                                      
Starting  >>> image_geometry                                                                                                 
Finished  <<< image_geometry                [ 0.4 seconds ]                                                                  
Finished  <<< cv_bridge                     [ 0.4 seconds ]                                                                  
Starting  >>> image_view2                                                                                                    
_____________________________________________________________________________________________________________________________
Errors     << image_view2:make /home/shumpeiwakabayashi/coral_ws/logs/image_view2/build.make.002.log                         
make[2]: *** No rule to make target '/usr/lib/libOpenNI2.so', needed by '/home/shumpeiwakabayashi/coral_ws/devel/.private/image_view2/lib/image_view2/points_rectangle_extractor'.  Stop.
make[1]: *** [CMakeFiles/points_rectangle_extractor.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** No rule to make target '/usr/lib/libOpenNI2.so', needed by '/home/shumpeiwakabayashi/coral_ws/devel/.private/image_view2/lib/image_view2/image_view2'.  Stop.
make[1]: *** [CMakeFiles/image_view2.dir/all] Error 2
make: *** [all] Error 2
cd /home/shumpeiwakabayashi/coral_ws/build/image_view2; catkin build --get-env image_view2 | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
.............................................................................................................................
Failed     << image_view2:make              [ Exited with code 2 ]                                                           
Failed    <<< image_view2                   [ 0.5 seconds ]                                                                  
[build] Summary: 2 of 3 packages succeeded.                                                                                  
[build]   Ignored:   12 packages were skipped or are blacklisted.                                                            
[build]   Warnings:  None.                                                                                                   
[build]   Abandoned: None.                                                                                                   
[build]   Failed:    1 packages failed.                                                                                      
[build] Runtime: 1.1 seconds total.                                    
```

出ているエラーの１つとしては，/usr/lib/libOpenNI2.soが見つからないということですが，
```
make[2]: *** No rule to make target '/usr/lib/libOpenNI2.so', needed by '/home/shumpeiwakabayashi/coral_ws/devel/.private/image_view2/lib/image_view2/points_rectangle_extractor'.  Stop.
```
/usr/lib/libOpenNI2.so自体は存在します．
```
$ apt-file search libOpenNI2.so
libopenni2-0: /usr/lib/libOpenNI2.so.0
libopenni2-dev: /usr/lib/libOpenNI2.so
```

しかし，以下のファイルは存在しませんでした．
```
/home/shumpeiwakabayashi/coral_ws/devel/.private/image_view2/lib/image_view2/points_rectangle_extractor
```
/home/shumpeiwakabayashi/coral_ws/devel/.private/image_view2/lib/image_view2以下が空でした．

そこで，その場しのぎの方法として， points_rectangle_extractor.cpp ファイルを [ここ](image_view2/lib/image_view2/points_rectangle_extractor) からコピーしてみました. image_view2 file についても同様です．しかし，ダメだったので戻しました．

`CMakeFiles/image_view2.dir/all.`をみてみると，いずれのフォルダにおいてもDependInfo.cmakeファイルでCMAKE_TARGET_LINKED_INFO_FILES の指定がされていて，恐らくこれの指定先がおかしいのかなと思いました，しかし，どこでこれを指定できるのかが分かりませんでした．

リンクファイルの呼び出し関係を見る方法がわからず，ご教授頂きたいです．

k-okada, 2020-05-24:
catkin clean -y
catkin build 2>&1 > build.log
として、結果のbuild.log をgistに貼りましょう。
--
◉ Kei Okada

shmpwk, 2020-05-24:
ありがとうございます．
[こちら](https://gist.github.com/shmpwk/f9f0ab17000a5cfa5c3ea029a00bde70)にbuild.logを貼りました．

shmpwk, 2020-05-24:
その時のコンソール画面の出力は以下のようになっていました．
```
$ catkin build 2>&1 > build.log
_________________________________________________________________________________
Warnings << cv_bridge:cmake /home/shumpeiwakabayashi/coral_ws/logs/cv_bridge/build.cmake.000.log
CMake Warning at /usr/share/cmake-3.10/Modules/FindBoost.cmake:1626 (message):
  No header defined for python3; skipping header check
Call Stack (most recent call first):
  CMakeLists.txt:11 (find_package)

cd /home/shumpeiwakabayashi/coral_ws/build/cv_bridge; catkin build --get-env cv_bridge | catkin env -si  /usr/bin/cmake /home/shumpeiwakabayashi/coral_ws/src/vision_opencv/cv_bridge --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/shumpeiwakabayashi/coral_ws/devel/.private/cv_bridge -DCMAKE_INSTALL_PREFIX=/home/shumpeiwakabayashi/coral_ws/install -DPYTHON_EXECUTABLE=/usr/bin/python3 -DPYTHON_INCLUDE_DIR=/usr/include/python3.6m -DPYTHON_LIBRARY=/usr/lib/x86_64-linux-gnu/libpython3.6m.so; cd -
.................................................................................
_________________________________________________________________________________
Warnings << cv_bridge:make /home/shumpeiwakabayashi/coral_ws/logs/cv_bridge/build.make.000.log
In file included from /home/shumpeiwakabayashi/coral_ws/src/vision_opencv/cv_bridge/src/module.cpp:35:0:
/home/shumpeiwakabayashi/coral_ws/src/vision_opencv/cv_bridge/src/module.hpp: In function ‘int do_numpy_import()’:
/home/shumpeiwakabayashi/coral_ws/src/vision_opencv/cv_bridge/src/module.hpp:39:5: warning: converting to non-pointer type ‘int’ from NULL [-Wconversion-null]
     import_array( );
     ^~~~~~~~~~~~
In file included from /home/shumpeiwakabayashi/coral_ws/src/vision_opencv/cv_bridge/src/module_opencv3.cpp:3:0:
/home/shumpeiwakabayashi/coral_ws/src/vision_opencv/cv_bridge/src/module.hpp: In function ‘int do_numpy_import()’:
/home/shumpeiwakabayashi/coral_ws/src/vision_opencv/cv_bridge/src/module.hpp:39:5: warning: converting to non-pointer type ‘int’ from NULL [-Wconversion-null]
     import_array( );
     ^~~~~~~~~~~~
cd /home/shumpeiwakabayashi/coral_ws/build/cv_bridge; catkin build --get-env cv_bridge | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
.................................................................................
_________________________________________________________________________________
Warnings << image_view2:cmake /home/shumpeiwakabayashi/coral_ws/logs/image_view2/build.cmake.000.log
** WARNING ** io features related to openni2 will be disabled
** WARNING ** io features related to ensenso will be disabled
** WARNING ** io features related to davidSDK will be disabled
** WARNING ** io features related to dssdk will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** io features related to libusb-1.0 will be disabled
** WARNING ** visualization features related to openni2 will be disabled
** WARNING ** visualization features related to ensenso will be disabled
** WARNING ** visualization features related to davidSDK will be disabled
** WARNING ** visualization features related to dssdk will be disabled
** WARNING ** visualization features related to rssdk will be disabled
cd /home/shumpeiwakabayashi/coral_ws/build/image_view2; catkin build --get-env image_view2 | catkin env -si  /usr/bin/cmake /home/shumpeiwakabayashi/coral_ws/src/jsk_common/jsk_ros_patch/image_view2 --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/shumpeiwakabayashi/coral_ws/devel/.private/image_view2 -DCMAKE_INSTALL_PREFIX=/home/shumpeiwakabayashi/coral_ws/install -DPYTHON_EXECUTABLE=/usr/bin/python3 -DPYTHON_INCLUDE_DIR=/usr/include/python3.6m -DPYTHON_LIBRARY=/usr/lib/x86_64-linux-gnu/libpython3.6m.so; cd -
.................................................................................
_________________________________________________________________________________
Errors << image_view2:make /home/shumpeiwakabayashi/coral_ws/logs/image_view2/build.make.000.log
make[2]: *** '/home/shumpeiwakabayashi/coral_ws/devel/.private/image_view2/lib/image_view2/points_rectangle_extractor' に必要なターゲット '/usr/lib/libOpenNI2.so' を make するルールがありません.  中止.
make[2]: *** 未完了のジョブを待っています....
make[2]: *** '/home/shumpeiwakabayashi/coral_ws/devel/.private/image_view2/lib/image_view2/image_view2' に必要なターゲット '/usr/lib/libOpenNI2.so' を make するルールがありません.  中止.
make[2]: *** 未完了のジョブを待っています....
make[1]: *** [CMakeFiles/points_rectangle_extractor.dir/all] Error 2
make[1]: *** 未完了のジョブを待っています....
make[1]: *** [CMakeFiles/image_view2.dir/all] Error 2
make: *** [all] Error 2
cd /home/shumpeiwakabayashi/coral_ws/build/image_view2; catkin build --get-env image_view2 | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
.................................................................................
Failed << image_view2:make                        [ Exited with code 2 ]    
```

k-okada, 2020-05-24:
catkin clean -y
catkin build -vi --make-args VERBOSE=1 2>&1 > build.log
でした。
--
◉ Kei Okada

shmpwk, 2020-05-24:
ありがとうございます．所望のログを得られました．今，リンク関係を確認しています．

shmpwk, 2020-05-24:
[build.log](https://gist.github.com/shmpwk/09a2cd44b5dab841b0b1a58c704015ca)のログファイルを確認したところ，/coral_ws/build/image_view2/CMakeFiles/points_rectangle_extractor.dir/build.makeにおいて，以下のエラーが起こっていることが判りました．
> '/home/shumpeiwakabayashi/coral_ws/devel/.private/image_view2/lib/image_view2/points_rectangle_extractor' に必要なターゲット '/usr/lib/libOpenNI2.so' を make するルールがありません.  中止.

そこで，ターゲットがlibOpenNI2.soとなっていたのをlibOpenNI2.so.0と変更するとこのエラーはなくなりました．

同様の方法で以下のエラーも消えました．
> '/home/shumpeiwakabayashi/coral_ws/devel/.private/image_view2/lib/image_view2/image_view2' に必要なターゲット '/usr/lib/libOpenNI2.so' を make するルールがありません.  中止.

しかし，このbuild.makeは冒頭に
> #CMAKE generated file: DO NOT EDIT!

と書かれていたので，根本的には問題の解決には至っていません．
そこで，このファイルを作った元を探しましたが，分かりませんでした．

shmpwk, 2020-05-24:
さらに，以上の方法を行った後は，別のエラーが発生しました．やはり，根本的な解決ができていないからだと思います．

```
catkin build -vi --make-args VERBOSE=1 2>&1 > build1.log
WARNING: Could not encode unicode characters. Please set the PYTHONIOENCODING environment variable to see complete output. (i.e. PYTHONIOENCODING=UTF-8)
[image_view2:make] /usr/bin/ld: -lOpenNI2 が見つかりません                                                                         
[image_view2:make] collect2: error: ld returned 1 exit status                                                                      
[image_view2:make] make[2]: *** [/home/shumpeiwakabayashi/coral_ws/devel/.private/image_view2/lib/image_view2/points_rectangle_extractor] Error 1
[image_view2:make] make[1]: *** [CMakeFiles/points_rectangle_extractor.dir/all] Error 2                                            
[image_view2:make] make[1]: *** 未完了のジョブを待っています....                                                                   
[image_view2:make] /usr/bin/ld: -lOpenNI2 が見つかりません                                                                         
[image_view2:make] collect2: error: ld returned 1 exit status                                                                      
[image_view2:make] make[2]: *** [/home/shumpeiwakabayashi/coral_ws/devel/.private/image_view2/lib/image_view2/image_view2] Error 1 
[image_view2:make] make[1]: *** [CMakeFiles/image_view2.dir/all] Error 2                                                           
[image_view2:make] make: *** [all] Error 2                          
```

build1.logは[こちら](https://gist.github.com/shmpwk/9e0447d2ea1009ac494ae3ecdcdea680)です．
このログでOpenNI2に関係する箇所を探したところ，
```
/opt/ros/melodic/lib/libnodeletlib.so /opt/ros/melodic/lib/libbondcpp.so -lpcl_common -lpcl_octree -lpcl_io -lboost_iostreams -lboost_serialization -lOpenNI -lOpenNI2
```
とあったので，libbondcpp.soが怪しいと推測しましたが，具体的にどう直せばよいかが分かりませんでした．

-lOpenNI つまりlibOpenNI2がインストールされていないか，またはこれのリンクがうまく貼られていないかという原因を推測しました．
しかし，以下を見てリンクのおかしい部分を発見できませんでした．
```
 ldconfig -p | grep libOpenNI2
	libOpenNI2.so.0 (libc6,x86-64) => /usr/lib/libOpenNI2.so.0
```

knorth55, 2020-05-25:
何かがおかしいですが、aptでいれると全てが解決するのでaptで入れるように心がけながら、いらないものを消していきましょう。

ワークスペースを綺麗にする（人力）
```
cd ~/coral_ws/
rm .catkin_tools/ build/ logs/ devel/ -rf
```

ワークスペースを再ビルド
```
source /opt/ros/melodic/setup.bash
cd ~/coral_ws
rosdep install --ignore-src --from-paths src -y -r
catkin init
catkin config -DPYTHON_EXECUTABLE=/usr/bin/python3 -DPYTHON_INCLUDE_DIR=/usr/include/python3.6m -DPYTHON_LIBRARY=/usr/lib/x86_64-linux-gnu/libpython3.6m.so
catkin build
```

shmpwk, 2020-05-25:
北川さん
ありがとうございます．
こちらすべてそっくりそのまま実行したところ，最初と同じエラーが出ました．
```
Errors     << image_view2:make /home/shumpeiwakabayashi/coral_ws/logs/image_view2/build.make.000.log
make[2]: *** '/home/shumpeiwakabayashi/coral_ws/devel/.private/image_view2/lib/image_view2/points_rectangle_extractor' に必要なターゲット '/usr/lib/libOpenNI2.so' を make するルールがありません.  中止.
make[2]: *** 未完了のジョブを待っています....
make[2]: *** '/home/shumpeiwakabayashi/coral_ws/devel/.private/image_view2/lib/image_view2/image_view2' に必要なターゲット '/usr/lib/libOpenNI2.so' を make するルールがありません.  中止.
make[2]: *** 未完了のジョブを待っています....
make[1]: *** [CMakeFiles/points_rectangle_extractor.dir/all] Error 2
make[1]: *** 未完了のジョブを待っています....
make[1]: *** [CMakeFiles/image_view2.dir/all] Error 2
make: *** [all] Error 2
cd /home/shumpeiwakabayashi/coral_ws/build/image_view2; catkin build --get-env image_view2 | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
```

knorth55, 2020-05-25:
先のIssueでいっていたようにOpenNI2に関して環境が壊れていますね。
これは正直パソコンを見せてもらわない限りは僕にはなんともわかりません。
岡田先生ほどリモートデバグ力がないので

shmpwk, 2020-05-25:
環境を破壊してしまったことは本当に悔やまれます．
もし学校が再開しても直っていない状態でしたらまたご相談させてください．

k-okada, 2020-05-26:
```
[image_view2:cmake] -- Could NOT find OpenNI2 (missing: OPENNI2_LIBRARY)
```
がわるそうなんだけど，以下のようになっているので，`sudo apt install --reinstall libopenni2-dev` してみてください．
```
$ apt-file search libopenni2.pc
libopenni2-dev: /usr/lib/pkgconfig/libopenni2.pc
$ cat /usr/lib/pkgconfig/libopenni2.pc
prefix=/usr
exec_prefix=${prefix}
libdir=${exec_prefix}/lib
includedir=${prefix}/include/openni2

Name: OpenNI2
Description: A general purpose driver for all OpenNI cameras.
Version: 2.2.0.3
Cflags: -I${includedir}
Libs: -L${libdir} -lOpenNI2
```
ちなみにapt-file search はapt のDBにファイルがあるか調べるもので，インストールされているか調べるにはdpkg -S かとおもいます．
```
$ dpkg -S /usr/lib/pkgconfig/libopenni2.pc
libopenni2-dev: /usr/lib/pkgconfig/libopenni2.pc
```

とおもったんだけど，そもそも`[image_view2:cmake] -- Could NOT find OpenNI2 (missing: OPENNI2_LIBRARY)` が僕の手元ではでなくて
```
[image_view2:cmake] ** WARNING ** io features related to ensenso will be disabled
[image_view2:cmake] ** WARNING ** io features related to davidSDK will be disabled
[image_view2:cmake] ** WARNING ** io features related to dssdk will be disabled
[image_view2:cmake] ** WARNING ** io features related to pcap will be disabled
[image_view2:cmake] ** WARNING ** io features related to png will be disabled
[image_view2:cmake] -- Boost version: 1.65.1
[image_view2:cmake] -- Found the following Boost libraries:
[image_view2:cmake] --   system
[image_view2:cmake] --   filesystem
[image_view2:cmake] --   thread
[image_view2:cmake] --   date_time
[image_view2:cmake] --   iostreams
[image_view2:cmake] --   serialization
[image_view2:cmake] --   chrono
[image_view2:cmake] --   atomic
[image_view2:cmake] --   regex
[image_view2:cmake] -- Could NOT find ensenso (missing: ENSENSO_LIBRARY ENSENSO_INCLUDE_DIR)
[image_view2:cmake] -- Could NOT find DAVIDSDK (missing: DAVIDSDK_LIBRARY DAVIDSDK_INCLUDE_DIR)
[image_view2:cmake] -- Could NOT find DSSDK (missing: _DSSDK_LIBRARIES)
[image_view2:cmake] -- The imported target "vtkRenderingPythonTkWidgets" references the file
[image_view2:cmake]    "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
```
なんで，もしかして，pclも自分でコンパイルしてインストールしているのかな
```
$ locate libpcl_
```
として，`/usr/local `　とかにあったら全部消して，
```
$ locate pcl_ | grep 1.8\\.pc$
```
も要チェック．`sudo apt-get remove libpcl-common1.8` みたいにしてとにかく１回消すのが確実だけど，（大量にremoveされるはずだから，必要不可欠なものが削除されないかは要チェック．こいうことをしたあと，PCが立ち上がらに人もよくいる）以下のように，とりあえずインストールされているものを再インストールしても良い．
```
dpkg --get-selections | grep libpcl | grep amd64 | cut -d\:  -f01 | xargs -n1 bash -c 'sudo apt install $0; '
```

shmpwk, 2020-05-26:
build 成功しました！
`sudo apt install --reinstall libopenni2-dev`でできました．

上記をやる前に以下を確認しました．
`$ apt-file search libopenni2.pc`と`$ cat /usr/lib/pkgconfig/libopenni2.pc`と`$ dpkg -S /usr/lib/pkgconfig/libopenni2.pc`については岡田先生と全く同じ結果が出力されました．
`$ locate libpcl_`と`$ locate pcl_ | grep 1.8\\.pc$`については，`/usr/local `以外にありました．

再インストールでなぜうまく行ったのかよくわからないですが，解決できて一安心です．
毎度ご丁寧にありがとうございました．

closed 2020-05-28 by shmpwk