jsk-ros-pkg/jsk_recognition #2509: PrimitiveShapeClassifierの使い方がわからない

Kanazawanaoaki, 2020-06-03:
jsk_pcl_rosの[PrimitiveShapeClassifier](https://jsk-docs.readthedocs.io/projects/jsk_recognition/en/latest/jsk_pcl_ros/nodes/primitive_shape_classifier.html?highlight=segmentation )を試しに使って見ようと思いましたが、ドキュメントにsampleも載っていなくて使い方がよくわかりませんでした。
[image]
画像のようなサンプルを試せるコードがあるか探して見たところ、rosbagもrvizを使っていないサンプルを見つけることは出来ました。
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/sample/sample_primitive_shape_classifier.launch

Kanazawanaoaki, 2020-06-03:
https://github.com/Kanazawanaoaki/cook_eyes/blob/master/kitchen_recognition/launch/primitive_shape_classifier_test.launch のようにlaunchファイルを作り実行すると、  
```
$ rostopic echo /primitive_shape_classifier/output
header: 
  seq: 0
  stamp: 
    secs: 1589959185
    nsecs: 342318050
  frame_id: "camera_rgb_optical_frame"
labels: [1, 1, 1, 1, 1, 1, 1, 1, 1, 1]
label_names: [circle, circle, circle, circle, circle, circle, circle, circle, circle, circle]
label_proba: [0.38947367668151855, 1.0, 0.5625, 0.9855072498321533, 0.6339285969734192, 0.6339285969734192, 0.9090909361839294, 0.9090909361839294, 0.9672130942344666, 1.0]
probabilities: []
classifier: "primitive_shape_classifier"
target_names: [box, circle, other]
---
header: 
  seq: 1
  stamp: 
    secs: 1589959187
    nsecs:  43953888
  frame_id: "camera_rgb_optical_frame"
labels: [1, 1, 1, 1, 1, 1, 1, 1]
label_names: [circle, circle, circle, circle, circle, circle, circle, circle]
label_proba: [0.31284916400909424, 1.0, 0.9428571462631226, 0.5111111402511597, 0.5714285969734192, 1.0, 1.0, 1.0]
probabilities: []
classifier: "primitive_shape_classifier"
target_names: [box, circle, other]
---
```
と出るようにはなったので認識は一応出来ていると思うのですが、認識結果をRvizでどのように可視化すれば良いのかわかりませんでした。  
メッセージの型は
```
$ rostopic type /primitive_shape_classifier/output
jsk_recognition_msgs/ClassificationResult
```
です。  
[image]

takayuki5168, 2020-06-03:
@Kanazawanaoaki RvizでAdd押してから，By display typeではなくてBy topicからだと選べたりしない？

Kanazawanaoaki, 2020-06-03:
By topicがあったんですね。  
`primitive_shape_classifier`関連のtopicは選択出来ませんでした。  
[image]
```
$ rostopic list | grep primitive_shape_classifier
/primitive_shape_classifier/debug/boundary_indices
/primitive_shape_classifier/debug/projected_cloud
/primitive_shape_classifier/output
/primitive_shape_classifier/parameter_descriptions
/primitive_shape_classifier/parameter_updates
```
存在はしているはずなのですが、可視化できないトピックなのでしょうか。

takayuki5168, 2020-06-03:
なるほど

`rostopic echo /primitive_shape_classifier/output`
の結果を見るとvisualizationのためのメッセージというよりは計算結果を羅列して出力しているように見えるから多分可視化出来ないトピックなのかな．

708yamaguchi, 2020-06-03:
自信はありませんが、金沢くんが最初にアップロードした画像の赤字付きのマーカーは以下のプログラムから出せるかも知れません。
https://github.com/jsk-ros-pkg/jsk_visualization/blob/master/jsk_rviz_plugins/scripts/classification_result_visualizer.py

試しにlaunchファイルを作ってみたので、以下の僕のコミット（ブランチ）を手元に持ってきて、以下のコマンドを試してみてください。
https://github.com/708yamaguchi/jsk_recognition/commit/dd0f38091dbd98fa0054f1f63c9b509fe0710db7
```
roslaunch jsk_pcl_ros sample_sample_primitive_shape_classifier.launch
```

実行結果
[image]

以下気づいたことです。
- 金沢くんが見つけてくれた`sample_primitive_shape_classifier.launch`ですが、sampleと頭についているものの、これは`primitive_shape_classifier.launch`という名前であったほうが分かりやすかったかもしれません。その上で、`primitive_shape_classifier.launch`やrosbag playをincludeした`sample_primitive_shape_classifier.launch`があったらより良かったと思います。
- `jsk_rviz_plugins/classification_result_visualizer.py`を使ったらいいよ、と言った手前申し訳ないですが、jsk_recognitionはjsk_rviz_pluginsに依存してはいけない方針みたいなので、僕がつくったlaunchファイルはjsk_recognitionにマージすることは出来ません。
https://github.com/jsk-ros-pkg/jsk_recognition/blob/d7294629bc39366107c264300d0934e2e5dbd613/jsk_pcl_ros/package.xml#L53

Kanazawanaoaki, 2020-06-03:
@708yamaguchi ありがとうございます。  
手元のrosbagでもマーカーを出す事が出来ました。  
[image]
なるほど`jsk_rviz_plugins/classification_result_visualizer.py`を`jsk_rviz_plugins`に依存しない形で`jsk_recognition`に持って来る必要があったから、PrimitiveShapeClassifierのサンプルが無かったんですね。

shmpwk, 2020-06-10:
山口さんのプログラムを動かしてみようとしました．
issueと直接関係なく，かなり初歩的な質問で恐縮ですが，
山口さんのプログラムで.bagファイルを置き換えれば動くものなのでしょうか．
.bagファイルは，73b2のfetchが出力していたrosbag(20200521100706_go_to_kitchen_rosbag.bag
)を使ってみました．
実行すると，rviz上は何も表示されませんでした．

$ roslaunch jsk_2020_4_carry_dish sample_sample_primitive_shape_classifier.launch 
... logging to /home/shumpeiwakabayashi/.ros/log/47e7fea6-aacb-11ea-8e3e-287fcf9948c3/roslaunch-shumpeiwakabayashi-ThinkPad-T495s-18982.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro.py is deprecated; please use xacro instead
started roslaunch server http://shumpeiwakabayashi-ThinkPad-T495s:35045/

SUMMARY
========

PARAMETERS
 * /classification_result_visualizer/text_offset_z: -0.3
 * /classification_result_visualizer/text_size: 0.05
 * /kinect_head/depth_rectify_depth/interpolation: 0
 * /kinect_head/depth_registered_rectify_depth/interpolation: 0
 * /kinect_head/disparity_depth/max_range: 4.0
 * /kinect_head/disparity_depth/min_range: 0.5
 * /kinect_head/disparity_registered_hw/max_range: 4.0
 * /kinect_head/disparity_registered_hw/min_range: 0.5
 * /kinect_head/disparity_registered_sw/max_range: 4.0
 * /kinect_head/disparity_registered_sw/min_range: 0.5
 * /kinect_head/kinect_head_nodelet_manager/num_worker_threads: 4
 * /primitive_shape_classifier/bilateral_filter/sigma_r: 0.1
 * /primitive_shape_classifier/bilateral_filter/sigma_s: 1.0
 * /primitive_shape_classifier/bilateral_filter/vital_rate: 0.1
 * /primitive_shape_classifier/euclidean_clustering/min_size: 100
 * /primitive_shape_classifier/euclidean_clustering/tolerance: 0.02
 * /primitive_shape_classifier/euclidean_clustering/vital_rate: 0.1
 * /primitive_shape_classifier/euclidean_segmentation_decomposer/align_boxes: True
 * /primitive_shape_classifier/euclidean_segmentation_decomposer/align_boxes_with_plane: True
 * /primitive_shape_classifier/euclidean_segmentation_decomposer/publish_clouds: False
 * /primitive_shape_classifier/euclidean_segmentation_decomposer/publish_tf: False
 * /primitive_shape_classifier/euclidean_segmentation_decomposer/use_pca: True
 * /primitive_shape_classifier/euclidean_segmentation_decomposer/vital_rate: 0.1
 * /primitive_shape_classifier/multi_plane_extraction/min_height: 0.03
 * /primitive_shape_classifier/multi_plane_extraction/sensor_frame: /head_mount_kinec...
 * /primitive_shape_classifier/multi_plane_extraction/use_sensor_frame: True
 * /primitive_shape_classifier/multi_plane_extraction/vital_rate: 0.1
 * /primitive_shape_classifier/organized_multi_plane_segmentation/estimate_normal: True
 * /primitive_shape_classifier/organized_multi_plane_segmentation/max_curvature: 0.01
 * /primitive_shape_classifier/organized_multi_plane_segmentation/publish_normal: True
 * /primitive_shape_classifier/organized_multi_plane_segmentation/vital_rate: 0.1
 * /primitive_shape_classifier/plane_concatenator/connect_distance_threshold: 0.2
 * /primitive_shape_classifier/plane_concatenator/min_area: 0.4
 * /primitive_shape_classifier/plane_concatenator/vital_rate: 0.1
 * /primitive_shape_classifier/primitive_shape_classifier/vital_rate: 0.1
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.5
 * /use_sim_time: True

NODES
  /kinect_head/
    depth_metric (nodelet/nodelet)
    depth_metric_rect (nodelet/nodelet)
    depth_points (nodelet/nodelet)
    depth_rectify_depth (nodelet/nodelet)
    depth_registered_hw_metric_rect (nodelet/nodelet)
    depth_registered_metric (nodelet/nodelet)
    depth_registered_rectify_depth (nodelet/nodelet)
    depth_registered_sw_metric_rect (nodelet/nodelet)
[... 59 lines cut ...]
process[kinect_head/depth_registered_metric-21]: started with pid [23004]
process[kinect_head/disparity_depth-22]: started with pid [23021]
process[kinect_head/disparity_registered_sw-23]: started with pid [23036]
process[kinect_head/disparity_registered_hw-24]: started with pid [23055]
process[primitive_shape_classifier_nodelet_manager-25]: started with pid [23069]
process[primitive_shape_classifier/input_relay-26]: started with pid [23083]
process[primitive_shape_classifier/bilateral_filter-27]: started with pid [23096]
process[primitive_shape_classifier/organized_multi_plane_segmentation-28]: started with pid [23108]
process[primitive_shape_classifier/plane_concatenator-29]: started with pid [23124]
process[primitive_shape_classifier/multi_plane_extraction-30]: started with pid [23145]
process[primitive_shape_classifier/euclidean_clustering-31]: started with pid [23157]
[ INFO] [1591760070.538018265]: Initializing nodelet with 8 worker threads.
process[primitive_shape_classifier/euclidean_segmentation_decomposer-32]: started with pid [23183]
process[primitive_shape_classifier/primitive_shape_classifier-33]: started with pid [23202]
process[classification_result_visualizer-34]: started with pid [23224]
process[rviz-35]: started with pid [23239]
[ INFO] [1591760071.134337352, 1590022807.669999972]: instantiating tf::TransformListener
[ WARN] [1591760076.121785200, 1590022812.679283217]: '/primitive_shape_classifier/multi_plane_extraction' subscribes topics only with child subscribers.
[ WARN] [1591760076.158549215, 1590022812.719545277]: '/primitive_shape_classifier/organized_multi_plane_segmentation' subscribes topics only with child subscribers.
[... cut ...]
[ WARN] [1591760076.317311092, 1590022812.880543002]: '/primitive_shape_classifier/euclidean_segmentation_decomposer' subscribes topics only with child subscribers.
[ WARN] [1591760076.347979211, 1590022812.910731664]: '/primitive_shape_classifier/primitive_shape_classifier' subscribes topics only with child subscribers.
[WARN] [1591760077.069591, 1590022813.628903]: [/classification_result_visualizer] subscribes topics only with child subscribers. Set '~always_subscribe' as True to have it subscribe always.

708yamaguchi, 2020-06-10:
若林くん、使ってくれてありがとうございます。

> issueと直接関係なく，かなり初歩的な質問で恐縮ですが，
> 山口さんのプログラムで.bagファイルを置き換えれば動くものなのでしょうか．

rosbagファイルに色付き点群のトピックが含まれていれば、rosbagファイルを入れ替えても大丈夫です。
その上で、色付き点群トピック名（`/head_camera/depth_registered/points`）とframe_id（`/head_camera_rgb_optical_frame`）を僕の`sample_sample_primitive_shape_classifier.launch`のargs（以下の`cloud_input`と`sensor_frame_id`）に入れればプログラムは動くはずです。
https://github.com/708yamaguchi/jsk_recognition/blob/dd0f38091dbd98fa0054f1f63c9b509fe0710db7/jsk_pcl_ros/sample/sample_sample_primitive_shape_classifier.launch#L7-L10

このfetchのrosbagはちょっとややこしくて、色付き点群がそのまま保存されていないため、rosbag playをするPCの中でRGB画像とDepth画像から色付き点群を復元するという作業が必要です。

そのためには、最近北川さんが作ってくれたプログラムを使うと楽です。
https://github.com/jsk-ros-pkg/jsk_robot/pull/1220
このプルリクエストを使って以下のようにrosbagを再生すれば、色付き点群の復元まで行ってくれます。
```bash
roslaunch jsk_fetch_startup rosbag_play.launch rosbag:=$HOME/Downloads/20200521100706_go_to_kitchen_rosbag.bag
```

点群をRvizで見たいときは、RvizのPointCloud2で指定しているトピック名を変更するのも忘れないようにしてください。デフォルトではPR2のrosbag用のトピック名になっているはずです。

shmpwk, 2020-06-15:
山口さんの`sample_sample_primitive_shape_classifier.launch`における，色付き点群のframe_idの指定の仕方がよくわからなかったのですが，以下の方法で分かりました．念のため情報共有をします．山口さん教えて頂きありがとうございます．

北川さんの[rosbag_play.launch](https://github.com/jsk-ros-pkg/jsk_robot/pull/1220)を走らせて，
`rostopic echo /head_camera/depth_registered/points --noarr -n1`
を実行する．(--noarrは配列部分を無視、-n1は一回だけsubscribeする)

closed 2021-09-11 by shmpwk