jsk-ros-pkg/jsk_recognition #2515: Add support for ROS neotic?

raysworld, 2020-07-10:
I am trying to use this package with Ubuntu 20.04 LTS + ROS neotic. Unfortunately I cannot find the available package to install. Would you be very kind to add support for that? Or if this has already been in progress, could you please let me know the status? Many thanks for your help.

knorth55, 2020-07-10:
@raysworld we are working on it, but before releasing jsk_recognition, we need to change all dependencies to support Ubuntu 20.04 + ROS noetic.
it would be not a easy task and may take several weeks or months.
If you want to try it now, you can maybe try this PR #2507 (but be careful, it is work in progress) and build this package from source.

JWhitleyWork, 2020-09-03:
@knorth55 Any update on this? We have had a request (https://github.com/Autoware-AI/messages/issues/4) to release Autoware.AI messages into `noetic` and they depend on `jsk_recognition_msgs` as well.

knorth55, 2020-09-03:
@JWhitleyWork we are moving forward for noetic #2507 .
we are preparing for the release.

JWhitleyWork, 2020-09-03:
Awesome, thanks!

Timple, 2020-09-22:
Now #2507 is merged, can a noetic release be made?

knorth55, 2020-09-22:
we need to wait for other dependency releases, so we cannot release it even if the test build succeeded.

closed 2022-06-13 by raysworld

ttezy, 2023-04-06:
Hi @knorth55 @raysworld , is there any update for Noetic support? I found this Issue is closed but I cannot find any instruction for Autoware.ai Noetic version. May I know if it is possible to build Autoware.ai on Ubuntu 20.04 + ROS Noetic?

knorth55, 2023-04-06:
Im not sure about autoware, but
jsk recognition is already released in noetic

https://github.com/ros/rosdistro/pull/26895