jsk-ros-pkg/jsk_recognition #2516: How to define the input topic for border_estimator

Ahmad-Drak, 2020-07-19:
How can i specify the input topic when launching  border_estimate.launch? [launch file](https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/launch/border_estimate.launch)

I tried `roslaunch jsk_pcl_ros border_estimate.launch input:=/camera/depth/color/points` but that wont work as its not an arg.

As seen it complains from not subscribing to a topic

> roslaunch jsk_pcl_ros border_estimate.launch 
> ... logging to /home/ahmad/.ros/log/66feb130-c9ee-11ea-90a7-54271e90ab2d/roslaunch-ahmad-stryker-7600.log
> Checking log directory for disk usage. This may take a while.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> started roslaunch server http://ahmad-stryker:35603/
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /border_estimator/model_type: laser
>  * /laser_cloud_from_camera_frame/target_frame_id: multisense/left_c...
>  * /rosdistro: melodic
>  * /rosversion: 1.14.6
> 
> NODES
>   /
>     border_estimator (jsk_pcl_ros/border_estimator)
>     border_points (nodelet/nodelet)
>     laser_cloud_from_camera_frame (jsk_pcl_ros/tf_transform_cloud)
>     shadow_points (nodelet/nodelet)
>     veil_points (nodelet/nodelet)
> 
> ROS_MASTER_URI=http://localhost:11311
> 
> process[laser_cloud_from_camera_frame-1]: started with pid [7615]
> process[border_estimator-2]: started with pid [7616]
> process[border_points-3]: started with pid [7617]
> process[veil_points-4]: started with pid [7618]
> process[shadow_points-5]: started with pid [7619]
> [ INFO] [1595186444.616147390]: Initializing nodelet with 12 worker threads.
> [ WARN] [1595186449.693401136, 1594809778.053502148]: This node/nodelet subscribes topics only when subscribed.
> [ WARN] [1595186449.711033648, 1594809778.053502148]: This node/nodelet subscribes topics only when subscribed.
> [ WARN] [1595186449.719633865, 1594809778.053502148]: This node/nodelet subscribes topics only when subscribed.
> [ WARN] [1595186449.907160615, 1594809778.053502148]: '/laser_cloud_from_camera_frame' subscribes topics only with child subscribers.
> [ WARN] [1595186449.963293686, 1594809778.053502148]: '/border_estimator' subscribes topics only with child subscribers.
>

knorth55, 2020-07-19:
There is a sample launch, so please try it.
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/sample/sample_border_estimator.launch

For more information, please read the doc.
https://jsk-docs.readthedocs.io/projects/jsk_recognition/en/latest/jsk_pcl_ros/nodes/border_estimator.html

Ahmad-Drak, 2020-07-20:
thanks @knorth55 . I installed the pkg via ros-melodic-jsk_pcl_ros. The samples there  seem incomplete (border_estimator isnt even there). Nonetheless thank you

closed 2020-07-20 by Ahmad-Drak