jsk-ros-pkg/jsk_recognition #2517: Installation / Building kinfu

jdeeee, 2020-07-22:
Hi!
I am having trouble installing the package jsk_pcl_ros; I have installed the package using "sudo apt-get install ros-kinetic-jsk-pcl-ros" and it seems I can use some of the functionality coming with the package (e.g. the "bounding_box_array_publisher"). However, I can not run the "kinfu" executable. This might not be surprising, since I can't find a corresponding executable. Also, I can only find one header file named "kinfu.h" in /opt/ros/kinetic/include/jsk_pcl_ros, but no matching cpp file. I am assuming I need to build parts of jsk_pcl_ros myself, but I do not really know how to. Unfortunately, I was unable to find a proper installation documentation that could help me. What steps do I need to execute?
I am using Ubuntu 16.04 and ROS Kinetic.
Thank you!

knorth55, 2020-07-23:
Kinfu requires CUDA enables PCL, so you need to build PCL with CUDA enabled flag first and build jsk_recognition from source.
You can find how to do that in the doc.
https://jsk-recognition.readthedocs.io/en/latest/jsk_pcl_ros/nodes/kinfu.html

jdeeee, 2020-07-23:
Okay, yes, thank you :) But I am still unsure about the step _"And rebuild all PCL related packges from source"._ 
Probably I am only required to rebuild packages that use PCL and either use KINFU or are used by KINFU? So it might be sufficient to build PCL with CUDA, download the code from https://github.com/jsk-ros-pkg/jsk_recognition and then build jsk_recognition myself?

knorth55, 2020-07-23:
Yes