jsk-ros-pkg/jsk_recognition #2529: DepthImageCreator output pointcloud doesn't appeared and is unorganized

shmpwk, 2020-09-24:
[DepthImageCreator](https://jsk-docs.readthedocs.io/projects/jsk_recognition/en/latest/jsk_pcl_ros/nodes/depth_image_creator.html?highlight=depth%20image%20creater)を使ってみたところ、
output_cloudで出力されるはずのpointcloudがtopicとしては出力されているものの、rviz上に現れませんでした。また、parameterのorganize_cloudをtrueと指定したものの、topicでheightが1であると確認されたので、unorganizedなpointcloudが出力されています。

[image]

```

$ rostopic echo camera/depth_registered/points 
(次の出力は一部だけ載せています。)
55, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 87, 67, 58, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 17, 0, 0, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 0, 0, 0, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 0, 0, 0, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 4, 0, 0, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 5, 0, 0, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 13, 0, 0, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 12, 0, 0, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 3, 0, 0, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 5, 0, 0, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 18, 0, 0, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 16, 0, 0, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 4, 0, 0, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 4, 0, 0, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 0, 0, 0, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 0, 0, 0, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 0, 0, 0, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 0, 0, 0, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 3, 0, 0, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 5, 0, 0, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 0, 0, 0, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 0, 0, 0, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 0, 0, 0, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 0, 0, 0, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 5, 0, 0, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 5, 0, 0, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 0, 0, 0, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 0, 0, 0, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
```

```
$ rostopic echo camera/depth_registered/points --noarr
header: 
  seq: 42355
  stamp: 
    secs: 1600952074
    nsecs: 360056870
  frame_id: "camera_rgb_optical_frame"
height: 480
width: 640
fields: "<array type: sensor_msgs/PointField, length: 4>"
is_bigendian: False
point_step: 32
row_step: 20480
data: "<array type: uint8, length: 9830400>"
is_dense: False
---
```
[DepthImageCreator](https://jsk-docs.readthedocs.io/projects/jsk_recognition/en/latest/jsk_pcl_ros/nodes/depth_image_creator.html?highlight=depth%20image%20creater)のサンプルをそのまま[こちら](https://github.com/shmpwk/grasp_recognition/blob/master/launch/depth_image_creator.launch)に書きました。
xtionのカメラを立ち上げて(`roslaunch openni2_launch openni.launch`)から、上記をlaunchします。

※このissueは僕の現在の作業に支障が出ないので、自分にとっては急いで解決する必要はありません。

shmpwk, 2020-09-24:
同じ問題を[#4](https://github.com/shmpwk/jsk_2020_04_carry_dish/issues/4)のissueでも触れています。
こちらは、gazeboシミュレーションと実機のpr1040を使った場合です。

shmpwk, 2020-09-24:
念の為
````
$ rostopic echo /camera/depth_registered/points --noarr
header: 
  seq: 67630
  stamp: 
    secs: 1600952921
    nsecs: 470426790
  frame_id: "camera_rgb_optical_frame"
height: 480
width: 640
fields: "<array type: sensor_msgs/PointField, length: 4>"
is_bigendian: False
point_step: 32
row_step: 20480
data: "<array type: uint8, length: 9830400>"
is_dense: False
```

```
$ rostopic echo /camera/rgb/camera_info 
header: 
  seq: 69071
  stamp: 
    secs: 1600952969
    nsecs: 617541544
  frame_id: "camera_rgb_optical_frame"
height: 480
width: 640
distortion_model: "plumb_bob"
D: [0.0, 0.0, 0.0, 0.0, 0.0]
K: [570.3422241210938, 0.0, 319.5, 0.0, 570.3422241210938, 239.5, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [570.3422241210938, 0.0, 319.5, 0.0, 0.0, 570.3422241210938, 239.5, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
---
```

knorth55, 2020-09-24:
can you paste the output of `rostopic hz /depth_image_creator/output_cloud` and `rostopic echo --noarr /depth_image_creator/output_cloud`?

shmpwk, 2020-09-27:
Here is an example of output
```
$ rostopic hz /depth_image_creator/output_cloud
subscribed to [/depth_image_creator/output_cloud]
no new messages
no new messages
average rate: 1.867
	min: 0.067s max: 1.375s std dev: 0.50991s window: 5
average rate: 7.628
	min: 0.032s max: 1.375s std dev: 0.27614s window: 25
average rate: 12.261
	min: 0.026s max: 1.375s std dev: 0.19517s window: 52
```

```
$ rostopic echo --noarr /depth_image_creator/output_cloud
header: 
  seq: 669
  stamp: 
    secs: 1601193070
    nsecs: 721627000
  frame_id: "camera_rgb_optical_frame"
height: 1
width: 307200
fields: "<array type: sensor_msgs/PointField, length: 4>"
is_bigendian: False
point_step: 32
row_step: 9830400
data: "<array type: uint8, length: 9830400>"
is_dense: True
---
header: 
  seq: 700
  stamp: 
    secs: 1601193071
    nsecs: 760525000
  frame_id: "camera_rgb_optical_frame"
height: 1
width: 307200
fields: "<array type: sensor_msgs/PointField, length: 4>"
is_bigendian: False
point_step: 32
row_step: 9830400
data: "<array type: uint8, length: 9830400>"
is_dense: True
---

````

k-okada, 2020-11-12:
試してみたけど，普通に動いている様に見えますが，，，
 `openni.launch`がなく，`roslaunch openni2_launch openni2.launch`と，https://github.com/shmpwk/grasp_recognition/blob/a791f4cdf64a4ce9275327b878b490c91a008d05/launch/depth_image_creator.launch を起ち上げて，以下のように確認しましたが，これで問題ないのではないでしょうか？
```
$ rostopic echo -n1 /camera/depth_registered/points/height
480
---
$ rostopic echo -n1  /depth_image_creator/output_cloud/height
480
---
```

というかそもそも，organizeかそうでないかにかかわらず，rvizには表示されるのでは？と思うんだけど違うでしょうか？あるいは，paramがセットされていなかったか．そういう勘違いがないように，https://github.com/jsk-ros-pkg/jsk_recognition/pull/2543 をつくりました．

shmpwk, 2020-11-12:
再度やってみたら，正常でした，，，

```
$ rostopic echo -n1 /camera/depth_registered/points/height
480
```
でしたし，
rviz上に/camera/depth_image_creator/output_cloudのpointcloud2も表示されました．

このissueを立ててから今までに，jsk_recognitionをbuildし直すなどしたのが，バグ解消に繋がったのかもしません．

k-okada, 2020-11-12:
リリースした物でも動いているはずなので，
source /opt/ros/melodic/setup.bash
でも確認してみてください．

shmpwk, 2020-11-13:
リリースされたものも正常でした．
rosparamでorganize_cloud: Trueとしたら，
```
$ rostopic echo -n1  /depth_image_creator/output_cloud/height
480
```

closed 2022-01-11 by shmpwk