jsk-ros-pkg/jsk_recognition #2547: multi_plane_sac_segmentationでの平面の認識がうまく行かない

Utaro-M, 2020-11-17:
以下のrosbagに対して、動いている板の平面の認識しようとしているのですが、動いている板は認識してくれずに関係のないところに平面がたくさん出てしまいます。
organizedのpointcloudに対し、organized_multi_plane_segmenatationを試したところ、板と大域平面を認識できています。
今回はunorganizedのpoint cloudに対して同様の認識をしたいと思っています。
パラメタはデフォルトのままで以下のlaunchファイルを実行しています。
[image]

rosbagファイル　https://drive.google.com/file/d/1u99347bS9DeI-9lME4jSsHik8XFATVgJ/view?usp=sharing
launchファイル　https://github.com/Utaro-M/rm_debris_vision/blob/master/launch/multi_plane_fail.launch
organized_multi_plane_segmentationを使った場合の動画https://drive.google.com/file/d/1gD7_bB5xDeLK3q5U48kjCXcwlE3wwlHN/view?usp=sharing

iory, 2020-11-17:
https://gist.github.com/iory/2ed329d430bb1587022cfd3799369f9b
https://gist.github.com/iory/2ed329d430bb1587022cfd3799369f9b
上のlaunchファイルとrvizを開くいてrosbagをやると添付の動画のように平面抽出できます。
表示されているbounding boxの中の点群だけを見て平面検出するようになっているので、bounding boxを変えると見たい部分にfocusするように変えられるので適宜変えてみてください。

https://drive.google.com/file/d/1egRjo2SHW5kXU2odzAgkXYfxy7YfGwto/view?usp=sharing

Utaro-M, 2020-11-18:
ありがとうございます。
自分で試していた時よりも平面をうまく絞り込むことができるようになりました。

Utaro-M, 2020-11-18:
ありがとうございます。
試してみたところ、自分で使っていたときよりもうまく平面を取れるようになりました。
これと同様のことを、unorganizedの点群に対してもできればと思うのですが難しいのでしょうか？
今回は、depth_image_creatorを使ってカメラフレーム以外の固定フレームからみた点群に対して動いている板の平面認識をしたいと思っていて、フレームの変更をして出力したorganize point cloudに対してorganized_multi_plane_segmentationを使ったところnanなどが含まれているせいなのか平面認識ができませんでした。
その解決のために、depth_image_creatorを通してunorganized pointcloudを出力して、それを用いて平面の認識ができればと思い、multi_plane_sac_segmentation使おうとしたところうまくいきませんでした。
今回のように平面を認識する場合には、pointcloudを工夫して、organizedのpoint cloudを使うorganized_multi_plane_segmentationを使ったほうが有利なのでしょうか？

iory, 2020-11-18:
いまいちゴールが分かっていなんだけど、
ゴールはカメラフレーム以外の固定フレームから見た座標系で動いている平面の頂点を出力したいということなんだろうか？

knorth55, 2020-11-18:
よくわかってないですが，２つのLaunchかなりdiffがありますが，結局何が問題だったのでしょうか？

@Utaro-M 
ここにsampleとして使っているLaunchをPRで出してください．
今後みなが共有して使えるようになると，それが便利かと思います．
https://github.com/jsk-ros-pkg/jsk_recognition/tree/master/jsk_pcl_ros/sample

二足歩行，ベースロボット，カメラさえあればいい，色々レイヤーがあると思いますが，カメラさえあればいい話っぽいので，欲をいうとrosbagとあわせてぱっと誰でも試せるようにしてくれると嬉しいです．

knorth55, 2020-11-18:
@Utaro-M 
ちなみにですが，ここらへんではダメだったのでしょうか？
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/launch/laser_multi_plane_detection.launch
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/launch/laser_multi_plane_segmentation.launch
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/launch/organized_multi_plane_segmentation.launch

個人的には`organize_multi_plane_segmentation.launch`にならって`unorganized_multi_plane_segmentation.launch`をつくってくれるとわかりやすいです．

Utaro-M, 2020-11-18:
今回、このissueの当初の問題はmulti_plane_sac_segmentationでの平面の認識がうまくいかないことだったのですが、同様のことをorganized_multi_plane_segmentationでより上手くできるということがわかったのでそちらのノードを使って進めようと思いました。
代わりの方法は見つかったのですが当初のmulti_plane_sac_segmentationを使った認識が上手くいかない問題についてはまだ解決はできていません。

knorth55, 2020-11-18:
@Utaro-M 
> organized_multi_plane_segmentationでより上手くできるということがわかったのでそちらのノードを使って進めようと思いました。
代わりの方法は見つかったのですが当初のmulti_plane_sac_segmentationを使った認識が上手くいかない問題についてはまだ解決はできていません。

つかったlaunchはこれですか？
もしも違ったり変更があるならPRをだしてほしいです，情報共有のために
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/launch/organized_multi_plane_segmentation.launch

Utaro-M, 2020-11-19:
> https://gist.github.com/iory/2ed329d430bb1587022cfd3799369f9b
https://gist.github.com/iory/0d2d6726656bfb00cd41aacebdbc7c1a
上のlaunchファイルとrvizを開くいてrosbagをやると添付の動画のように平面抽出できます。

を少しだけ改良したもののPRを出しておきました。確認お願いします。
また、ファイルの中でbagファイルを呼ぶようにしているのですが、bagファイルはどこへ保存すればよいか教えていただけますか？

knorth55, 2020-11-19:
@Utaro-M 

rosbagはまずgoogle driveの共有フォルダにあげます．
https://drive.google.com/drive/folders/0B9P1L--7Wd2vUGplQkVLTFBWcFE
JSKのひとならアップロードできるかと思います．

次にgoogleのURLからサンプルデータをダウンロードするスクリプトを編集します．
`jsk_pcl_ros/scripts/install_sample_data.py`にあります．
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/scripts/install_sample_data.py

こんな感じで書きます．
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/scripts/install_sample_data.py#L37-L42
ファイルのmd5は`md5sum`コマンドでえることができます．
Google DriveのURLは共有から得られますが，`https://drive.google.com/uc?id=<id>`に変更しなければならないのが注意点です．

そうすれば，`rosrun jsk_pcl_ros install_sample_data.py`でサンプルデータが一括でダウンロードできるようになります．