jsk-ros-pkg/jsk_recognition #2552: depth_image_creatorの実機での使い方がわからない

ishiguroJSK, 2020-11-20:
Gazebo環境で使っているときは以下のように使えていたが，
PR1040実機で使えなくて困っています．
```
(略)
  <node name="depth_image_creator" pkg="nodelet" type="nodelet" args="load jsk_pcl/DepthImageCreator sample_manager">
    <remap from="~input" to="/cluster_point_indices_decomposer/debug_output" />
    <remap from="~info" to="/$(arg device)/rgb/camera_info" />
    <rosparam>                                                                                                                                            
      use_approximate: true                                                                                                                               
      max_queue_size: 3                                                                                                                                   
      max_pub_queue_size: 1                                                                                                                               
      max_sub_queue_size: 1                                                                                                                               
    </rosparam>
  </node>
```
デバッグ1
```
<launch>
  <node name="depth_image_creator"
        pkg="nodelet" type="nodelet"
        args="standalone jsk_pcl/DepthImageCreator" output="screen">
    <remap from="~input" to="/kinect_head/depth_registered/points" />
    <remap from="~info" to="/kinect_head/depth_registered/camera_info" />
    <rosparam>
      use_approximate: true
      max_queue_size: 3
      max_pub_queue_size: 1
      max_sub_queue_size: 1
    </rosparam>
  </node>
</launch>
```
から
`rostopic echo /depth_image_creator/output_image`→でない，

デバッグ２
`rosrun jsk_pcl_ros depth_image_creator ~input:=/kinect_head/depth_registered/points ~info:=/kinect_head/depth_registered/camera_info`
から
`rostopic echo /JSK_NODELET_jsk_pcl_ros_depth_image_creator/output_image` →でない

knorth55, 2020-11-20:
message_filterのsynchronizeができていないとうにおもえます．
`jsk_topic_tools`の`is_synchronized`が確認には便利だと思います．

```
$ rosrun jsk_topic_tools is_synchronized --help
usage: is_synchronized [-h] [-t TIMEOUT] [-q QUEUE_SIZE] [-a] [--slop SLOP]
                       topics [topics ...]

positional arguments:
  topics                topics which should be synchronized

optional arguments:
  -h, --help            show this help message and exit
  -t TIMEOUT, --timeout TIMEOUT
                        Timeout for the test of synchronization
  -q QUEUE_SIZE, --queue-size QUEUE_SIZE
                        Size of queue for the synchronization
  -a, --approximate-sync
                        Flag to use approximate synchronization
  --slop SLOP           Allowed time delta in approximate synchronization
```

ちなみに今試してみましたが，`queue_size=3`では以下のように`not synchronized`が返ってきます．

```
$ rosrun jsk_topic_tools is_synchronized --approximate-sync --queue-size 3 /kinect_head/depth_registered/camera
_info /kinect_head/depth_registered/points
****************** Test Condition ******************
sync_policy: Approximate
topics:
  - /kinect_head/depth_registered/camera_info [sensor_msgs/CameraInfo]
  - /kinect_head/depth_registered/points [sensor_msgs/PointCloud2]
timeout: 5 seconds
queue_size: 3
slop: 0.1
****************************************************
listening these topics for at most 5 seconds in rostime
timeout for 5 seconds
these topics are: 'not synchronized'
```

これを適当に`queue_size=100`くらいにすると`synchronized`になります．
```bash
$ rosrun jsk_topic_tools is_synchronized --approximate-sync --queue-size 100 /kinect_head/depth_registered/came
ra_info /kinect_head/depth_registered/points
****************** Test Condition ******************
sync_policy: Approximate
topics:
  - /kinect_head/depth_registered/camera_info [sensor_msgs/CameraInfo]
  - /kinect_head/depth_registered/points [sensor_msgs/PointCloud2]
timeout: 5 seconds
queue_size: 100
slop: 0.1
****************************************************
listening these topics for at most 5 seconds in rostime
synchronized in 0.5 seconds
these topics are: 'synchronized'
```

なので，`max_queue_size`を高めに設定すると解決すると思います．

ishiguroJSK, 2020-11-24:
なるほど，ありがとう．うまく行きました．

色々試して理解できた状況としては，ロボット体内ならcamera_infoもpointsも30Hzでqueue_size=3でもsynchronizedだけど，
ロボット体外とか，あるいはクラスタリング処理とかしてcamera_info(30Hz) vs points(1Hz) とかになると，queue_size=3ではあっという間にnot synchronizedになるわけだね．

closed 2020-11-24 by ishiguroJSK

knorth55, 2020-11-24:
そうですね，camera_infoとpointsは基本的に同じtimestampをもっているトピックがでますが，処理がはいるとlatencyが発生したり，周期もおちてしまうので，queue_sizeを大きくしないとsyncはできなくなってしまいます．
queue_size分しか比較してくれないので，1Hzと30Hzの場合だと，Timestampが同じpointとcamera_infoがあったとしても，queue_sizeは最低でも30は必要になり，latencyを考えるともっと大きな100くらいをセットするということになります．

k-okada, 2020-11-26:
@ishiguroJSK  折角よい情報を教えてもらったので以下で注意点（＋デバッグ方法）を書き加えましょう．
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/doc/jsk_pcl_ros/nodes/depth_image_creator.md

あと，デフォルトを100にしてもよいかもしれないです．