jsk-ros-pkg/jsk_recognition #2578: euclidean_segmentation.launch : failed to run

gachiemchiep, 2021-03-05:
Hello. 
Thank you for your team's great work. 

I tried the sample at [euclidean_segmentation_sample.launch](https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/sample/euclidean_segmentation_sample.launch). 
I edited my launch file as follow so I can run the sample with Intel realsense

```bash
<launch>
  <include file="$(find realsense2_camera)/launch/rs_rgbd.launch">
    <arg name="align_depth" value="true"/>
  </include>

  <include file="$(find object_analytics_nodelet)/launch/euclidean_segmentation.launch">
    <arg name="input_point" value="/camera/depth_registered/points"/>
  </include>

  <node pkg="rviz" type="rviz" name="rviz" args=" -d $(find jsk_pcl_ros)/sample/rviz/euclidean_segmentation.rviz"/>
```

Unfortunately, the segmented are unable to segment the input pointcloud. 

[image]

I also tried to change ClusterPointIndicesDecomposerZAxis node into ClusterPointIndicesDecomposer node [euclidean_segmentation_sample.launch](https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/launch/euclidean_segmentation.launch). But it didn't help. 

Would you mind help me solve this?
Thank you.

k-okada, 2021-03-05:
hi, can you provide rosbag files so that we can reproduce your problem?

gachiemchiep, 2021-03-09:
@k-okada 

Hello sorry for the late response. 
Here is the bag file : 
https://drive.google.com/file/d/1aFWRphwv8JjnbVAL8DjFbTCmObILV8JG/view?usp=sharing

I run the euclidean_segmentation_sample.launch. then record all topics. 
So I think this file will give you enough information .

k-okada, 2021-03-09:
oh, we prefer bagfile that only contains input
of  euclidean_segmentation_sample, we will
rosbag play yourfile.bag
roslaunch euclidean_segmentation_sample.launch
and see how it goes.
--
◉ Kei Okada

closed 2021-03-12 by gachiemchiep