jsk-ros-pkg/jsk_recognition #2587: How to properly install octree voxel grid

malith1992, 2021-04-20:
I want to run my realsense camera with octree voxel grid. But I get this error

> [octree_voxel_grid.launch] is neither a launch file in package [jsk_pcl_ros] nor is [jsk_pcl_ros] a launch file name

Seems like I have an installation error.  Please help me to make this work.

knorth55, 2021-04-20:
can you tell us how you installed jsk_pcl_ros and which version of jsk_pcl_ros you are using?

knorth55, 2021-04-20:
can you tell me how you installed jsk_pcl_ros and which version of jsk_pcl_ros you are using?

malith1992, 2021-04-20:
> can you tell us how you installed jsk_pcl_ros and which version of jsk_pcl_ros you are using?

I used following commands

> wget https://github.com/PointCloudLibrary/pcl/archive/pcl-1.8.1.tar.gz
> tar xvzf pcl-1.8.1.tar.gz
> cd pcl-pcl-1.8.1
> mkdir build
> cd build
> cmake .. -DCMAKE_BUILD_TYPE=Release -DPCL_ENABLE_SSE:BOOL=FALSE
> make -j 4
> sudo make install

knorth55, 2021-04-20:
that is how you install PCL not jsk_pcl_ros.
can you paste output of `rospack find jsk_pcl_ros`?

malith1992, 2021-04-20:
> rospack find jsk_pcl_ros

This is the output

> [rospack] Error: package 'jsk_pcl_ros' not found

It seems like I installed only PCL but not  jsk_pcl_ros. Please tell me how to install  jsk_pcl_ros.

knorth55, 2021-04-20:
you need to install ros first, if you don't install ros
and then run `sudo apt install ros-$ROS_DISTRO-jsk-pcl-ros`.
if you are not familiar with ros, it is better to learn it first.

malith1992, 2021-04-21:
> you need to install ros first, if you don't install ros
> and then run `sudo apt install ros-$ROS_DISTRO-jsk-pcl-ros`.
> if you are not familiar with ros, it is better to learn it first.

Thank you for the help. I have installed jsk-pcl-ros using the command you gave me. Now I get the following output for the command "roslaunch jsk_pcl_ros octree_voxel_grid.launch"

> ... logging to /home/nvidia/.ros/log/5c485f30-a264-11eb-81cd-06d6c33466cd/roslaunch-nvidia-desktop-21543.log
> Checking log directory for disk usage. This may take a while.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> started roslaunch server http://nvidia-desktop:37171/
> 
> SUMMARY
> 
> CLEAR PARAMETERS
>  * /octree_voxel_grid/octgrid/
> 
> PARAMETERS
>  * /octree_voxel_grid/octgrid/publish_marker: True
>  * /rosdistro: melodic
>  * /rosversion: 1.14.10
> 
> NODES
>   /octree_voxel_grid/
>     octgrid (nodelet/nodelet)
>     octree_voxel_grid_manger (nodelet/nodelet)
> 
> ROS_MASTER_URI=http://localhost:11311
> 
> process[octree_voxel_grid/octree_voxel_grid_manger-1]: started with pid [21559]
> process[octree_voxel_grid/octgrid-2]: started with pid [21560]
> type is jsk_pcl/OctreeVoxelGrid
> [ INFO] [1618985883.900222334]: Initializing nodelet with 4 worker threads.
> [ WARN] [1618985893.237390080]: '/octree_voxel_grid/octgrid' subscribes topics only with child subscribers.

I tried to subscribe to the topics using rviz. But I can't see the voxel grided cloud from point cloud as in this link https://jsk-docs.readthedocs.io/projects/jsk_recognition/en/latest/jsk_pcl_ros/nodes/octree_voxel_grid.html

This is my rviz output. It shows nothing. 

[image]

Do I need to change any parameters in rviz ?

knorth55, 2021-04-21:
please change the frame id of rviz.

malith1992, 2021-04-21:
> please change the frame id of rviz.

what should be the frame id ? It's currently map type

knorth55, 2021-04-21:
it should be same frame id as the voxel grid topic.

malith1992, 2021-04-21:
> it should be same frame id as the voxel grid topic.

voxel grid topic name is "/octree_voxel_grid/octgrid/output" . I added it to the fixed frame, but still nothing.

[image]

knorth55, 2021-04-21:
i mean the frame_id field in the topic. you need to select it from the rviz menu.

malith1992, 2021-04-26:
> i mean the frame_id field in the topic. you need to select it from the rviz menu.

There is no menu in rviz. And there is no field named as `frame_id`