jsk-ros-pkg/jsk_recognition #2594: With cluster_point_indices_decomposer, a few tf of segmentation_decomposeroutput appear in the reference frame instead of the target frame.

shmpwk, 2021-05-06:
The [cluster_point_indices_decomposer](https://jsk-docs.readthedocs.io/projects/jsk_recognition/en/latest/jsk_pcl_ros/nodes/cluster_point_indices_decomposer.html) node publishes tf of each segmentated  bounding box when the parameter of publish_tf is true.
It is expected to show target frame of the bounding box.
The photo shows the each tf of segmentation_decomposeroutput. 
As you can see the photo, a few tf unexpectedly show on the robot head where I set as the reference frame. 

[image]

shmpwk, 2021-05-06:
My temporary solution is to change [here](https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/src/cluster_point_indices_decomposer_nodelet.cpp#L432) from true to false. 
It blocks publishing topics including tf when segmented cloud size is 0.

knorth55, 2021-05-06:
can you upload your rosbag file and launch file to reproduce this error?
I think you got the core idea of this error `segmented cloud size is zero`.
You can make a PR with your solution and discuss it.
Your solution looks good to me, so please make a PR.

knorth55, 2021-05-06:
however, your solution will change the order of output bounding boxes.
I think it is better to skip the tf publication section only.

shmpwk, 2021-05-06:
> I think it is better to skip the tf publication section only.
Yes, I will.

shmpwk, 2021-07-15:
Close with #2599

closed 2021-07-15 by shmpwk