jsk-ros-pkg/jsk_recognition #2595: [jsk_perception] Broken links in install_trained_data.py

pazeshun, 2021-05-11:
When I built jsk_perception, I faced `Permission denied` errors of GoogleDrive links as follows:
```
$ catkin build jsk_perception --no-deps
---------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/melodic
Workspace:                   /home/ushun/ros/ws_jsk_apc
---------------------------------------------------------------
Build Space:        [exists] /home/ushun/ros/ws_jsk_apc/build
Devel Space:        [exists] /home/ushun/ros/ws_jsk_apc/devel
Install Space:      [unused] /home/ushun/ros/ws_jsk_apc/install
Log Space:          [exists] /home/ushun/ros/ws_jsk_apc/logs
Source Space:       [exists] /home/ushun/ros/ws_jsk_apc/src
DESTDIR:            [unused] None
---------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
---------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
---------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
---------------------------------------------------------------
Workspace configuration appears valid.
---------------------------------------------------------------
[build] Found '103' packages in 0.0 seconds.                                   
[build] Package table is up to date.                                           
Starting  >>> jsk_perception                                                   
_______________________________________________________________________________
Warnings   << jsk_perception:make /home/ushun/ros/ws_jsk_apc/logs/jsk_perception/build.make.005.log
Please install CuPy

    sudo pip install cupy-cuda[your cuda version]
i.e.
    sudo pip install cupy-cuda91

/usr/lib/python2.7/dist-packages/h5py/__init__.py:36: FutureWarning: Conversion of the second argument of issubdtype from `float` to `np.floating` is deprecated. In future, it will be treated as `np.float64 == np.dtype(float).type`.
  from ._conv import register_converters as _register_converters
[/home/ushun/ros/ws_jsk_apc/src/jsk-ros-pkg/jsk_recognition/jsk_perception/sample/data/2016-11-11-12-53-06_in_lab.bag] File exists or not writable, so skipping creating symlink.
[/home/ushun/ros/ws_jsk_apc/src/jsk-ros-pkg/jsk_recognition/jsk_perception/sample/data/2016-10-15-23-21-42_moving_bottle.bag] File exists or not writable, so skipping creating symlink.
Permission denied: https://drive.google.com/uc?id=0B9P1L--7Wd2vTDV3ZzUyTlBFZE0
Maybe you need to change permission over 'Anyone with the link'?
Permission denied: https://drive.google.com/uc?id=0B9P1L--7Wd2vQVBodlFsMnpGbkU
Maybe you need to change permission over 'Anyone with the link'?
Permission denied: https://drive.google.com/uc?id=0B9P1L--7Wd2vWG5MeUEwWmxudU0
Maybe you need to change permission over 'Anyone with the link'?
Permission denied: https://drive.google.com/uc?id=0B9P1L--7Wd2vQVBodlFsMnpGbkU
Maybe you need to change permission over 'Anyone with the link'?
[/home/ushun/.ros/data/jsk_perception/resnet152_from_caffe.npz] could not download file. maximum download trial exceeded.
Permission denied: https://drive.google.com/uc?id=0B9P1L--7Wd2vTDV3ZzUyTlBFZE0
Maybe you need to change permission over 'Anyone with the link'?
[/home/ushun/.ros/data/jsk_perception/resnet_lsvrc2012_mean.npy] could not download file. maximum download trial exceeded.
Permission denied: https://drive.google.com/uc?id=0B9P1L--7Wd2vWG5MeUEwWmxudU0
Maybe you need to change permission over 'Anyone with the link'?
[/home/ushun/.ros/data/jsk_perception/fcn8s_voc.npz] could not download file. maximum download trial exceeded.
cd /home/ushun/ros/ws_jsk_apc/build/jsk_perception; catkin build --get-env jsk_perception | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Finished  <<< jsk_perception                [ 22.3 seconds ]                   
[build] Summary: All 1 packages succeeded!                                     
[build]   Ignored:   102 packages were skipped or are blacklisted.             
[build]   Warnings:  1 packages succeeded with warnings.                       
[build]   Abandoned: None.                                                     
[build]   Failed:    None.                                                     
[build] Runtime: 22.6 seconds total.
```

I tried to access these links from a web browser, but the browser returned `404 Not Found` errors even when I logged in with my JSK account.
When I searched in jsk_perception, I found that these broken links are listed in [install_trained_data.py in jsk_perception](https://github.com/jsk-ros-pkg/jsk_recognition/blob/2bc0fab0be749e419fd668f097464e6f02a35e91/jsk_perception/scripts/install_trained_data.py) and were added by @wkentaro :
https://github.com/jsk-ros-pkg/jsk_recognition/commit/c9a7944dd3ad759291f9133d666e9c83b651e6a3
https://github.com/jsk-ros-pkg/jsk_recognition/commit/19d7a2ac09bab2b470a8b06e0ed98d072b4958d4

@wkentaro Do you know anything about where linked files are?
If you have these files, could you send me them?
I'll upload them to a common space and fix the links.

knorth55, 2021-06-19:
@wkentaro ping.

wkentaro, 2021-06-19:
Oh sorry, I overlooked this.
I’ll have a look in a week.

closed 2021-07-10 by knorth55