jsk-ros-pkg/jsk_recognition #2618: Coral TPU USBで認識したものをdeep sort trackerなどでトラッキングしたい．

knorth55, 2021-08-16:
@a-ichikura さんからの質問．

Coral TPU USBのSSDで物体認識は出来たのですが，そこからペン1，ペン2としてトラッキングしたいです．

@sktometometo はどのようにやっていますか？
launchファイルなどのURLなどを共有してもらえると助かります．

sktometometo, 2021-08-16:
@knorth55 @a-ichikura

Spot で Multi Object Tracking をしたい時、今は jsk_perception の [DeepSortノード](https://jsk-docs.readthedocs.io/projects/jsk_recognition/en/latest/jsk_perception/nodes/deep_sort_tracker.html) を使っています。

このノードはObjectDetectionノードと組み合わせて使うもので、[jsk_perceptionのサンプル](https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_perception/sample/sample_deep_sort_tracker.launch)では、[CPUorGPU上で動くSSDノードと](https://jsk-docs.readthedocs.io/projects/jsk_recognition/en/latest/jsk_perception/nodes/ssd_object_detector.html)と組み合わせて使っていますが、Spotでは以下のlaunchファイルのようにCoralTPUでの物体認識と組み合わせて使っています。

https://github.com/sktometometo/jsk_robot/blob/develop/spot/jsk_spot_robot/jsk_spot_startup/launch/object_detection_and_tracking.launch

https://github.com/sktometometo/jsk_robot/blob/develop/spot/jsk_spot_robot/jsk_spot_startup/launch/include/multi_object_detector.launch

sktometometo, 2021-08-16:
@a-ichikura

このDeepSortノードはCPU or GPUで動作するため、リアルタイムに動作させるにはGPUを使えるように（chainer + cupy 6.7.0環境）セットアップする必要があります。
もしやり方がわからなければ聞いてください。

a-ichikura, 2021-08-16:
@sktometometo ありがとうございます。GPUのセットアップがわからないですが手元にCoralTPUはあるのでやってみます。

k-okada, 2021-08-16:
この手のものは，思った通りにうまくいかないときのプランBも考えながら進めるのが良いですね．

以下は勝手な想像ですが，例えば，
（１：やりたいこと）2つの同じ物体があって，それを覚えていれば，これはAさんのです，とか言いたい．とか，AさんからBさんにプレゼントした物体を覚えておいて，あとでそれを見たときに，Aさん，このまえはプレゼントありがとうと言いたい
→
（２：実験イメージ）画面に人が2人いて，「この前あげたペン貸して」「こちだったけ」「違います」とペンのやり取りをしている．
→
（３：実際のロボットでデータを取る）人が2人写っていて，その前に机があって，人がペンが2つの受け渡しをしている．
→
（４：画像処理を考える）ペンと人を物体認識して，トラッキングしたら，やりたいことができそう

となっているとして，１）をやるには４）は別の可能性もいくつもあって，その中には画像処理が得意なものや不得意なものがいろいろあって，とくに，画像処理研究としては４）の段階の可能性を広げたい，というところなんだろうけど，そうでないほとんどの人は，現状の技術の４）の範囲内で，どこまで１）を達成するか，というのがポイントになります．

なので，質問として４）を例示すると，答えている人はその範疇でこたえようとしますが，１）－３）まで含めて，４）を説明すると，２）や３）でこういう風にしたら，もっと簡単にできるよ，という答えがある可能性があります．

画像処理の能力の一つの指標は，画像上での見え方の変化にどれぐらい対応できるか，という点で，これが具体的に何かは，AIBOの付属品を見るとよくて，（ http://www.e-kasuga.co.jp/home/etc_aibo/ers-7w.htm )，これまでの流れは以下のようになる．
・例えば色付きの球体などどの向きからも円に見えるものの画像処理が簡単，AIBOが最初からこのボールを持っているのはこれが理由．
・そのあと，トランプの柄を認識するみたいにに，視線の向きによって見え方が変化しても，なんとか見えるようになってきている，最近のAIBOがカードを持っているのはこれが理由．しかも，対象の柄の3次元の位置姿勢が2次元画像からわかったりしている．
・見え方だけでなくて，対象の形状が変わってしまうもの．かつ，個体差が多いもの．生き物系はだいたいこれで，その代表例は人間だけど，こういうのはずっと難しかったんだけど，最近のDepp系でできるようになってきている．ただ，逆に，新しい方法で，簡単な球とか，対象の位置姿勢がわかるかというと難しい．

トラッキングも同じような流れになっていて，結局見え方さえ変わらなければ，トラッキングできていた時代から，見え方が変わっても，いろいろできる，という時代になってきている．特にトラッキングだとオクルージョンといって，追跡対象が少し隠れたりしても大丈夫，みたいなところが，今までは難しかった．
https://www.ijcai.org/Proceedings/93-1/Papers/108.pdf
https://ishikawa-vision.org/mvf/SaccadeMirrorFullHD/index-e.html

なので，画像処理の説明は認識できます，トラッキングできます，と書いてあるんだけど，実際には，（ある特定のカテゴリのものが，ある条件で，今まで難しかったこういうものが）物体認識できて，（ある特定の対象がある特定の条件で，しかも今まで難しかったこういう場面で）トラッキングできます，と注釈が付いているもので，それも踏まえて，何パターンか対象・シナリオを用意しておいて，うまく実装できるものを選択するとよいですね．

@sktometometo も数カ月前にCoralで人を認識してトラッキングしたけど，今一つ安定しません，といっていたけど，あれは治ったかな？それも含めてPR出しておきましょう．

@a-ichikura まずは練習で @sktometometo がやっているのと同じことをやって，自分の思っている対象で認識してみて，どうだったか教えてください．で，ほかにもいろいろパターンが思いつくので，うまくいかなくてもあきらめずにチャレンジするのが良いです．

思いつくのは
- 対象物が画面中に小さいと大変なので，大きいほうが良い．
- トラッキングは相手が何かわからないけど追跡する，という処理以外に，毎フレーム対象物を認識して，追跡したとすることもできます．例えば，ペンでなくてコップにして，その表面の柄で区別がつくとすると，かなり正確にトラッキングできるはずです
- Deep系が得意なのは生き物系なの，例えば2つのケーキをトラッキングしていますといいつつ，処理自体はショートケーキの認識とタルトの認識を毎フレームしている，みたいなのは，従来の認識法だと難しそうだけど，Deepでできている例なのでよいと思います．ぬいぐるみとかもよいかと思います．

などなどでしょうか．

knorth55, 2021-08-24:
@k-okada 

> @sktometometo も数カ月前にCoralで人を認識してトラッキングしたけど，今一つ安定しません，といっていたけど，あれは治ったかな？それも含めてPR出しておきましょう．

こちらの件ですが，Panorama画像を分割して認識していたため，人一人に対して複数バウンディングボックスがでる，もしくは出ない，といった現象がおきていました．
これは画像の分割部分などでおきていて，バウンディングボックスをマージする必要がありました．
現在CoralTPUの方でnmsをつかってマージするようにしたので，少し安定したかと思います．
https://github.com/knorth55/coral_usb_ros/pull/65#issuecomment-877168429