jsk-ros-pkg/jsk_recognition #2630: CI failures due to test data download failures after Google Drive's change

wkentaro, 2021-10-03:
## Problem

It seems the full test on noetic in CI fails always recently.
https://github.com/jsk-ros-pkg/jsk_recognition/actions/workflows/noetic-full-test.yml
[image]

I think this is because of the change of Google Drive's sharing permissions and that disables the download of test data from Google Drive as the following. Previously the sign-in was not required. 
[image]

## Context

We upload files and download them from [JSK_public_data](https://drive.google.com/drive/u/1/folders/0B9P1L--7Wd2vUGplQkVLTFBWcFE?resourcekey=0-qlPyB_oRQm887pgRGiPhgg), so **ideally** the permission should be

(**config-ideal**)
- writable for anyone in JSK
- readable from anyone with link

We couldn't find a way of how to make it writable only for people in JSK, so we have been using

(**config-actual**)
- writable for anyone with link
- readable from anyone with link

But now, Google Drive has changed its specification and we cannot use **config-actual** anymore (as it requires sign-in even for just downloading files via links).

## Solution

I think the solution would be adding a group to JSK_public_data that includes all members in JSK (something like jsk-all-members@jsk.imi.i.u-tokyo.ac.jp). Please let me know if there's already such group (maybe it would be better to let me know via email not to make it public).

knorth55, 2021-10-04:
@wkentaro 
Even if we make a group like `jsk-all-members`, `gdown` cannot download the data because it requires sign-in.
Is my understanding correct?
So we need to download data with authentication such as `pydrive2`, right?

sktometometo, 2021-10-04:
I have created a shared drive for JSK lab and make JSK_public_data directory to it.
https://drive.google.com/drive/folders/1PPOZDoWbJnxM7XkPMPP3iKQqrji56Vgq?usp=sharing

Downloading from this directory doesn't require sign-in and standard users can only download data and JSK lab member can upload data to this directory. This can realize config-actual.

knorth55, 2021-10-04:
@wkentaro can you make @knorth55 @sktometometo as an owner?
we can move this directory to shared drive.

wkentaro, 2021-10-04:
> @wkentaro
> Even if we make a group like `jsk-all-members`, `gdown` cannot download the data because it requires sign-in.
> Is my understanding correct?
> So we need to download data with authentication such as `pydrive2`, right?

No, because Google Drive has 2 different sharing options:

- Share with people and groups
- Get link

With `jsk-all-members`, we can use `Share with people and groups` to give write access to JSK members, so we can make `Get link` to read-only access to anyone with link.

wkentaro, 2021-10-04:
> I have created a shared drive for JSK lab and make JSK_public_data directory to it. https://drive.google.com/drive/folders/1PPOZDoWbJnxM7XkPMPP3iKQqrji56Vgq?usp=sharing
>
> Downloading from this [...]

It seems I cannot upload files to it. Are you sure all members have the write access?

https://user-images.githubusercontent.com/4310419/135833922-6d6a7fec-9a1c-4be4-b6a5-939b10174a06.mp4

wkentaro, 2021-10-04:
> @wkentaro can you make @knorth55 @sktometometo as an owner? we can move this directory to shared drive.

Let's find the solution first. Also, I can do that if you don't mind (I can also create Shared Drives).

wkentaro, 2021-10-04:
@sktometometo I also thought about Shared Drives before, but it seemed there was no option to give All Members write access, and probably we still need to add members as a person or a group. (which makes sense as write access to everyone sounds dangerous especially when you have alumnis in the organization like me)
[image]

sktometometo, 2021-10-04:
@wkentaro I have added gakusei-kyoin@jsk.imi.i.u-tokyo.ac.jp to members of the shared drive ( and you ). All jsk members are in this group, so all jsk member can upload files. Can you try again?

knorth55, 2021-10-04:
あんまりgithub issueでメールアドレスとかかいて，こういう話をすべきでない気がします．
スパムの原因になりそう．
あとJSKの管理下に置きたいので，まず和田さん管理になっているJSK_public_dataはデータをダウンロードできる，できない関係なくShared Driveに移行すべきです

knorth55, 2021-10-04:
そしてShared Driveに移行したらあとは現JSKメンバで対応してくれると思います．
とりあえずまずCIが通ること．
そして理想形についてはShared driveでメンバに現JSKメンバをいれるようにすることでJSKメンバはアップロードできて，他の人はURLでダウンロードできるようになります．（ただし和田さんはアップロードできなくなります）

wkentaro, 2021-10-04:
その通りだと思います。
ただ僕が懸念しているのは解決策がわからないままに移動させたりすると、移動を繰り返しているうちにファイルのURLが変わったりしてすべてのtest_dataのURLを書き換えないといけなくなるんじゃないかということです。（どっちにしろSharedDrivesに移動したら変わるのかもしれませんが変わらない可能性があるのならそっちのほうがいいだろうと。）

knorth55, 2021-10-04:
ただ個人のドライブ管理をやめるというのは，もうJSKの方針になってしまっているので，SharedDriveに移行してほしいです．
卒業したあと管理者と連絡が取れなくなるとかが発生したので．
SharedDriveに移行してもURLは変わらないんじゃないかなと信じています．
ただ変わってしまうことよりも個人で管理していることが問題なので，，，
（今回みたいに和田さんが気づいてくれたからいいですが，今後気づかなくなったらおわりなので）

sktometometo, 2021-10-04:
最近、共有アイテムとして共有されていたものを共有ドライブに移した際は, URLは変更しておらず、今さっき共有ドライブから個人ドライブに移動させた際もURLは変更していませんでした。

knorth55, 2021-10-04:
そして共有ドライブに移動することで権限の問題は解決されます．

wkentaro, 2021-10-04:
@sktometometo 先の例では僕が権限を与えてもらっているので、アップロードできても当然だと思うのですが、jsk-all-membersが追加されている状態であれば現在メンバーは問題なくアップロードできているのでしょうか？

knorth55, 2021-10-04:
@wkentaro できました

sktometometo, 2021-10-04:
@wkentaro 他の人に試してもらいましたができているようです。

sktometometo, 2021-10-04:
新規に入った人も, gakusei-kyoin グループに入っていればgakusei-kyoinが追加されている共有ドライブにはアクセスできているので問題ないかと思います

wkentaro, 2021-10-04:
確認ありがとうございます。
個人フォルダはオーナーを移行できないようなので、僕が共有ドライブに移動してOwner権限を移譲するのが安全だと思うのですがそれでいいでしょうか。

ここにアップロードできるか一応試してもらえますか？　@sktometometo 
https://drive.google.com/drive/u/2/folders/0AJDdHz203zXSUk9PVA
https://drive.google.com/drive/u/2/folders/0AJDdHz203zXSUk9PVA

sktometometo, 2021-10-04:
@wkentaro アップロードできました
gakusei-kyoin の権限を, コンテンツ管理者から管理者に変更していただければ幸いです

wkentaro, 2021-10-04:
了解です。

sktometometo, 2021-10-04:
JSKPublicDataの方に移動していただけますか？

sktometometo, 2021-10-04:
命名規則的にこちらを利用したいです

wkentaro, 2021-10-04:
@sktometometo さんが作った方ってこと？

knorth55, 2021-10-04:
なぜわざわざもう一つ作ったのでしょう．．．

sktometometo, 2021-10-04:
もともと作成しています(上で共有したもの)

wkentaro, 2021-10-04:
Mangerアクセスもらえますか。

wkentaro, 2021-10-04:
ああ、大丈夫でした。

wkentaro, 2021-10-04:
権限のせいではなく、フォルダが移動できないだけでした。

wkentaro, 2021-10-04:
JSKPublicData/JSK_public_dataに移動すればいいですか？

sktometometo, 2021-10-04:
@wkentaro そうです、おねがいします。

wkentaro, 2021-10-04:
@sktometometo 今いくつか移動しましたが、こんな感じで問題ないでしょうか？

sktometometo, 2021-10-04:
@wkentaro ありがとうございます。大丈夫そうです。最終的に大丈夫かどうか含めて対応させていただきます。

wkentaro, 2021-10-04:
確認ありがとうございます。全部移動し終わったらまた報告します。

wkentaro, 2021-10-04:
@sktometometo 一部のファイル以外は移動し終わりました。僕のアクセスも消しておきました。

移動できなかったファイルShinjoくんのもので、個人メールアドレスからアップロードされているようで、それが原因で移動できないようです。（ファイルのオーナーがOrganization内でないといけないようです）
残りのファイルは以下にあります（もとのJSK_public_data）。Shinjoくんの個人メールアドレスとJSKメールアドレスにEditor権限を与えたので、うまく移動できないか試してもらえますか？
https://drive.google.com/drive/u/2/folders/0B9P1L--7Wd2vUGplQkVLTFBWcFE?resourcekey=0-qlPyB_oRQm887pgRGiPhgg
[image]

sktometometo, 2021-10-04:
@wkentaro 移動できました。これですべてだと思います。ありがとうございます。

wkentaro, 2021-10-04:
@sktometometo ありがとうございます。お手数おかけしました。CIでのチェックなど残りの処理もよろしくお願いします。

wkentaro, 2021-10-04:
@sktometometo 
共有ドライブへの移行後にダウンロードを試したので結果を共有します。

少し試した程度ですが、以下の例のようにファイルの元オーナーが共有フォルダに権限があるかどうかで
ダウンロードする際にSign-inが必要かどうかが変わるようです。（Google Drive上でViewするだけならSign-inなしでできます）

以下の動画での現象から、

- Yanokura IoriのファイルはYanokura Ioriがjsk-all-members（Manager）のメンバーなのでSign-inなしでダウンロードできる
- Kentaro WadaのファイルはKentaro Wadaが何の権限もないのでSign-inが必要

となっているのではないかと考えています。

Shinjoくんの個人メールアドレスがオーナーになっているファイルでも同じことが起こると思うので、どのような権限を与えたらSign-inが要らなくなるのか試してみてもらえればと思います。できればManagerやContent Managerのような高い権限を与えなくてよければ理想的ですが、最悪でもファイルごとに設定（Kentaro WadaがOwnerになっているファイルのみManager権限を与える）すれば解決できるのではないかと思います。

https://user-images.githubusercontent.com/4310419/135920708-6aca50ee-8bf5-4168-939e-6921807858fc.mp4

sktometometo, 2021-10-05:
@wkentaro  知見の共有ありがとうございます。共有ドライブでこのような問題が起こらないのは, アップロードしたあとのファイルのOwnerなどの扱いが共有アイテムなどとは異なるのかもですね。

https://user-images.githubusercontent.com/9410362/135953707-144d2344-452a-4971-bfa2-db82434470c1.mp4

wkentaro, 2021-10-05:
僕の環境で起こっていることとしては、現象として

- GetLinkで返されるLinkからはSign-inなしではダウンロードできないファイルがある。
- Sign-inせずにJSK_public_dataフォルダをブラウザをみてDownloadボタンを押すとおそらくすべてのファイルがダウンロードできる。

ということみたいですね。
GetLinkで返されるLinkからダウンロードできないと今回の問題の解決にはまだなっていなそうですが、 @sktometometo さんの環境ではどうでしょうか？

sktometometo, 2021-10-05:
とりあえず、 https://github.com/jsk-ros-pkg/jsk_recognition/issues/2631 のいくつかのファイルはそもそも存在しなかったりしたので, jsk_perception, jsk_pcl_ros, jsk_pcl_ros_utils でダウンロードできないものはアップロードしなおしました。

管理者に入っていない人がアップロードしたファイルはログインが必要な件ですが、僕の個人アカウントからアップロードしたものはログイン無しでダウンロードできるので, 例外もありそうです。

eng2_subway_tablenavigation.walk.tar.gz
https://drive.google.com/file/d/1Lye5ZfFLgmU-G8TdevSiHG72kN04XnFt/view