jsk-ros-pkg/jsk_recognition #2631: Sample data does not exists

sktometometo, 2021-10-05:
Sample data and Traned data does not exists.

## Trained Data

- drill_svm.xml
  + URL: https://drive.google.com/uc?id=0B5hRAGKTOm_KWW11R0FTX0xjTDg

- vgg16_fast_rcnn.chainermodel
  + URL: https://drive.google.com/uc?id=0B9P1L--7Wd2vX015UzB4aC13cVk

- apc2015_sample_bof.pkl.gz
  + URL: https://drive.google.com/uc?id=0B9P1L--7Wd2vemVRaDBOWDVpb28

- vgg_cnn_m_1024.chainermodel
  + URL: https://drive.google.com/uc?id=0B9P1L--7Wd2vZzJuaFRIdDMtLWc

- bvlc_alexnet.chainermodel
  + URL: https://drive.google.com/uc?id=0B5DV6gwLHtyJZkd1ZTRiNUdrUXM

- bvlc_vgg16.chainermodel
  + URL: https://drive.google.com/uc?id=0B9P1L--7Wd2vSlFjQlJFQjM5TEk

- vgg16_bn_apc2015_496000.chainermodel
  + URL: https://drive.google.com/uc?id=0B9P1L--7Wd2vQ2tCN1hoYV84eHM

- apc2015_sample_clf.pkl.gz
  + URL: https://drive.google.com/uc?id=0B9P1L--7Wd2veFY5ZFNqbzAzNmc

## Sample Data

- 2016-11-11-12-53-06_in_lab.bag
  + URL: https://drive.google.com/uc?id=0B9P1L--7Wd2vdE9IRERDLXRoUWs

- 2016-10-15-23-21-42_moving_bottle.bag
  + URL: https://drive.google.com/uc?id=0B9P1L--7Wd2vTDZOZGhrTTNvc00

- 2017-06-20-12-00-00_people_images_in_lab.bag.tgz
  + URL: https://drive.google.com/uc?id=0B_NiLAzvehC9Yzl4YzB1WnlFcXM

- apc2016_object_imgs_and_masks.bag
  + URL: https://drive.google.com/uc?id=0B9P1L--7Wd2veVJ4N21tVlB3MGs

- apc2016_object_imgs_and_masks_templates.tgz
  + URL: https://drive.google.com/uc?id=0B9P1L--7Wd2vX0NqcXNGYXEzUlE

- resnet_features_apc2016.npz
  + URL: https://drive.google.com/uc?id=0B9P1L--7Wd2vM3hhOTBGNjA1eTQ

knorth55, 2021-10-05:
@sktometometo I sent you the original data.

sktometometo, 2021-10-05:
I have uploaded and re-uploaded files above.

closed 2021-10-05 by sktometometo