jsk-ros-pkg/jsk_recognition #2655: Plane coefficients direction is unstable in PlaneConcatenator.

Tacha-S, 2022-01-06:
Plane coefficients direction flip according to the position of the point cloud TF.
I think this problem is due to not executing [`forceToDirectOrigin`, which is executed in OrganizedMultiPlaneSegmentation](https://github.com/jsk-ros-pkg/jsk_recognition/blob/6ff031d51d425bf77456d89fbe84b61e58da3aea/jsk_pcl_ros/src/organized_multi_plane_segmentation_nodelet.cpp#L139-L160) after refinement.

Is there any plan to implement this in the future?

iory, 2022-01-06:
Thank you for your reports.

I think it is common to have a node that receives coefficients and aligns them.
How do you expect the plane coefficient direction to be aligned?

Tacha-S, 2022-01-07:
The expected output is to be aligned to the side where the depth sensor is detecting the plane (or vice versa).
For example, in the case of an optical frame, it should be aligned so that the angle with the unit z is less than pi/2 (or vice versa).

k-okada, 2022-01-07:
@Tacha-S Can you create PR that enables your request?

PR #2658 (2022-11-19): Force plane coefficient to direct origin in plane concatinator
Fixed a problem in `PlaneConcatinator` that caused the coefficients to flip depending on the angle between the point cloud frame and the plane.
- fix #2655 

Before the improvement, the Z axis is inverted depending on the angle between the point cloud frame and the plane, as shown in the following pictures.
Since the normal vector of the plane in the pictures is calculated from the polygon without using coefficients, the orientation of the coefficients is determined by the centroid_pose_array of the `ClusterPointIndicesDecomposer`. [... cut]
closed 2022-11-19 by k-okada