jsk-ros-pkg/jsk_recognition #2660: Buildに失敗する

a-ichikura, 2022-01-10:
spotkinovaのjsk_perceptionをbuildしようとしてもFailedになります。

`leus@ubuntu:~/spotkinova_ws/src$ wstool 
workspace: /home/leus/spotkinova_ws/src

 Localname                   S SCM Version (Spec) UID  (Spec)  URI  (Spec) [http(s)://...]
 ---------                   - --- -------------- -----------  ---------------------------
 tork-a/visualization_rwt      git kinetic-devel  9c015164e95e github.com/tork-a/visualization_rwt.git
 tork-a/roswww                 git fetch15        fcb215a9331a github.com/knorth55/roswww.git
 spot-ros                      git develop/spot   21c2e0bd7384 github.com/sktometometo/spot_ros.git
 ros_kortex_vision             git master         e65349f08be7 github.com/Kinovarobotics/ros_kortex_vision
 ros_kortex                    git melodic-devel  1f9d60ae3e6e github.com/kinovarobotics/ros_kortex
 knorth55/app_manager_utils    git master         4d32e427220f github.com/knorth55/app_manager_utils.git
 jsk-ros-pkg/jsk_robot       M git develop/spot   9e59b7d50f0c github.com/sktometometo/jsk_robot.git
 jsk-ros-pkg/jsk_recognition M git master         be6e319d2979 github.com/jsk-ros-pkg/jsk_recognition.git
 jsk-ros-pkg/jsk_demos         git develop/spot   9f2765db099d github.com/sktometometo/jsk_demos.git
 jsk-ros-pkg/jsk_common        git develop/spot   ea3f9ceb7a03 github.com/sktometometo/jsk_common.git
 jsk-ros-pkg/jsk_3rdparty      git develop/spot   be0570485d7d github.com/sktometometo/jsk_3rdparty.git
 aerial_robot                  git develop/spot   c7f246236a0b github.com/sktometometo/aerial_robot.git
 PR2/app_manager               git kinetic-devel  9ed2f9254d3c github.com/PR2/app_manager.git
`
`___________________________________________________________________________________________________________________________________________________________________________________________________________
Errors     << jsk_perception:make /home/leus/spotkinova_ws/logs/jsk_perception/build.make.001.log                                                                                                          
Please install CuPy

    sudo pip install cupy-cuda[your cuda version]
i.e.
    sudo pip install cupy-cuda91

/usr/lib/python2.7/dist-packages/h5py/__init__.py:36: FutureWarning: Conversion of the second argument of issubdtype from `float` to `np.floating` is deprecated. In future, it will be treated as `np.float64 == np.dtype(float).type`.
  from ._conv import register_converters as _register_converters
[/home/leus/spotkinova_ws/src/jsk-ros-pkg/jsk_recognition/jsk_perception/sample/data/2016-11-11-12-53-06_in_lab.bag] File exists or not writable, so skipping creating symlink.
[/home/leus/spotkinova_ws/src/jsk-ros-pkg/jsk_recognition/jsk_perception/sample/data/2016-10-15-23-21-42_moving_bottle.bag] File exists or not writable, so skipping creating symlink.
[/home/leus/spotkinova_ws/src/jsk-ros-pkg/jsk_recognition/jsk_perception/sample/data/2020-11-11-15-58-47-d435-floor.bag] File exists or not writable, so skipping creating symlink.
[/home/leus/spotkinova_ws/src/jsk-ros-pkg/jsk_recognition/jsk_perception/sample/data/sample_mask_rcnn_73b2_kitchen.bag] File exists or not writable, so skipping creating symlink.
[/home/leus/spotkinova_ws/src/jsk-ros-pkg/jsk_recognition/jsk_perception/sample/data/sample_mask_rcnn_73b2_kitchen_japanese_label.bag] File exists or not writable, so skipping creating symlink.
CMakeFiles/jsk_perception.dir/src/bing.cpp.o: 関数 `jsk_perception::Bing::onInit()' 内:
/home/leus/spotkinova_ws/src/jsk-ros-pkg/jsk_recognition/jsk_perception/src/bing.cpp:89: `cv::saliency::ObjectnessBING::setTrainingPath(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)' に対する定義されていない参照です
collect2: error: ld returned 1 exit status
make[2]: *** [/home/leus/spotkinova_ws/devel/.private/jsk_perception/lib/libjsk_perception.so] Error 1
make[1]: *** [CMakeFiles/jsk_perception.dir/all] Error 2
make[1]: *** 未完了のジョブを待っています....
make: *** [all] Error 2
cd /home/leus/spotkinova_ws/build/jsk_perception; catkin build --get-env jsk_perception | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...........................................................................................................................................................................................................
Failed     << jsk_perception:make                  [ Exited with code 2 ]                                                                                                                                  
Failed    <<< jsk_perception                       [ 25.4 seconds ]                                                                                                                                        
[build] Summary: 12 of 13 packages succeeded.                                                                                                                                                              
[build]   Ignored:   157 packages were skipped or are blacklisted.                                                                                                                                         
[build]   Warnings:  None.                                                                                                                                                                                 
[build]   Abandoned: None.                                                                                                                                                                                 
[build]   Failed:    1 packages failed.                                                                                                                                                                    
[build] Runtime: 29.0 seconds total. `

@sktometometo とbuildしていたのですが、詳しい原因がわからず現状jsk_pcl_rosだけ別でbuildしました。

a-ichikura, 2022-01-10:
@sktometometo です。手元で同様のworkspaceを再現するために, vcstoolでワークスペース構成を出力しました。

```yaml
repositories:
  PR2/app_manager:
    type: git
    url: https://github.com/PR2/app_manager.git
    version: 9ed2f9254d3c5c81ee99a7eecab40a93174b71d6
  aerial_robot:
    type: git
    url: https://github.com/sktometometo/aerial_robot.git
    version: c7f246236a0b908ba0467f5e25022d0f04a96435
  jsk-ros-pkg/jsk_3rdparty:
    type: git
    url: https://github.com/sktometometo/jsk_3rdparty.git
    version: be0570485d7d92e39273bbad7a3137eccb23b8f4
  jsk-ros-pkg/jsk_common:
    type: git
    url: https://github.com/sktometometo/jsk_common.git
    version: ea3f9ceb7a030bf0e37460a08fba0f419aa13318
  jsk-ros-pkg/jsk_demos:
    type: git
    url: https://github.com/sktometometo/jsk_demos.git
    version: 9f2765db099d6028672549f23665d63f731daf3c
  jsk-ros-pkg/jsk_recognition:
    type: git
    url: https://github.com/jsk-ros-pkg/jsk_recognition.git
    version: be6e319d29797bafb10c589fdff364c3d333a605
  jsk-ros-pkg/jsk_robot:
    type: git
    url: https://github.com/sktometometo/jsk_robot.git
    version: 9e59b7d50f0c204fa787ce976b925545b05ef6e5
  knorth55/app_manager_utils:
    type: git
    url: https://github.com/knorth55/app_manager_utils.git
    version: 4d32e427220f0ec58c7c8cff706ad80cffc28ef5
  ros_kortex:
    type: git
    url: https://github.com/kinovarobotics/ros_kortex
    version: 1f9d60ae3e6e12a332f96678d18e496d365f20ea
  ros_kortex_vision:
    type: git
    url: https://github.com/Kinovarobotics/ros_kortex_vision
    version: e65349f08be77f3f4b4ef79a4e76373756a3b255
  spot-ros:
    type: git
    url: https://github.com/sktometometo/spot_ros.git
    version: 21c2e0bd7384abb70b16745e011af7ce45281899
  tork-a/roswww:
    type: git
    url: https://github.com/knorth55/roswww.git
    version: fcb215a9331a7ac4086e7b63df9ddd6cac3fdd50
  tork-a/visualization_rwt:
    type: git
    url: https://github.com/tork-a/visualization_rwt.git
    version: 9c015164e95ec13492dac9bf433c082355168ce4
```

sktometometo, 2022-01-10:
僕の手元ではそのままbuildできました。ちょっとログの回収できてないのですが、@a-ichikura のPC内で apt でインストールされている OpenCVのバージョンは3.2.0にも関わらず `-v` オプションをつけてjsk_perceptionをbuildした際に, OpenCV 4.5.1 FOUND と出ていたので, apt で入れた以外のOpenCVが入っているかもしれません。

BINGはcontribの方なので、aptで入れずにOpenCVをインストールした場合は方法によってはライブラリがなくてリンクに失敗するかもです。

とりあえず https://github.com/jsk-ros-pkg/jsk_recognition/blob/be6e319d29797bafb10c589fdff364c3d333a605/jsk_perception/CMakeLists.txt#L290-L292 をコメントアウトすることでbuildできているので、そのままデモの確認をしてもらっています。

@a-ichikura
過去にCUDAを利用したOpenCVの機能を利用するためなどで、OpenCVをソースからbuildしてインストールするなどしていてもし不要なら、削除して置いたほうがいいかもです。

a-ichikura, 2022-01-11:
@sktometometo 
ありがとうございます。何も解決していませんがご報告です。

OpenCV 4.5.1がpython2とpython3で呼び出したときに見つかりました。
```bash 
leus@ubuntu:~$ python3
Python 3.6.9 (default, Dec  8 2021, 21:08:43) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import cv2
>>> cv2.__version__
'4.5.1'
>>> 
```
```bash
leus@ubuntu:~$ python2
Python 2.7.17 (default, Feb 27 2021, 15:10:58) 
[GCC 7.5.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import cv2
>>> cv2.__version__
'4.5.1'
>>> 
```
なので、単純にアンインストールしたら良いのかと思ってpip uninstallしたらライブラリがないと言われました。 
```bash
leus@ubuntu:~$ pip3 list | grep cv
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
cv-bridge (1.13.0)
leus@ubuntu:~$ pip list | grep cv
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
chainercv                          0.13.1
cv-bridge                          1.13.0
webrtcvad                          2.0.10
leus@ubuntu:~$ pip uninstall opencv-python opencv-contrib-python
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
WARNING: Skipping opencv-contrib-python as it is not installed.
WARNING: Skipping opencv-python as it is not installed.
leus@ubuntu:~$ pip3 uninstall opencv-python opencv-contrib-python
Cannot uninstall requirement opencv-python, not installed
```

shmpwk, 2022-01-11:
```
$ python2
>>> import cv2
>>> cv2
```
```
$ which pip
```
```
$ which python 
```
これらの出力を見たい

a-ichikura, 2022-01-11:
@shmpwk 
```bash
leus@ubuntu:~$ python2
Python 2.7.17 (default, Feb 27 2021, 15:10:58) 
[GCC 7.5.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import cv2
>>> cv2
<module 'cv2' from '/usr/local/lib/python2.7/dist-packages/cv2/python-2.7/cv2.so'>
>>> exit()
leus@ubuntu:~$ which pip
/home/leus/.local/bin/pip
leus@ubuntu:~$ which python
/usr/bin/python
```

knorth55, 2022-01-11:
```
ldd /home/leus/spotkinova_ws/devel/.private/jsk_perception/lib/libjsk_perception.so | grep opencv -i
```
?

my workspace example
```
$ ldd ~/ros/melodic/devel/.private/jsk_perception/lib/libjsk_perception.so  | grep cv -i
        libcv_bridge.so => /opt/ros/melodic/lib/libcv_bridge.so (0x00007f579c3d1000)
        libopencv_calib3d.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_calib3d.so.3.2 (0x00007f579bbf5000)
        libopencv_core.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_core.so.3.2 (0x00007f579b7ba000)
        libopencv_highgui.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_highgui.so.3.2 (0x00007f579b5ae000)
        libopencv_imgcodecs.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2 (0x00007f579b375000)
        libopencv_imgproc.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_imgproc.so.3.2 (0x00007f579ae24000)
        libopencv_ml.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_ml.so.3.2 (0x00007f579ab7a000)
        libopencv_video.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_video.so.3.2 (0x00007f579a917000)
        libopencv_saliency.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_saliency.so.3.2 (0x00007f579a6ea000)
        libopencv_flann.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_flann.so.3.2 (0x00007f5794af5000)
        libopencv_features2d.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_features2d.so.3.2 (0x00007f578af4c000)
```

knorth55, 2022-01-11:
if you are using `anaconda` or `miniconda`, please check opencv in `anaconda`, too.

a-ichikura, 2022-01-11:
I don't use  ```conda``` ```anaconda```, or ```miniconda```.

```bash
leus@ubuntu:~/euslib$ ldd /home/leus/spotkinova_ws/devel/.private/jsk_perception/lib/libjsk_perception.so | grep opencv -i
	libopencv_calib3d.so.4.5 => /usr/local/lib/libopencv_calib3d.so.4.5 (0x00007f1611498000)
	libopencv_core.so.4.5 => /usr/local/lib/libopencv_core.so.4.5 (0x00007f16102cb000)
	libopencv_highgui.so.4.5 => /usr/local/lib/libopencv_highgui.so.4.5 (0x00007f1610087000)
	libopencv_imgcodecs.so.4.5 => /usr/local/lib/libopencv_imgcodecs.so.4.5 (0x00007f160fe22000)
	libopencv_imgproc.so.4.5 => /usr/local/lib/libopencv_imgproc.so.4.5 (0x00007f160e0d2000)
	libopencv_ml.so.4.5 => /usr/local/lib/libopencv_ml.so.4.5 (0x00007f160de45000)
	libopencv_video.so.4.5 => /usr/local/lib/libopencv_video.so.4.5 (0x00007f160dbc8000)
	libopencv_features2d.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_features2d.so.3.2 (0x00007f160d25d000)
	libopencv_features2d.so.4.5 => /usr/local/lib/libopencv_features2d.so.4.5 (0x00007f1607f56000)
	libopencv_flann.so.4.5 => /usr/local/lib/libopencv_flann.so.4.5 (0x00007f1607cee000)
	libopencv_dnn.so.4.5 => /usr/local/lib/libopencv_dnn.so.4.5 (0x00007f1602d8d000)
	libopencv_imgcodecs.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2 (0x00007f1602b54000)
	libopencv_imgproc.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_imgproc.so.3.2 (0x00007f16022fb000)
	libopencv_core.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_core.so.3.2 (0x00007f1601ec0000)
	libopencv_calib3d.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_calib3d.so.3.2 (0x00007f1601749000)
	libopencv_flann.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_flann.so.3.2 (0x00007f16014f9000)
	libopencv_objdetect.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_objdetect.so.3.2 (0x00007f15fe6ce000)
	libopencv_video.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_video.so.3.2 (0x00007f15fe46b000)
	libopencv_rgbd.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_rgbd.so.3.2 (0x00007f15fe1e5000)
	libopencv_text.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_text.so.3.2 (0x00007f15fdec6000)
	libopencv_ml.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_ml.so.3.2 (0x00007f15e6a6b000)
```

knorth55, 2022-01-11:
```
	libopencv_imgcodecs.so.4.5 => /usr/local/lib/libopencv_imgcodecs.so.4.5 (0x00007f160fe22000)
```
!!!!

mqcmd196, 2022-01-11:
If you find difficult, I can debug tomorrow.

a-ichikura, 2022-01-11:
I tried to find the directory where I built, but I don't find it. I will ask someone tomorrow. Thank you.

a-ichikura, 2022-01-12:
I found OpenCV source tree by `locate OpenCV` and ran
```
leus@ubuntu:~/Downloads/OpenCV/build$ sudo make uninstall
```
to uninstall opencv files under `/usr/local`

after that, I confirmed opencv version by
```
leus@ubuntu:~/Downloads/OpenCV/build$ pkg-config opencv --modversion
3.2.0
leus@ubuntu:~/Downloads/OpenCV/build$ pkg-config opencv --cflags
-I/usr/include/opencv
leus@ubuntu:~/Downloads/OpenCV/build$ pkg-config opencv --libs
-lopencv_shape -lopencv_stitching -lopencv_superres -lopencv_videostab -lopencv_aruco -lopencv_bgsegm -lopencv_bioinspired -lopencv_ccalib -lopencv_datasets -lopencv_dpm -lopencv_face -lopencv_freetype -lopencv_fuzzy -lopencv_hdf -lopencv_line_descriptor -lopencv_optflow -lopencv_video -lopencv_plot -lopencv_reg -lopencv_saliency -lopencv_stereo -lopencv_structured_light -lopencv_phase_unwrapping -lopencv_rgbd -lopencv_viz -lopencv_surface_matching -lopencv_text -lopencv_ximgproc -lopencv_calib3d -lopencv_features2d -lopencv_flann -lopencv_xobjdetect -lopencv_objdetect -lopencv_ml -lopencv_xphoto -lopencv_highgui -lopencv_videoio -lopencv_imgcodecs -lopencv_photo -lopencv_imgproc -lopencv_core
```

mqcmd196, 2022-01-12:
Please reopen if you still have problem.

closed 2022-01-12 by mqcmd196