jsk-ros-pkg/jsk_recognition #2665: How to use BoundingBoxArray to crop input pointcloud ?

gachiemchiep, 2022-02-03:
御忙しいの中、すみませんが？
ClusterPointIndicesDecomposer の出力結果（BoundingBoxArray）を用いて、入力pointcloudをcropする関数を教えてもらいませんか？

現在、私は pointcloudからmask imageを作りたいです。
https://github.com/jsk-ros-pkg/jsk_recognition/pull/765 の説明とおり、入力pointcloudにpassthrough filter適用すれば、mask imageを生成できます。
[image]

mask imageがちょっと汚いので、Euclid clusteringして、物体のboundingboxを抽出します。抽出されたboundingboxを用いて、入力pointcloudをcropして、mask imageを生成します。現在、boundingboxを用いて、入力pointcloud crop仕方を分からないので、関数またはやり方を教えてもらいませんか？以下の画像はEuclid clusteringの結果です。
[image]

knorth55, 2022-02-03:
Sorry but we don't have enough time or human resource to solve your issue properly.
Also we don't do consulting for your business.
But if you are using tabletop environment, I recommend to use tabletop detector launch.
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/sample/tabletop_object_detector.launch

gachiemchiep, 2022-02-03:
@knorth55 
Thank you for your reply. 
Please ignore the non-sense detail I post above. 

I just want to know is there any function in jsk package which can crop input pointcloud using output of ClusterPointIndicesDecomposer (~boxes (jsk_recognition_msgs/BoundingBoxArray)) as setting parameters ?

knorth55, 2022-02-03:
It may be AttentionClipper
https://jsk-docs.readthedocs.io/projects/jsk_recognition/en/latest/jsk_pcl_ros/nodes/attention_clipper.html

iory, 2022-02-03:
@gachiemchiep 
https://github.com/jsk-ros-pkg/jsk_recognition/pull/2666
This PR will help you learn the sample cases.

gachiemchiep, 2022-02-05:
@iory  @knorth55 
Thank you for your help. I'm able to crop the pointcloud.

closed 2022-02-05 by gachiemchiep