jsk-ros-pkg/jsk_recognition #2670: how to install jsk_recognition on ros-neotic ubuntu 20.04?

18lalala, 2022-03-25:


k-okada, 2022-03-27:
we already released jsk_recognition in noetic https://github.com/ros/rosdistro/pull/26895, https://build.ros.org/view/Nbin_uF64/job/Mbin_uB64__jsk_recognition__ubuntu_bionic_amd64__binary/
try 
```
sudo apt update
sudo apt install ros-noetic-jsk-recognition
```

greatman666, 2024-05-29:
how to install jsk-rviz-plugins on ros-neotic ubuntu 20.04?

k-okada, 2024-05-29:
please try
```
sudo apt update
sudo apt install ros-noetic-jsk-rviz-plugins
```

greatman666, 2024-05-29:
> please try
> 
> ```
> sudo apt update
> sudo apt install ros-noetic-jsk-rviz-plugins
> ```

I've tried the above command last night, but I still have the error that the package can't be downloaded

k-okada, 2024-05-29:
please provide entire log messages.

greatman666, 2024-05-29:
> please provide entire log messages.

The computer is not nearby at the moment, and I can only upload the error log at evening

greatman666, 2024-05-29:
https://mirrors.tuna.tsinghua.edu.cn/help/ros/   
like thest  blog say: I change the  /etc/apt/sources.list.d/ros-latest.list  to "deb https://mirrors.tuna.tsinghua.edu.cn/ros/ubuntu/ focal main"  and  rechange back to " deb http://packages.ros.org/ros/ubuntu / focal main" ; and it worked