jsk-ros-pkg/jsk_recognition #2680: Node says "received signal SIGSEGV, Segmentation fault." and die

Kanazawanaoaki, 2022-04-28:
jsk_topic_tools(jsk_common)をソースからbuildしていて，jsk_recognitionはaptのものを使っているときにnodeがエラーメッセージ無しで死んでしまいました．
```
[pr2_imitation/apply_mask_image-5] process has died [pid 2224, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet standalone jsk_perception/ApplyMaskImage pr2_imitation_manager ~input:=/kinect_head/rgb/image_decompress_throttle ~input/mask:=mask ~output:=processed_image __name:=apply_mask_image __log:=/home/kanazawa/.ros/log/2d4cea82-c51f-11ec-be71-d05099c29feb/pr2_imitation-apply_mask_image-5.log].
log file: /home/kanazawa/.ros/log/2d4cea82-c51f-11ec-be71-d05099c29feb/pr2_imitation-apply_mask_image-5*.log
```
http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB の`launch-prefix="gdb -ex run --args"`を使って調べると，"received signal SIGSEGV, Segmentation fault."といっていて`bt`するとjsk_topic_toolsに関連していそうでした．
```
[pr2_imitation/apply_mask_image-5] process has died [pid 2224, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet standalone jsk_perception/ApplyMaskImage pr2_imitation_manager ~input:=/kinect_head/rgb/image_decompress_throttle ~input/mask:=mask ~output:=processed_image __name:=apply_mask_image __log:=/home/kanazawa/.ros/log/2d4cea82-c51f-11ec-be71-d05099c29feb/pr2_imitation-apply_mask_image-5.log].
log file: /home/kanazawa/.ros/log/2d4cea82-c51f-11ec-be71-d05099c29feb/pr2_imitation-apply_mask_image-5*.log
[tcsetpgrp failed in terminal_inferior: デバイスに対する不適切なioctlです]
[New Thread 0x7fff98e2b700 (LWP 2362)]
[New Thread 0x7fff9862a700 (LWP 2363)]

Thread 7 "mask_image_gene" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffdffff700 (LWP 2334)]
0x00007fffd741b115 in jsk_perception::MaskImageGenerator::generate(boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&) () from /opt/ros/melodic/lib//libjsk_perception.so
(gdb) bt
#0  0x00007fffd741b115 in jsk_perception::MaskImageGenerator::generate(boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&) () at /opt/ros/melodic/lib//libjsk_perception.so
#1  0x00007fffd52c3937 in boost::_mfi::mf1<void, jsk_topic_tools::DiagnosticNodelet, diagnostic_updater::DiagnosticStatusWrapper&>::operator()(jsk_topic_tools::DiagnosticNodelet*, diagnostic_updater::DiagnosticStatusWrapper&) const ()
    at /home/kanazawa/ros/jsk_demo_ws/devel/lib/libjsk_topic_tools.so
#2  0x00007fffd52c36e9 in void boost::_bi::list2<boost::_bi::value<jsk_topic_tools::DiagnosticNodelet*>, boost::arg<1> >::operator()<boost::_mfi::mf1<void, jsk_topic_tools::DiagnosticNodelet, diagnostic_updater::DiagnosticStatusWrapper&>, boost::_bi::rrlist1<diagnostic_updater::DiagnosticStatusWrapper&> >(boost::_bi::type<void>, boost::_mfi::mf1<void, jsk_topic_tools::DiagnosticNodelet, diagnostic_updater::DiagnosticStatusWrapper&>&, boost::_bi::rrlist1<diagnostic_updater::DiagnosticStatusWrapper&>&, int) ()
    at /home/kanazawa/ros/jsk_demo_ws/devel/lib/libjsk_topic_tools.so
#3  0x00007fffd52c3523 in void boost::_bi::bind_t<void, boost::_mfi::mf1<void, jsk_topic_tools::DiagnosticNodelet, diagnostic_updater::DiagnosticStatusWrapper&>, boost::_bi::list2<boost::_bi::value<jsk_topic_tools::DiagnosticNodelet*>, boost::arg<1> > >::operator()<diagnostic_updater::DiagnosticStatusWrapper&>(diagnostic_updater::DiagnosticStatusWrapper&) ()
    at /home/kanazawa/ros/jsk_demo_ws/devel/lib/libjsk_topic_tools.so
#4  0x00007fffd52c33ad in boost::detail::function::void_function_obj_invoker1<boost::_bi::bind_t<void, boost::_mfi::mf1<void, jsk_topic_tools::DiagnosticNodelet, diagnostic_updater::DiagnosticStatusWrapper&>, boost::_bi::list2<boost::_bi::value<jsk_topic_tools::DiagnosticNodelet*>, boost::arg<1> > >, void, diagnostic_updater::DiagnosticStatusWrapper&>::invoke(boost::detail::function::function_buffer&, diagnostic_updater::DiagnosticStatusWrapper&) () at /home/kanazawa/ros/jsk_demo_ws/devel/lib/libjsk_topic_tools.so
#5  0x00007fffd529d440 in boost::function1<void, diagnostic_updater::DiagnosticStatusWrapper&>::operator()(diagnostic_updater::DiagnosticStatusWrapper&) const () at /home/kanazawa/ros/jsk_demo_ws/devel/lib/libjsk_topic_tools.so
#6  0x00007fffd529bf82 in diagnostic_updater::DiagnosticTaskVector::DiagnosticTaskInternal::run(diagnostic_updater::DiagnosticStatusWrapper&) const () at /home/kanazawa/ros/jsk_demo_ws/devel/lib/libjsk_topic_tools.so
#7  0x00007fffd529c607 in diagnostic_updater::Updater::force_update() ()
    at /home/kanazawa/ros/jsk_demo_ws/devel/lib/libjsk_topic_tools.so
#8  0x00007fffd529c3d2 in diagnostic_updater::Updater::update() () at /home/kanazawa/ros/jsk_demo_ws/devel/lib/libjsk_topic_tools.so
#9  0x00007fffd529bc92 in jsk_topic_tools::TimeredDiagnosticUpdater::update() ()
    at /home/kanazawa/ros/jsk_demo_ws/devel/lib/libjsk_topic_tools.so
#10 0x00007fffd529bcbd in jsk_topic_tools::TimeredDiagnosticUpdater::timerCallback(ros::TimerEvent const&) ()
    at /home/kanazawa/ros/jsk_demo_ws/devel/lib/libjsk_topic_tools.so
#11 0x00007fffd52a2dcf in boost::_mfi::mf1<void, jsk_topic_tools::TimeredDiagnosticUpdater, ros::TimerEvent const&>::operator()(jsk_topic_tools::TimeredDiagnosticUpdater*, ros::TimerEvent const&) const () at /home/kanazawa/ros/jsk_demo_ws/devel/lib/libjsk_topic_tools.so
#12 0x00007fffd52a253f in void boost::_bi::list2<boost::_bi::value<jsk_topic_tools::TimeredDiagnosticUpdater*>, boost::arg<1> >::operator()<boost::_mfi::mf1<void, jsk_topic_tools::TimeredDiagnosticUpdater, ros::TimerEvent const&>, boost::_bi::rrlist1<ros::TimerEvent const&> >(boost::_bi::type<void>, boost::_mfi::mf1<void, jsk_topic_tools::TimeredDiagnosticUpdater, ros::TimerEvent const&>&, boost::_bi::rrlist1<ros::TimerEvent const&>&, int) () at /home/kanazawa/ros/jsk_demo_ws/devel/lib/libjsk_topic_tools.so
#13 0x00007fffd52a1b73 in void boost::_bi::bind_t<void, boost::_mfi::mf1<void, jsk_topic_tools::TimeredDiagnosticUpdater, ros::TimerEvent const&>, boost::_bi::list2<boost::_bi::value<jsk_topic_tools::TimeredDiagnosticUpdater*>, boost::arg<1> > >::operator()<ros::TimerEvent const&>(ros::TimerEvent const&) () at /home/kanazawa/ros/jsk_demo_ws/devel/lib/libjsk_topic_tools.so
#14 0x00007fffd52a1235 in boost::detail::function::void_function_obj_invoker1<boost::_bi::bind_t<void, boost::_mfi::mf1<void, jsk_topic_tools::TimeredDiagnosticUpdater, ros::TimerEvent const&>, boost::_bi::list2<boost::_bi::value<jsk_topic_tools::TimeredDiagnosticUpdater*>, boost::arg<1> > >, void, ros::TimerEvent const&>::invoke(boost::detail::function::function_buffer&, ros::TimerEvent const&) ()
    at /home/kanazawa/ros/jsk_demo_ws/devel/lib/libjsk_topic_tools.so
#15 0x00007ffff7619b07 in ros::TimerManager<ros::Time, ros::Duration, ros::TimerEvent>::TimerQueueCallback::call() ()
    at /opt/ros/melodic/lib/libroscpp.so
#16 0x00007ffff763e559 in ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*) () at /opt/ros/melodic/lib/libroscpp.so
#17 0x00007ffff763f845 in ros::CallbackQueue::callOne(ros::WallDuration) () at /opt/ros/melodic/lib/libroscpp.so
#18 0x00007ffff7bb625d in nodelet::detail::CallbackQueue::callOne() () at /opt/ros/melodic/lib/libnodeletlib.so
#19 0x00007ffff7bb8c05 in nodelet::detail::CallbackQueueManager::workerThread(nodelet::detail::CallbackQueueManager::ThreadInfo*) ()
    at /opt/ros/melodic/lib/libnodeletlib.so
#20 0x00007ffff58d7bcd in  () at /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.65.1
#21 0x00007ffff56ae6db in start_thread (arg=0x7fffdffff700) at pthread_create.c:463
#22 0x00007ffff6adf61f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```
jsk_recognitionもソースからbuildして最新にするとエラーが起こらずに実行できました．

Kanazawanaoaki, 2022-04-28:
https://github.com/jsk-ros-pkg/jsk_recognition/issues/2532 に関係していそうです．

k-okada, 2022-04-28:
```
$ ldd ./lib/libjsk_perception.so  | grep jsk_topic
	libjsk_topic_tools.so => /opt/ros/melodic/lib/libjsk_topic_tools.so (0x00007fd2381b2000)
```
と，jsk_recognition -> jsk_topic_tools という依存関係があって，
apt の jsk_recognition　はaptで入っている 2.2.11 を使ってコンパイルされているけど，
ソースの jsk_topic_tools は　https://github.com/jsk-ros-pkg/jsk_common/compare/2.2.11...master?D=jsk_topic_tools#diff-5d46b01523b8fe2b6071ffbac5b961969f18bc27e6e76597a5c95a797c143e8e　みたいライブラリのヘッダ部分に差分があって，呼び出しルールが変わってしまっているのでjsk_topic_toolsをソースからコンパイルして使っている人は，それに依存しているすべてのプログラムをコンパイルし直さないといけない，となります．
ちなみに，jsk_topic_tools をソースから入れている理由は何でしょうか？

Kanazawanaoaki, 2022-04-28:
ありがとうございます．なるほど，ヘッダーファイル部分の差分が原因なんですね．  
jsk_topic_toolsをソースから入れている理由は，jsk_dataのdata_collection_serverでrosbagをrecordする(https://github.com/jsk-ros-pkg/jsk_common/pull/1708 )ためにjsk_commonのmasterを使いたいからです．