jsk-ros-pkg/jsk_recognition #2685: Regression caused by update objectdetection_tf_publisher for publishing simple tf #457

amirtu, 2022-05-13:
Could someone place the line https://github.com/jsk-ros-pkg/jsk_recognition/blob/39747cc75031e3ff599450189a3f3645b31145a5/checkerboard_detector/src/objectdetection_tf_publisher.py#L19
after these two lines
https://github.com/jsk-ros-pkg/jsk_recognition/blob/39747cc75031e3ff599450189a3f3645b31145a5/checkerboard_detector/src/objectdetection_tf_publisher.py#L20-L21
as it was before [#457](https://github.com/jsk-ros-pkg/jsk_recognition/commit/c5478102d20e9b946f47277347698e9c1c64ddb8)

knorth55, 2022-05-13:
You can make a PR.
This repo is open-source, so you can join by yourself.
Don't hesitate and please feel free to submit your suggestion and describe the reason why you want it in a PR.

amirtu, 2022-05-13:
@knorth55 

Thanks for your prompt reply, I have created a PR for this change.

knorth55, 2022-05-13:
can you explain the detailed error you encountered?
can you also paste the error output?

closed 2023-09-14 by amirtu