jsk-ros-pkg/jsk_recognition #2705: RobotToMaskImage doesn't align

Affonso-Gui, 2022-06-30:
RobotToMaskImage ([doc](https://jsk-docs.readthedocs.io/projects/jsk_recognition/en/latest/jsk_perception/nodes/robot_to_mask_image.html?highlight=Robot) ) seems to be a bit off.

[image]

The robot is not moving at all in the image above, so time lag is not a factor.
Is this kind of error expected?

```
  <node name="robot_to_mask_image"
        pkg="nodelet" type="nodelet"
        args="standalone jsk_perception/RobotToMaskImage">
    <remap from="~input/camera_info" to="/kinect_head/rgb/camera_info"/>
    <rosparam>
      self_see_links:
        - name: l_gripper_palm_link
        - name: l_gripper_l_finger_link
        - name: l_gripper_l_finger_tip_link
        - name: l_gripper_r_finger_link
        - name: l_gripper_r_finger_tip_link
        <!-- - name: r_gripper_palm_link -->
        <!-- - name: r_gripper_l_finger_link -->
        <!-- - name: r_gripper_l_finger_tip_link -->
        <!-- - name: r_gripper_r_finger_link -->
        <!-- - name: r_gripper_r_finger_tip_link -->
      self_see_default_padding: 0.0
    </rosparam>
  </node>
```

iory, 2022-06-30:
Could you show the image overlaying `RobotModel` with `Camera` in rviz?
 http://wiki.ros.org/rviz/DisplayTypes/Camera

Affonso-Gui, 2022-06-30:
Yep, that seems to be the problem:
[image]

Should we recallibrate the kinect camera then?

iory, 2022-06-30:
I see. It's a problem with hand-eye calibration. 
There seem to be positions (also rotations) errors where the current camera is attached.
We can manually update them.

http://wiki.ros.org/pr2_calibration/Tutorials/Calibrating%20the%20PR2#Update_the_system

```
applications@pr1012:/etc/ros/indigo/urdf $ ls /etc/ros/indigo/urdf
pr2_1.11.14.urdf.xacro                 robot_calibrated_2019_07_05_17_09.xml  robot_uncalibrated_1.11.14.xml  robot.xml.backup
pr2_1.12.2.urdf.xacro                  robot_calibrated_2019_11_11.xml        robot_uncalibrated_1.12.2.xml
README                                 robot_calibrated_2019_11_22.xml        robot_uncalibrated.xml
robot_calibrated_2015_09_10_21_46.xml  robot_calibrated_2020_01_30.xml        robot.xml
```