jsk-ros-pkg/jsk_recognition #2706: How to Invert a Mask Image?

Affonso-Gui, 2022-06-30:
How to invert a mask image in the jsk_recognition pipeline?
I have the following:
[image]

And want this:
[image]

iory, 2022-06-30:
You can use `ApplyMaskImage` with `~negative` option.
https://jsk-docs.readthedocs.io/projects/jsk_recognition/en/latest/jsk_perception/nodes/apply_mask_image.html

Affonso-Gui, 2022-06-30:
Thanks, it worked for me with `negative: true` and `clip: false`, although with significant drops in performance (10Hz -> 6Hz).

Maybe we could also add a negative option directly to the RobotToMask node?

iory, 2022-06-30:
> Maybe we could also add a negative option directly to the RobotToMask node?

I think the general direction is to create a node that inverts the mask.

Affonso-Gui, 2022-06-30:
Looking forward to an InvertMask node.

closed 2025-10-19 by mqcmd196