jsk-ros-pkg/jsk_recognition #2752: PanoramaInfo msg can have image height and width?

knorth55, 2022-12-07:
`rects_array_in_panorama_to_bounding_box_array` node subscribe full panorama image,
but it uses only image height and image width image.

https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_perception/node_scripts/rect_array_in_panorama_to_bounding_box_array.py#L53-L68

But when the network connection is low, it is hard to subscribe full panorama image.

We thought `PanoramaInfo` msg can have `image` height and `image` width information

@sktometometo what do you think?

cc. @AyumuIwa

sktometometo, 2022-12-07:
I didn't design PanoramaInfo without Image size since it is redundant, but it sounds good idea to put it in PanoramaInfo.

knorth55, 2022-12-07:
Thank you, I will make a PR.
@sktometometo do you want to keep the current behavior and add new option?
or just rewrite to subscribe only `panorama info`?

sktometometo, 2022-12-07:
@knorth55 Let's make it simple. Please remove a subscriber of panorama image.

knorth55, 2022-12-07:
@sktometometo OK, I will