jsk-ros-pkg/jsk_recognition #2757: sample_line_segment_detector not working in noetic

DuffRumkins, 2022-12-28:
I installed the ROS noetic version of jsk-ros-pkg and wanted to test the sample_line_segment_detector using the following command:
`
roslaunch jsk_pcl_ros sample_line_segment_detector.launch
`

Unfortunately, every time I run it I get the following error
`
[FATAL] [1672237403.125360436]: Error opening file: /opt/ros/noetic/share/jsk_pcl_ros_utils/sample/data/2017-02-05-16-11-09_shelf_bin.bag
`
I also get the following error stating that the process has died
`
[rosbag_play-1] process has died [pid 4157, exit code 1, cmd /opt/ros/noetic/lib/rosbag/play /opt/ros/noetic/share/jsk_pcl_ros_utils/sample/data/2017-02-05-16-11-09_shelf_bin.bag --clock --loop __name:=rosbag_play __log:=/home/<username>/.ros/log/ce9fb696-86ba-11ed-9a55-79ec321c301d/rosbag_play-1.log].
log file: /home/<username>/.ros/log/ce9fb696-86ba-11ed-9a55-79ec321c301d/rosbag_play-1*.log
`

I have checked and I saw that the 2017-02-05-16-11-09_shelf_bin.bag file is indeed missing and I could not find it anywhere online. 

Does anyone have any experience fixing this?

I would ultimately like to run the line_segment_detector on a real time stream from an **Intel RealSense D435**, so if anyone has experience with getting this to work without the 2017-02-05-16-11-09_shelf_bin.bag, please share!

Thanks!

Affonso-Gui, 2022-12-29:
Hello, 

To keep the installation lightweight, sample bag files are only downloaded when building from source (related: https://github.com/jsk-ros-pkg/jsk_recognition/issues/2692).

A quick search in the repository shows the link for the bag file in question:
https://github.com/jsk-ros-pkg/jsk_recognition/blob/f32cf54cd229a66527e8044bb8189a73440a4056/jsk_pcl_ros_utils/scripts/install_sample_data.py#L34-L38

Hopefully the sample should work after you download the bag file and place it on the designated spot (`/opt/ros/noetic/share/jsk_pcl_ros_utils/sample/data/2017-02-05-16-11-09_shelf_bin.bag`, or change the expected path in https://github.com/jsk-ros-pkg/jsk_recognition/blob/659a25c7bed3c8786aefa6ef1ce16a9bd53aaecb/jsk_pcl_ros_utils/sample/include/play_rosbag_shelf_bin.xml#L7).

DuffRumkins, 2022-12-29:
Thanks for your help! I was able to download the file and place it in the correct folder. The original error is solved, but this was immediately replaced by a new one.

When I launch the `sample_line_segment_detector` I get the following warning repeated over and over:

`[ WARN] [1672318442.285249951, 1486278999.838602169]: Detected jump back in time of 2.75951s. Clearing TF buffer.`

And nothing shows in rviz, but I can see that there is an error in the line marker field stating that the number of colors does not match the number of points.

More often than not, the rviz also crashes soon after launching giving me the following error:

`[rviz-22] process has died [pid 7072, exit code -11, cmd /opt/ros/noetic/lib/rviz/rviz -d /opt/ros/noetic/share/jsk_pcl_ros/sample/rviz/line_segment_detector.rviz __name:=rviz __log:=/home/seamus/.ros/log/d43e3622-8778-11ed-8800-bf1f7ac7ffb3/rviz-22.log].
log file: /home/<username>/.ros/log/d43e3622-8778-11ed-8800-bf1f7ac7ffb3/rviz-22*.log`

When I check this log file it is completely empty.

Do you have any idea what might be causing this?

Affonso-Gui, 2022-12-31:
`[ WARN] [1672318442.285249951, 1486278999.838602169]: Detected jump back in time of 2.75951s. Clearing TF buffer.`
This is just a notification that the bag file has been restarted due to the `--loop` argument, and is not relevant to this problem. The bag file is only a few seconds long, which is why you are getting this warning fairly often.

```bash
$ rosbag info 2017-02-05-16-11-09_shelf_bin.bag
path:        2017-02-05-16-11-09_shelf_bin.bag
version:     2.0
duration:    2.8s
start:       Feb 05 2017 16:16:39.83 (1486278999.83)
end:         Feb 05 2017 16:16:42.61 (1486279002.61)
size:        101.1 MB
messages:    828
compression: none [98/98 chunks]
types:       sensor_msgs/CameraInfo [c9a58c1b0b154e0e6da7578cb991d214]
             sensor_msgs/Image      [060021388200f6f0f447d0fcd9c64743]
             sensor_msgs/JointState [3066dcd76a6cfaef579bd0f34173e9fd]
             tf2_msgs/TFMessage     [94810edda583a504dfda3829e70d7eec]
topics:      /right_hand_camera/depth_registered/camera_info    50 msgs    : sensor_msgs/CameraInfo
             /right_hand_camera/depth_registered/image_raw      50 msgs    : sensor_msgs/Image
             /right_hand_camera/rgb/camera_info                 47 msgs    : sensor_msgs/CameraInfo
             /right_hand_camera/rgb/image_raw                   47 msgs    : sensor_msgs/Image
             /robot/joint_states                               279 msgs    : sensor_msgs/JointState
             /tf                                               355 msgs    : tf2_msgs/TFMessage     (5 connections)
```

When I tested, the detection results (`~output/inliers` and `~output/coefficients`, [doc page](https://jsk-docs.readthedocs.io/projects/jsk_recognition/en/latest/jsk_pcl_ros/nodes/line_segment_detector.html)) seem to be turning out fine, so you can probably use the node even without being able to visualize it.
In my case the rviz itself didn't crash though, so there might be something wrong with your installation?

>  there is an error in the line marker field stating that the number of colors does not match the number of points

This seems to be the real issue here. Indeed, the number of colors did not seem to match:

```bash
$ rostopic echo /line_segment_detector/debug/line_marker -n1
header: 
  seq: 1892
  stamp: 
    secs: 1486279001
    nsecs:  53697817
  frame_id: "right_hand_camera_rgb_optical_frame"
ns: ''
id: 0
type: 5
action: 0
pose: 
  position: 
    x: 0.0
    y: 0.0
    z: 0.0
  orientation: 
    x: 0.0
    y: 0.0
    z: 0.0
    w: 1.0
scale: 
  x: 0.01
  y: 0.0
  z: 0.0
color: 
  r: 1.0
  g: 0.0
  b: 0.0
  a: 1.0
lifetime: 
  secs: 0
  nsecs:         0
frame_locked: False
points: 
  - 
    x: -0.448338061571
    y: 0.130320519209
    z: 0.856072425842
  - 
    x: -0.391091704369
    y: 0.201518550515
    z: 1.30683338642
  - 
    x: -0.445450752974
    y: 0.0764372721314
    z: 0.852948606014
  - 
    x: -0.395038932562
    y: 0.150961756706
    z: 1.24994850159
  - 
    x: -0.363396078348
    y: -0.16954216361
    z: 0.709487855434
  - 
    x: -0.311520725489
    y: -0.193637043238
    z: 0.939999997616
  - 
    x: 0.471455156803
    y: 0.43302500248
    z: 1.29349994659
  - 
    x: 0.604960858822
    y: 0.328063368797
    z: 1.50100004673
  - 
    x: 0.195011809468
    y: -0.23387183249
    z: 0.818000018597
  - 
    x: -0.044523704797
    y: -0.217198565602
    z: 0.882000029087
colors: 
  - 
    r: 0.121569000185
    g: 0.466666996479
    b: 0.705882012844
    a: 1.0
  - 
    r: 0.682353019714
    g: 0.780391991138
    b: 0.909803986549
    a: 1.0
  - 
    r: 1.0
    g: 0.498039007187
    b: 0.0549019984901
    a: 1.0
  - 
    r: 0.172548994422
    g: 0.627451002598
    b: 0.172548994422
    a: 1.0
  - 
    r: 0.839215993881
    g: 0.152941003442
    b: 0.156863003969
    a: 1.0
text: ''
mesh_resource: ''
mesh_use_embedded_materials: False
---
```

@iory

DuffRumkins, 2023-01-02:
An update on this issue.

I have since tried to use the Line Segment Detector on the stream from an Intel RealSense depth camera by copying the sample_line_segment_detector.launch file to a new package and modifying the input to be the `sensor_msgs/PointCloud2` topic output by the camera's driver.

Unfortunately, I still get the same error in rviz which leads me to believe that the error lays within the code itself and not the input point cloud.

Any help here would be greatly appreciated!

Affonso-Gui, 2023-01-03:
> the same error in rviz

Does it still crashes or is just unable to show the results?

DuffRumkins, 2023-01-03:
I still get both issues.

The crashing is not that big a deal since I can just relaunch a couple times and it will eventually work.

The main issue is the number of colors not matching since I can not find any fix for this.

DuffRumkins, 2023-01-03:
After running some more tests, I see there seem to always be twice as many marker points as there are colors.

Any idea what could be causing this?

SangbaekYoo, 2023-10-19:
I'm having exactly the same problem. 
```
Number of colors doesn't match number of points.
```

Has anyone else solved this?