jsk-ros-pkg/jsk_recognition #2759: [audio_to_spectrogram] spectrum_plot does not work with apt-version matplotlib (?)

pazeshun, 2023-01-20:
When I launched `audio_to_spectrogram.launch` first time after https://github.com/jsk-ros-pkg/jsk_recognition/pull/2654 was merged, I faced the following error:
```
$ roslaunch audio_to_spectrogram audio_to_spectrogram.launch device:=hw:2,0 mic_sampling_rate:=44100 spectrogram_period:=0.3
... logging to /home/leus/.ros/log/02d876ee-98a6-11ed-8d16-a0cec8428594/roslaunch-jsk-moonshot-laptop-19908.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://133.11.216.138:40935/

SUMMARY
========

PARAMETERS
 * /audio_amplitude_plot/bitdepth: 16
 * /audio_amplitude_plot/mic_sampling_rate: 44100
 * /audio_amplitude_plot/n_channel: 1
 * /audio_capture/channels: 1
 * /audio_capture/depth: 16
 * /audio_capture/device: hw:2,0
 * /audio_capture/format: wave
 * /audio_capture/sample_rate: 44100
 * /audio_to_spectrum/bitdepth: 16
 * /audio_to_spectrum/fft_exec_rate: 50
 * /audio_to_spectrum/fft_sampling_period: 0.3
 * /audio_to_spectrum/high_cut_freq: 800
 * /audio_to_spectrum/low_cut_freq: 1
 * /audio_to_spectrum/mic_sampling_rate: 44100
 * /audio_to_spectrum/n_channel: 1
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /spectrogram_view/colormap: 2
 * /spectrogram_view/do_dynamic_scaling: True
 * /spectrum_to_spectrogram/image_height: 300
 * /spectrum_to_spectrogram/image_width: 300
 * /spectrum_to_spectrogram/spectrogram_period: 0.3

NODES
  /
    audio_amplitude_plot (audio_to_spectrogram/audio_amplitude_plot.py)
    audio_amplitude_view (image_view/image_view)
    audio_capture (audio_capture/audio_capture)
    audio_to_spectrum (audio_to_spectrogram/audio_to_spectrum.py)
    spectrogram_view (image_view/image_view)
    spectrum_plot (audio_to_spectrogram/spectrum_plot.py)
    spectrum_to_spectrogram (audio_to_spectrogram/spectrum_to_spectrogram.py)
    spectrum_view (image_view/image_view)

auto-starting new master
process[master]: started with pid [19929]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 02d876ee-98a6-11ed-8d16-a0cec8428594
process[rosout-1]: started with pid [19951]
started core service [/rosout]
process[audio_capture-2]: started with pid [19958]
process[audio_to_spectrum-3]: started with pid [19959]
process[spectrum_to_spectrogram-4]: started with pid [19960]
process[audio_amplitude_plot-5]: started with pid [19962]
process[spectrum_plot-6]: started with pid [19978]
process[audio_amplitude_view-7]: started with pid [19981]
process[spectrum_view-8]: started with pid [19982]
process[spectrogram_view-9]: started with pid [19984]
[ERROR] [1674207433.577344]: bad callback: <bound method SpectrumPlot._cb of <__main__.SpectrumPlot object at 0x7f0ec5d9c650>>
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/leus/hasegawa_moonshot_ws/src/jsk_recognition/audio_to_spectrogram/scripts/spectrum_plot.py", line 53, in _cb
    img = convert_matplotlib_to_img(self.fig)
  File "/home/leus/hasegawa_moonshot_ws/src/jsk_recognition/audio_to_spectrogram/src/audio_to_spectrogram/convert_matplotlib.py", line 14, in convert_matplotlib_to_img
[... 186 lines cut ...]
    spectrum_to_spectrogram (audio_to_spectrogram/spectrum_to_spectrogram.py)
    spectrum_view (image_view/image_view)

auto-starting new master
process[master]: started with pid [22155]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to ee26f520-98a7-11ed-8d16-a0cec8428594
process[rosout-1]: started with pid [22177]
started core service [/rosout]
process[audio_capture-2]: started with pid [22184]
process[audio_to_spectrum-3]: started with pid [22185]
process[spectrum_to_spectrogram-4]: started with pid [22186]
process[audio_amplitude_plot-5]: started with pid [22188]
process[spectrum_plot-6]: started with pid [22204]
process[audio_amplitude_view-7]: started with pid [22207]
process[spectrum_view-8]: started with pid [22209]
process[spectrogram_view-9]: started with pid [22210]

# plots are correctly displayed
```

Currently, I have no time to debug this, so I left this issue until I can make some time.

cc. @iory

nakane11, 2023-02-23:
I faced the same error and it disappeared by inserting the following lines before  `import matplotlib.pyplot as plt`:
```
import matplotlib
matplotlib.use('Agg')
```

708yamaguchi, 2023-02-23:
Thank you very much for your reports.

### Problem

We use `TKinter` in the callback functions.
https://github.com/jsk-ros-pkg/jsk_recognition/blob/d72e493c9ff16925a19afe29e1f941391351667d/audio_to_spectrogram/scripts/audio_amplitude_plot.py#L79
https://github.com/jsk-ros-pkg/jsk_recognition/blob/d72e493c9ff16925a19afe29e1f941391351667d/audio_to_spectrogram/scripts/spectrum_plot.py#L53

However, It seems that `Tkinter` GUI loop from outside the main thread is not recommended.
https://stackoverflow.com/questions/14694408/runtimeerror-main-thread-is-not-in-main-loop
https://stackoverflow.com/questions/54237067/how-to-make-tkinter-gui-thread-safe

### Solution

My current solution is to change the matplotlib backend from the default `Tkagg` to non-interactive backend `Agg`.
(This is the same solution as the previous comment https://github.com/jsk-ros-pkg/jsk_recognition/issues/2759#issuecomment-1441194826 )

I don't know much about it, but it seems that using `Agg` backend would be effective for non-interactive environment.
https://matplotlib.org/stable/users/explain/backends.html#selecting-a-backend
http://www2.yukawa.kyoto-u.ac.jp/~koudai.sugimoto/dokuwiki/doku.php?id=python:matplotlib:cui%E7%92%B0%E5%A2%83

FYI: How to print the current backend.
```
In [1]: import matplotlib

In [2]: matplotlib.get_backend()
Out[2]: u'TkAgg'
```

### WIP
I am investigating why this problem was resolved between 2.1.2 and 2.2.0 of matplotlib.

708yamaguchi, 2023-02-23:
@iory 
If you have any ideas, please let me know.

iory, 2023-02-23:
What about using a catkin virtualenv? The version of the python library changes frequently. Also, backward compatibility is not guaranteed.

708yamaguchi, 2023-02-23:
Thank you.
Good idea.

pazeshun, 2023-02-24:
I tried both solutions (1. using `Agg` in apt-version matplotlib (2.1.1), 2. using default `TkAgg` in pip-version matplotlib (2.2.5)) and found that the former is slightly faster than the latter:
```bash
# Using Agg in apt-version matplotlib (2.1.1)
$ rostopic hz /audio_amplitude_plot/output/viz
subscribed to [/audio_amplitude_plot/output/viz]
average rate: 6.928
	min: 0.126s max: 0.172s std dev: 0.01614s window: 7
average rate: 7.207
	min: 0.123s max: 0.172s std dev: 0.01517s window: 14
average rate: 7.180
	min: 0.123s max: 0.172s std dev: 0.01341s window: 21
average rate: 7.083
	min: 0.123s max: 0.172s std dev: 0.01512s window: 28
average rate: 7.046
	min: 0.123s max: 0.172s std dev: 0.01416s window: 35
^Caverage rate: 7.096
	min: 0.121s max: 0.172s std dev: 0.01378s window: 41

# Using default TkAgg in pip-version matplotlib (2.2.5)
$ rostopic hz /audio_amplitude_plot/output/viz
subscribed to [/audio_amplitude_plot/output/viz]
average rate: 6.373
	min: 0.149s max: 0.164s std dev: 0.00534s window: 6
average rate: 6.265
	min: 0.149s max: 0.183s std dev: 0.00851s window: 12
average rate: 6.171
	min: 0.149s max: 0.183s std dev: 0.00894s window: 18
average rate: 6.259
	min: 0.147s max: 0.183s std dev: 0.00889s window: 25
average rate: 6.227
	min: 0.147s max: 0.183s std dev: 0.00992s window: 31
average rate: 6.144
	min: 0.147s max: 0.195s std dev: 0.01261s window: 36
average rate: 5.980
	min: 0.147s max: 0.211s std dev: 0.01695s window: 42
average rate: 6.024
	min: 0.147s max: 0.211s std dev: 0.01645s window: 48
average rate: 6.032
	min: 0.147s max: 0.211s std dev: 0.01605s window: 54
^Caverage rate: 6.045
	min: 0.147s max: 0.211s std dev: 0.01587s window: 56
```

@nakane11 @708yamaguchi @iory How about in your environment?

708yamaguchi, 2023-02-24:
I tested it with `$ roslaunch audio_to_spectrogram sample_audio_to_spectrogram.launch` commend.
I got the same result as @pazeshun .

1. `Agg` and matplotlib (2.1.1). About 5.5Hz.

```
$ rostopic hz /audio_amplitude_plot/output/viz -w10
subscribed to [/audio_amplitude_plot/output/viz]
WARNING: may be using simulated time
average rate: 5.747
	min: 0.141s max: 0.232s std dev: 0.03543s window: 5
average rate: 5.382
	min: 0.141s max: 0.232s std dev: 0.02922s window: 10
average rate: 4.894
	min: 0.171s max: 0.275s std dev: 0.02843s window: 10
average rate: 4.956
	min: 0.171s max: 0.275s std dev: 0.02990s window: 10
average rate: 5.538
	min: 0.151s max: 0.232s std dev: 0.02526s window: 10
average rate: 6.408
	min: 0.141s max: 0.171s std dev: 0.00865s window: 10
average rate: 6.178
	min: 0.141s max: 0.232s std dev: 0.02638s window: 10
```

2. `TkAgg` and matplotlib (2.2.5). About 4.5Hz.

```
$ rostopic hz /audio_amplitude_plot/output/viz -w10
subscribed to [/audio_amplitude_plot/output/viz]
WARNING: may be using simulated time
average rate: 4.656
	min: 0.201s max: 0.232s std dev: 0.01261s window: 4
average rate: 4.340
	min: 0.201s max: 0.274s std dev: 0.02655s window: 8
average rate: 4.249
	min: 0.201s max: 0.274s std dev: 0.02810s window: 10
average rate: 4.231
	min: 0.211s max: 0.263s std dev: 0.01906s window: 10
average rate: 4.376
	min: 0.191s max: 0.252s std dev: 0.01788s window: 10
average rate: 4.271
	min: 0.191s max: 0.272s std dev: 0.02326s window: 10
average rate: 4.214
	min: 0.211s max: 0.272s std dev: 0.02018s window: 10
```

iory, 2023-02-24:
@pazeshun 

I created a PR to make agg the default. Please tell me your opinion.

pazeshun, 2023-02-27:
> I created a PR to make agg the default. Please tell me your opinion.

Thank you! I checked and approved:
https://github.com/jsk-ros-pkg/jsk_recognition/pull/2764#pullrequestreview-1314897737

pazeshun, 2023-02-27:
As I introduced in https://github.com/jsk-ros-pkg/jsk_recognition/pull/2764#pullrequestreview-1314897737, I found that pip-version matplotlib (2.2.5) is slow even when `Agg` is used (both https://github.com/jsk-ros-pkg/jsk_recognition/issues/2759#issuecomment-1441194826 and https://github.com/jsk-ros-pkg/jsk_recognition/pull/2764):
```
$ rostopic hz /spectrum_plot/output/viz 
subscribed to [/spectrum_plot/output/viz]
average rate: 9.538
	min: 0.091s max: 0.120s std dev: 0.00987s window: 9
average rate: 9.323
	min: 0.091s max: 0.123s std dev: 0.00939s window: 18
average rate: 9.241
	min: 0.091s max: 0.123s std dev: 0.00888s window: 27
average rate: 9.155
	min: 0.091s max: 0.125s std dev: 0.00886s window: 36
average rate: 9.195
	min: 0.091s max: 0.125s std dev: 0.00848s window: 45
average rate: 9.192
	min: 0.091s max: 0.125s std dev: 0.00840s window: 54
^Caverage rate: 9.187
	min: 0.091s max: 0.125s std dev: 0.00828s window: 58
```
So I concluded the cause of slowing down is other than `TkAgg`.
Even with #2764 , we will face slowing down when the apt-version matplotlib is updated in the future.
But I think this is not a problem because slowing down is not so big.

pazeshun, 2023-02-27:
Closed via #2764

closed 2023-02-27 by pazeshun