jsk-ros-pkg/jsk_recognition #2776: sample_craft_node.launchが立ち上がらない

sawada10, 2023-05-12:
jsk_perceptionのocrを使って文字認識をしたくて、
https://github.com/jsk-ros-pkg/jsk_recognition/pull/2650
を参考に、sample_craft_node.launchをnaoのカメラで実行しようとしたところ、エラーになりました。
```
$ roslaunch jsk_perception sample_craft_node.launch INPUT_IMAGE:=/nao_robot/naoqi_driver/camera/front/image_raw
... logging to /home/leus/.ros/log/ad7c2642-f074-11ed-ba03-606d3cba6c3b/roslaunch-leus-ThinkPad-P1-Gen-4i-5547.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://leus-ThinkPad-P1-Gen-4i:41591/

SUMMARY
========

CLEAR PARAMETERS
 * /craft_node/
 * /draw_rects/
 * /ocr_node/

PARAMETERS
 * /craft_node/gpu: 0
 * /craft_node/link_threshold: 0.2
 * /craft_node/low_text: 0.2
 * /craft_node/model_path: /home/leus/db_ws/...
 * /craft_node/refine_model_path: /home/leus/db_ws/...
 * /craft_node/text_threshold: 0.7
 * /draw_rects/use_classification_result: False
 * /image_publisher/encoding: bgr8
 * /image_publisher/file_name: /home/leus/db_ws/...
 * /ocr_node/font_path: /home/leus/db_ws/...
 * /ocr_node/language: eng
 * /ocr_node/subscribe_polygon: True
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /
    craft_node (jsk_perception/craft_node.py)
    draw_rects (jsk_perception/draw_rects.py)
    image_publisher (jsk_perception/image_publisher.py)
    image_view00 (image_view/image_view)
    image_view01 (image_view/image_view)
    ocr_node (jsk_perception/ocr_node.py)

ROS_MASTER_URI=http://localhost:11311

process[image_publisher-1]: started with pid [5603]
process[craft_node-2]: started with pid [5604]
process[ocr_node-3]: started with pid [5605]
process[draw_rects-4]: started with pid [5606]
process[image_view00-5]: started with pid [5607]
process[image_view01-6]: started with pid [5608]
Traceback (most recent call last):
  File "/home/leus/db_ws/src/jsk_perception/node_scripts/image_publisher.py", line 24, in <module>
    from jsk_recognition_utils.depth import depth_to_compressed_depth
  File "/home/leus/db_ws/devel/lib/python2.7/dist-packages/jsk_recognition_utils/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 8, in <module>
  File "/home/leus/db_ws/src/jsk_recognition_utils/python/jsk_recognition_utils/mask.py", line 5, in <module>
    from skimage.morphology import binary_closing
  File "/usr/lib/python2.7/dist-packages/skimage/__init__.py", line 158, in <module>
    from .util.dtype import *
  File "/usr/lib/python2.7/dist-packages/skimage/util/__init__.py", line 7, in <module>
    from .arraycrop import crop
  File "/usr/lib/python2.7/dist-packages/skimage/util/arraycrop.py", line 8, in <module>
    from numpy.lib.arraypad import _validate_lengths
ImportError: cannot import name _validate_lengths
Traceback (most recent call last):
  File "/home/leus/db_ws/src/jsk_perception/node_scripts/craft_node.py", line 27, in <module>
    import craft.craft as craft
  File "/home/leus/db_ws/jsk_recognition/jsk_perception/node_scripts/craft/craft.py", line 14, in <module>
    from craft.basenet.vgg16_bn import init_weights
  File "/home/leus/db_ws/jsk_recognition/jsk_perception/node_scripts/craft/basenet/vgg16_bn.py", line 6, in <module>
    from torchvision import models
ImportError: No module named torchvision
Traceback (most recent call last):
  File "/home/leus/db_ws/src/jsk_perception/node_scripts/ocr_node.py", line 20, in <module>
    from jsk_recognition_utils.put_text import put_text_to_image
  File "/home/leus/db_ws/devel/lib/python2.7/dist-packages/jsk_recognition_utils/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 8, in <module>
  File "/home/leus/db_ws/src/jsk_recognition_utils/python/jsk_recognition_utils/mask.py", line 5, in <module>
    from skimage.morphology import binary_closing
  File "/usr/lib/python2.7/dist-packages/skimage/__init__.py", line 158, in <module>
    from .util.dtype import *
  File "/usr/lib/python2.7/dist-packages/skimage/util/__init__.py", line 7, in <module>
    from .arraycrop import crop
  File "/usr/lib/python2.7/dist-packages/skimage/util/arraycrop.py", line 8, in <module>
    from numpy.lib.arraypad import _validate_lengths
ImportError: cannot import name _validate_lengths
Traceback (most recent call last):
  File "/home/leus/db_ws/src/jsk_perception/node_scripts/draw_rects.py", line 10, in <module>
    from jsk_recognition_utils.color import labelcolormap
  File "/home/leus/db_ws/devel/lib/python2.7/dist-packages/jsk_recognition_utils/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 8, in <module>
  File "/home/leus/db_ws/src/jsk_recognition_utils/python/jsk_recognition_utils/mask.py", line 5, in <module>
    from skimage.morphology import binary_closing
  File "/usr/lib/python2.7/dist-packages/skimage/__init__.py", line 158, in <module>
    from .util.dtype import *
  File "/usr/lib/python2.7/dist-packages/skimage/util/__init__.py", line 7, in <module>
    from .arraycrop import crop
  File "/usr/lib/python2.7/dist-packages/skimage/util/arraycrop.py", line 8, in <module>
    from numpy.lib.arraypad import _validate_lengths
ImportError: cannot import name _validate_lengths
[image_publisher-1] process has died [pid 5603, exit code 1, cmd /home/leus/db_ws/src/jsk_perception/node_scripts/image_publisher.py __name:=image_publisher __log:=/home/leus/.ros/log/ad7c2642-f074-11ed-ba03-606d3cba6c3b/image_publisher-1.log].
log file: /home/leus/.ros/log/ad7c2642-f074-11ed-ba03-606d3cba6c3b/image_publisher-1*.log
[craft_node-2] process has died [pid 5604, exit code 1, cmd /home/leus/db_ws/src/jsk_perception/node_scripts/craft_node.py ~input:=/nao_robot/naoqi_driver/camera/front/image_raw __name:=craft_node __log:=/home/leus/.ros/log/ad7c2642-f074-11ed-ba03-606d3cba6c3b/craft_node-2.log].
log file: /home/leus/.ros/log/ad7c2642-f074-11ed-ba03-606d3cba6c3b/craft_node-2*.log
[ocr_node-3] process has died [pid 5605, exit code 1, cmd /home/leus/db_ws/src/jsk_perception/node_scripts/ocr_node.py ~input:=/nao_robot/naoqi_driver/camera/front/image_raw ~input/polygons:=craft_node/output/polygons ~input/rects:=craft_node/output/rects __name:=ocr_node __log:=/home/leus/.ros/log/ad7c2642-f074-11ed-ba03-606d3cba6c3b/ocr_node-3.log].
log file: /home/leus/.ros/log/ad7c2642-f074-11ed-ba03-606d3cba6c3b/ocr_node-3*.log
[draw_rects-4] process has died [pid 5606, exit code 1, cmd /home/leus/db_ws/src/jsk_perception/node_scripts/draw_rects.py ~input:=/nao_robot/naoqi_driver/camera/front/image_raw ~input/rects:=craft_node/output/rects __name:=draw_rects __log:=/home/leus/.ros/log/ad7c2642-f074-11ed-ba03-606d3cba6c3b/draw_rects-4.log].
log file: /home/leus/.ros/log/ad7c2642-f074-11ed-ba03-606d3cba6c3b/draw_rects-4*.log
^C[image_view00-5] killing on exit
[image_view01-6] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

調べた原因としては、numpyのバージョンが1.15だとおそらくうまく行くのですが、1.16だと出るようです。
Pythonのバージョンは2.7です。
どうしたら良いでしょうか。
@iory

mqcmd196, 2023-05-12:
Please tell me your numpy version executing `python -c "import numpy; print(numpy.__version__)" `

sawada10, 2023-05-12:
Thank you for your response.
I don't have my PC right now, so I can't test the command, but from memory, numpy version was `16.1.5`.
As for numpy path, I don't remember, so I will tell you it next time I go to the lab (maybe next Monday).

mqcmd196, 2023-05-12:
Sure, and I only have noetic environment so let's see next week together.

tkmtnt7000, 2023-05-12:
Maybe related to https://github.com/scikit-image/scikit-image/issues/3906?

sawada10, 2023-05-15:
1. whlファイルを使ってtorchとtorchvisionをinstallするのではなく、`pip install --user torch==1.1.0 torchvision==0.3.0`を実行する。
2. このときにinstallされるnumpy-1.16.6を、`pip uninstall numpy`する。
3. aptで入るnumpyを使う。これがnumpy-1.13.3（ubuntu 18）

mqcmd196, 2023-05-15:
whlファイルで `torchvision` をインストールすると torchのバージョンがアップグレードされてしまう